moveit/moveit2 #1428: moveit_servo build on Rolling build farm may be exceeding resource limits.

nuclearsandwich, 2022-07-13:
This issue is largely informational regarding build farm builds and it can be closed without action.
If there is a more preferred communication channel to use in the future, please let me know! 🙇🏼 

### Your environment
* ROS Distro: Rolling so far, but I think Humble is also susceptible.
* OS Version: Ubuntu 22.04 amd64
* Binary deb package: 2.5.1-1

---

Two prior builds ([12](https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__moveit_servo__ubuntu_jammy_amd64__binary/12/) and [16](https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__moveit_servo__ubuntu_jammy_amd64__binary/16)) of `moveit_servo` have failed during compilation on the build farm  while compiling `servo_calcs.cpp.o`. 

Build agents on the build farm are quad-core instances with 8GiB of RAM (specifically AWS c5.xlarge / c5a.xlarge instances), but are configured to run four single core builds in parallel which yields a rough RAM allocation of 8GiB/4 = 2GiB per package build.
This limit is not strictly enforced by the build isolation mechanism but if a build agent is saturated and memory is exhausted the host operating system may take action and kill build processes consuming high amounts of memory. The killed process is usually, but not always, the highest memory consumer among the build processes so there is the possibility of a "false positive" where `moveit_servo` is getting killed but is not actually consuming more than the expected resource limits. However, since the build was stopped in the same place both times. It is most likely that this compile operation in `moveit_servo` is the high memory consumer.

It is still possible for builds of `moveit_servo` to succeed if they are triggered at a time when the build farm is relatively quiet and there are fewer builds running on the agent alongside it. This is why build [17](https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__moveit_servo__ubuntu_jammy_amd64__binary/17/), which I triggered manually this morning, succeeded. However rebuilds of Rolling and new releases of `moveit` may see regular build failures for `moveit_servo` while the build farm is under load.

### Build output

```
02:39:42 [ 19%] Building CXX object CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o
02:40:13 /usr/lib/ccache/c++ -DBACKWARD_HAS_DW=1 -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DRCUTILS_ENABLE_FAULT_INJECTION -Dmoveit_servo_lib_EXPORTS -I/tmp/binarydeb/ros-rolling-moveit-servo-2.5.1/include -isystem /opt/ros/rolling/include/control_msgs -isystem /opt/ros/rolling/include/geometry_msgs -isystem /opt/ros/rolling/include -isystem /opt/ros/rolling/include/moveit_msgs -isystem /opt/ros/rolling/include/pluginlib -isystem /opt/ros/rolling/include/rclcpp -isystem /opt/ros/rolling/include/rclcpp_components -isystem /opt/ros/rolling/include/sensor_msgs -isystem /opt/ros/rolling/include/std_msgs -isystem /opt/ros/rolling/include/std_srvs -isystem /opt/ros/rolling/include/tf2_eigen -isystem /opt/ros/rolling/include/trajectory_msgs -isystem /usr/include/eigen3 -isystem /usr/include/bullet -isystem /opt/ros/rolling/include/rcutils -isystem /opt/ros/rolling/include/rclcpp_action -isystem /opt/ros/rolling/include/action_msgs -isystem /opt/ros/rolling/include/builtin_interfaces -isystem /opt/ros/rolling/include/rosidl_runtime_c -isystem /opt/ros/rolling/include/rosidl_typesupport_interface -isystem /opt/ros/rolling/include/unique_identifier_msgs -isystem /opt/ros/rolling/include/fastcdr -isystem /opt/ros/rolling/include/rosidl_runtime_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/rolling/include/rmw -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/rolling/include/ament_index_cpp -isystem /opt/ros/rolling/include/libstatistics_collector -isystem /opt/ros/rolling/include/rcl -isystem /opt/ros/rolling/include/rcl_interfaces -isystem /opt/ros/rolling/include/rcl_logging_interface -isystem /opt/ros/rolling/include/rcl_yaml_param_parser -isystem /opt/ros/rolling/include/libyaml_vendor -isystem /opt/ros/rolling/include/tracetools -isystem /opt/ros/rolling/include/rcpputils -isystem /opt/ros/rolling/include/statistics_msgs -isystem /opt/ros/rolling/include/rosgraph_msgs -isystem /opt/ros/rolling/include/rosidl_typesupport_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_c -isystem /opt/ros/rolling/include/angles -isystem /opt/ros/rolling/include/shape_msgs -isystem /opt/ros/rolling/include/object_recognition_msgs -isystem /opt/ros/rolling/include/octomap_msgs -isystem /usr/include/libqhull_r -isystem /opt/ros/rolling/include/resource_retriever -isystem /opt/ros/rolling/include/visualization_msgs -isystem /opt/ros/rolling/include/urdf -isystem /opt/ros/rolling/include/urdf_parser_plugin -isystem /opt/ros/rolling/include/urdfdom_headers -isystem /opt/ros/rolling/include/urdfdom -isystem /opt/ros/rolling/include/class_loader -isystem /opt/ros/rolling/include/tf2 -isystem /opt/ros/rolling/include/tf2_ros -isystem /opt/ros/rolling/include/message_filters -isystem /opt/ros/rolling/include/rcl_action -isystem /opt/ros/rolling/include/tf2_msgs -isystem /opt/ros/rolling/include/tf2_geometry_msgs -isystem /opt/ros/rolling/include/kdl_parser -isystem /opt/ros/rolling/include/composition_interfaces -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-moveit-servo-2.5.1=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Wall -Wextra -Wwrite-strings -Wunreachable-code -Wpointer-arith -Wredundant-decls -Wcast-qual -Wno-unused-parameter -Wno-unused-function -Wno-maybe-uninitialized -std=c++17 -MD -MT CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o -MF CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o.d -o CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o -c /tmp/binarydeb/ros-rolling-moveit-servo-2.5.1/src/servo_calcs.cpp
02:40:13 c++: fatal error: Killed signal terminated program cc1plus
```

vatanaksoytezer, 2022-07-13:
Hello @nuclearsandwich really appreciate informing us with this issue. I think all the maintainers are actively looking out for issues so this is a great place to communicate. Do you happen to have any suggestions (maybe a compile flag or sorts) to avoid this issue, and make our build jobs more stable, I'm happy to investigate what can be done but just asking if you have a handy solution you have used before?

tylerjw, 2022-07-13:
I think the issue might just be that it creates a huge translation unit with how much code it includes. Thank you for this and hopefully, soon I can find time to profile that part of our build and refactor it so it is multiple translation units.

AndyZe, 2022-07-13:
I think it might be the large number of integration tests.

tylerjw, 2022-07-13:
> I think it might be the large number of integration tests.

He specifically said it was:

> `moveit_servo` have failed during compilation on the build farm while compiling `servo_calcs.cpp.o`

I don't think compiling any of those integration tests takes much time as at worst they link against `servo_calcs.cpp.o`.

nuclearsandwich, 2022-07-15:
> Do you happen to have any suggestions (maybe a compile flag or sorts) to avoid this issue, and make our build jobs more stable, I'm happy to investigate what can be done but just asking if you have a handy solution you have used before?

I'm not aware of any turnkey fixes. Whenever we've encountered issues with compile sizes someone on our team does exactly what @tylerjw mentioned:

> I hope soon I can find time to profile that part of our build and refactor it so it is multiple translation units.

Ultimately it's one of those trade-offs between effort, developer velocity, and resource requirements that we have to juggle. Reducing compile times and memory usage with more build profiling is something that's come up a few times recently and the recent consensus among my colleagues is that these sorts of issues are things that you investigate when they cause problems but that pro-actively trying to set restrictions or keep things "under control" is too arbitrary and creates a lot of additional work unless you have very exact goals in mind.

Good luck with the profiling and refactoring work @tylerjw! I hope it's fruitful.

henningkayser, 2024-02-27:
I'm closing this since we've merged a major refactoring of Servo a while ago, and I've not seen build resource issues with Servo anymore. Please reopen if you think this is still a problem.

closed 2024-02-27 by henningkayser