moveit/moveit2 #1431: Use clang-format's `QualifierAlignment` upon upgrade to clang-format-14

ChrisThrasher, 2022-07-14:
https://releases.llvm.org/14.0.0/tools/clang/docs/ClangFormatStyleOptions.html

clang-format-14 adds a new feature called `QualifierAlignment` which among other benefits, lets us automate the decision to use `east const` or `const west`. Once we can safely start requiring clang-format-14, enable this feature to enforce whatever style is already more common in MoveIt. Decisions about potentially changing our style will be saved for a later discussion once the tooling is in place to enforce a particular style.

[image]

tylerjw, 2023-08-23:
We do use clang-format-14 for this project on the main branch. Do we want to enable this rule?

ChrisThrasher, 2023-08-23:
I've started using it [elsewhere](https://github.com/SFML/SFML/pull/2608) and quite like it. It removes yet another detail of code style that would otherwise have to get caught and fixed in code review.

Shobuj-Paul, 2023-09-14:
Are we proceeding with this? 
If yes, should I make the change?