moveit/moveit2 #1432: [Hybrid planning] demo test, the panda arm does not move

luozhijing, 2022-07-16:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 22.04
* Source build
* 
* main

### Steps to reproduce
run the command:
"ros2 launch moveit_hybrid_planning hybrid_planning_demo.launch.py"

### Expected behaviour
Tell us what should happen

### Actual behaviour
replan all the time and the arm stand still

### Backtrace or Console output

[component_container_mt-1] [INFO] [1657971775.037030867] [global_planner_component]: Received global planning goal request
[hybrid_planning_demo_node-8] [INFO] [1657971775.037170729] [test_hybrid_planning_client]: Global goal accepted by server
[component_container_mt-1] [WARN] [1657971775.037292578] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'panda_arm' using planner 'ompl'. Will use defaults instead.
[component_container_mt-1] [WARN] [1657971775.037419699] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container_mt-1] [INFO] [1657971775.037582584] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container_mt-1] [INFO] [1657971775.110870262] [local_planner_component]: The local planner has been stuck for several iterations. Aborting.
[component_container_mt-1] [INFO] [1657971775.134308723] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'panda_arm'
[component_container_mt-1] [INFO] [1657971775.135283870] [global_planner_component]: Received global planning goal request
[hybrid_planning_demo_node-8] [INFO] [1657971775.135498669] [test_hybrid_planning_client]: Global goal accepted by server
[component_container_mt-1] [WARN] [1657971775.135638872] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'panda_arm' using planner 'ompl'. Will use defaults instead.

luozhijing, 2022-07-16:
Can someone help me?

AndyZe, 2022-07-16:
I'm sorry I don't have a good answer. I would `grep` for this line in the code and start putting print statements around it, because it seems like this is the issue:

`[component_container_mt-1] [INFO] [1657971775.110870262] [local_planner_component]: The local planner has been stuck for several iterations. Aborting.`

Did you change the start position of the robot, perhaps?

Have you modified any code, anywhere?

Are you building moveit2 repositories from source? In particular, try deleting the debian of `moveit_resources` and clone/build it from source.

luozhijing, 2022-07-17:
Thanks for your reply, it works! @AndyZe

closed 2022-07-17 by luozhijing

AndyZe, 2022-07-17:
Good to hear. Was it building from source that fixed it?