moveit/moveit2 #1435: support HPP-FCL as collision checker

v4hn, 2022-07-19:
See https://github.com/humanoid-path-planner/hpp-fcl

Authors of this heavily-changed version of FCL published [a paper at RSS this year](https://arxiv.org/abs/2205.09663) about speedups of factor 2 for convex collision checking over standard GJK.

Together with providing support to mark meshes as convex in general (!), it could accelerate our current pipelines *a lot*.

vatanaksoytezer, 2022-07-19:
> Authors of this heavily-changed version of FCL published [a paper at RSS this year](https://arxiv.org/abs/2205.09663) about speedups of factor 2 for convex collision checking over standard GJK.

This sounds amazing, definitely, +1 to adding this to be MoveIt should be one of our short-term goals.

simonschmeisser, 2022-08-24:
Please see (and comment on ...) https://github.com/ros-planning/moveit/pull/3167 as well for improving on the broad phase part of the collision checking.

sea-bass, 2023-11-21:
+1 this would be such a great enhancement. I've been trying out Pinocchio, which uses HPP-FCL, and it's quite nice. Also has Python bindings which plays well with our goals with MoveIt.

jcarpent, 2024-02-28:
Hi all,

You may look at our latest benchmarks (see https://github.com/humanoid-path-planner/hpp-fcl?tab=readme-ov-file#a-high-performance-library) done by @lmontaut to see the speed of hpp-fcl compared to alternative open-source solutions:

[image]

jcarpent, 2024-03-03:
@simonschmeisser Do you have a set of scenes you would like to test?
@lmontaut Interested in incorporating the broad phases into your benchmarks?

jcarpent, 2024-03-03:
By the way, [here](https://hal.science/hal-04070039/) is a follow-up of our R:SS conference with several accelerated variants of GJK.