moveit/moveit2 #1439: Configure the number of waypoints passed to Ruckig at once

AndyZe, 2022-07-20:
The Ruckig trajectory smoother currently stretches the entire trajectory in time until all kinematic limits are obeyed. This works but it can slow the trajectory more than necessary. It's better to feed in just a few waypoints at a time, so that only the waypoints that need it get stretched.

The downside of passing just a few waypoints at once is very long runtimes if the trajectory is very large. So, I think the number of waypoints passed to Ruckig at once should be configurable.

Attached is some code to feed in a configurable number of waypoints.

[process_waypoints_in_chunks.txt](https://github.com/ros-planning/moveit2/files/9150216/process_waypoints_in_chunks.txt)

swiz23, 2023-01-06:
Hi @AndyZe,

Decided to test out the PR for adjustable waypoint batch sizes yesterday. Saw something interesting. Refer to the table below and the attached 2 CSV files showing all waypoints. You'll notice that when ruckig is run in batches, it can lengthen the trajectory an unexpected amount. It's unclear to me why. Any ideas? Note that I left the default arguments in runRuckigInBatches.

[image]

[Motion_1.txt](https://github.com/ros-planning/moveit2/files/10362381/Motion_1.txt)

[Motion_2.txt](https://github.com/ros-planning/moveit2/files/10362383/Motion_2.txt)

AndyZe, 2023-02-19:
@swiz23 thanks for testing. I'll likely be looking into this soon. I'll open a new issue for this.

Closing this issue b/c it was completed by #1719

AndyZe, 2023-02-19:
New issue to track the bug noticed by @swiz23:  #1960

closed 2023-02-19 by AndyZe