moveit/moveit2 #1440: Servo singularity warning can speed up unexpectedly

AndyZe, 2022-07-20:
The code [here](https://github.com/ros-planning/moveit2/blob/b9543801be0fdcac6b157d3907d5f33c83ef438c/moveit_ros/moveit_servo/src/servo_calcs.cpp#L793) causes some undesired behavior:

```
  // If this dot product is positive, we're moving toward singularity ==> decelerate
  double dot = vector_toward_singularity.dot(commanded_velocity);
  if (dot > 0)
  {
    velocity_scale = ... // decelerate
```

If the dot product suddenly becomes (equal to or greater than 0), `velocity_scale` jumps back to 1.0 and the robot can speed up suddenly.

This should get low-pass filtered down the line, but it still is undesired behavior.

nbbrooks, 2022-07-20:
If I'm understanding this correctly, this behavior would be addressed by making this behavior default https://github.com/ros-planning/moveit2/pull/620

Henry was working on updating and merging the PR, but might have gotten pulled onto something else

henrygerardmoore, 2022-07-27:
I think we decided that #620 was ready for another review @nbbrooks. I also tested locally and it seemed to achieve the desired effect.

AndyZe, 2022-08-16:
Closed by #620

closed 2022-08-16 by AndyZe