moveit/moveit2 #1442: Error while installing ros-humble-pilz-industrial-motion-planner binary

Yadunund, 2022-07-21:
### Description

Installing moveit2 from binaries on `humble` + `Ubuntu 22.04` seems to throw unmet dependency error. Specifically due to `ros-humble-pilz-industrial-motion-planner`

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary install
* `2.5.1-1jammy.20220625.084800_amd64`

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.
```
sudo apt update && sudo apt upgrade -y
sudo apt install ros-humble-moveit
```

### Expected behaviour
All binaries install without any errors

### Actual behaviour
```
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 ros-humble-moveit-planners : Depends: ros-humble-pilz-industrial-motion-planner but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-moveit] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-resources-panda-moveit-config] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-configs-utils] failed
  apt: command [sudo -H apt-get install -y ros-humble-control-msgs] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-core] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-ros-perception] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-ros-planning-interface] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-servo] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-hybrid-planning] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-visual-tools] failed
  apt: command [sudo -H apt-get install -y ros-humble-rviz-visual-tools] failed
  apt: command [sudo -H apt-get install -y ros-humble-geometric-shapes] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-common] failed
  apt: command [sudo -H apt-get install -y ros-humble-moveit-ros-planning] failed
  apt: Failed to detect successful installation of [ros-humble-moveit]
  apt: Failed to detect successful installation of [ros-humble-moveit-resources-panda-moveit-config]
  apt: Failed to detect successful installation of [ros-humble-moveit-configs-utils]
  apt: Failed to detect successful installation of [ros-humble-moveit-core]
  apt: Failed to detect successful installation of [ros-humble-moveit-ros-perception]
  apt: Failed to detect successful installation of [ros-humble-moveit-ros-planning-interface]
  apt: Failed to detect successful installation of [ros-humble-moveit-servo]
  apt: Failed to detect successful installation of [ros-humble-moveit-hybrid-planning]
  apt: Failed to detect successful installation of [ros-humble-moveit-visual-tools]
  apt: Failed to detect successful installation of [ros-humble-rviz-visual-tools]
  apt: Failed to detect successful installation of [ros-humble-geometric-shapes]
  apt: Failed to detect successful installation of [ros-humble-moveit-common]
  apt: Failed to detect successful installation of [ros-humble-moveit-ros-planning]
```

### Additional infromation:
Running `sudo apt --fix-broken install` returns

```
The following NEW packages will be installed:
  ros-humble-pilz-industrial-motion-planner
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
37 not fully installed or removed.
Need to get 0 B/1,756 kB of archives.
After this operation, 8,659 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
(Reading database ... 311446 files and directories currently installed.)
Preparing to unpack .../ros-humble-pilz-industrial-motion-planner_2.5.1-1jammy.20220625.084800_amd64.
deb ...
Unpacking ros-humble-pilz-industrial-motion-planner (2.5.1-1jammy.20220625.084800) ...
dpkg: error processing archive /var/cache/apt/archives/ros-humble-pilz-industrial-motion-planner_2.5.
1-1jammy.20220625.084800_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/include/joint_limits/joint_limits.hpp', which is also in packag
e ros-humble-joint-limits 2.12.1-1jammy.20220714.172107
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ros-humble-pilz-industrial-motion-planner_2.5.1-1jammy.20220625.084800_amd64
.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

### Temporary solution

```
sudo dpkg -i --force-overwrite /var/cache/apt/archives/ros-humble-pilz-industrial-motion-planner_2.5.1-1jammy.20220625.084800_amd64.deb

sudo apt --fix-broken install

```

mikeferguson, 2022-07-21:
Looks like this now exists in ros2_control upstream, and these files should be removed from moveit. Per the comment in our codebase:

```
// TODO(henning): This file is copied from the DRAFT PR https://github.com/ros-controls/ros2_control/pull/462 since the
// current ros2_control implementation does not offer the desired joint limits API, yet. Remove when ros2_control has an
// upstream version of this.
```

mikeferguson, 2022-07-21:
Opened a PR to fix this - but won't actually fix anything until we get a new Debian release and a sync

mikeferguson, 2022-07-21:
@vatanaksoytezer I know you just cut a release, but we'll probably want another one after we resolve this issue (since the Debians aren't installable)

tylerjw, 2022-07-22:
We just cut a release but haven't bloomed it yet due to https://answers.ros.org/question/403942/how-to-change-trim-sub-directory/

mikeferguson, 2022-07-22:
@tylerjw I'm fairly certain that release will still have this issue - since the joint_limits folder is being installed by us and ros2_control's new "joint_limits" package.

tylerjw, 2022-07-22:
Oof, I'll investigate this to see if I can find a solution.

mikeferguson, 2022-07-22:
That PR I opened (#1443) would in theory fix the problem - but it's not as straightforward as the TODO makes it seem - it looks like what eventually got merged upstream is not what we've been using (they dropped support for the namespace part, and it looks like we do leverage that in the pilz planner)

vatanaksoytezer, 2022-07-25:
> @vatanaksoytezer I know you just cut a release, but we'll probably want another one after we resolve this issue (since the Debians aren't installable)

Sorry I'm late to the party, I'm happy to cut a release any time we need, just let me know when you need one. it takes only 10-15 minutes. 2.5.2 is also now bloomed and should be available in testing repositories.

vatanaksoytezer, 2022-07-29:
2.5.3 released!

closed 2022-07-29 by henningkayser