moveit/moveit2 #1445: Moveit Setup Assistant installed via apt doesn't launch GUI and throws error

rajeshpachaikani, 2022-07-24:
### Description
Installed moveit setup assistant via apt and unable to launch moveit setup assistant GUI

### Your environment
* ROS Distro: humble
* OS Version: Ubuntu 22.04
* Binary build
* 2.5.1

### Steps to reproduce
Installed moveit set up assistant via apt using 
`sudo apt-get install ros-humble-moveit-setup-assistant`
Launch moveit setup assistant via either
`ros2 launch moveit_setup_assistant setup_assistant.launch` 
or 
`ros2 run moveit_setup_assistant moveit_setup_assistant`

### Expected behaviour
Moveit Setup assistant GUI should be launched 

### Actual behaviour
Receiving the following error message

### Backtrace or Console output
when using ros2 run
`No executable found` 

when using ros2 launch
```[INFO] [launch]: All log files can be found below /home/rajesh/.ros/log/2022-07-24-14-12-02-635655-rajesh-GL552VX-52601
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [launch]: Error trying to process rule "substitution":

Unknown substitution: arg
```

mikeferguson, 2022-07-24:
The 2.5.2 release will be the first with a truly functional moveit_setup_assistant (the release has been tagged, but not yet made it to the build farm or sync).

rajeshpachaikani, 2022-07-24:
Thank you for the quick reply. May I know the estimated time for 2.5.2 release in apt package manager?

closed 2022-07-24 by rajeshpachaikani

tylerjw, 2022-07-25:
> Thank you for the quick reply. May I know the estimated time for 2.5.2 release in apt package manager?

We are running bloom today which will open a PR to rosdistro. When that will make it into apt depends then on when the next Humble or Rolling (depending on which one you use) sync is. You can follow this tag to find out when the next sync is: https://discourse.ros.org/c/release/16