moveit/moveit2 #1447: Error calling IKCallbackFn from IK Solver

ravnicas, 2022-07-25:
### Description

On humble and higher an error is thrown using a port of trac-ik solver. Error wasn't obvious or didnt trigger on galactic or lower, until 2.4.x.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22
* Source or Binary build? Source
* If binary, which release version? Humble, tags: 2.5.0, 2.5.1, 2.5.2

### Steps to reproduce
Clone and build this [repo](https://github.com/ravnicas/trac_ik). Plan any robot with a Pose Goal.

### Expected behaviour
A solution should be returned.

### Actual behaviour
This error is thrown.

As far as got, the solver does actually solve the ik problem, the error is thrown somewhere in the IKCallbackFn.

`[move_group-4] terminate called after throwing an instance of 'boost::wrapexcept<boost::bad_function_call>'
[move_group-4]   what():  call to empty boost::function
[move_group-4] Stack trace (most recent call last) in thread 88151:
[move_group-4] #17   Object "", at 0xffffffffffffffff, in
[move_group-4] #16   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f8efa0389ff]
[move_group-4] #15   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f8ef9fa6b42]
[move_group-4] #14   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8efa2362c2, in
[move_group-4] #13   Object "/opt/ros/humble/lib/x86_64-linux-gnu/libompl.so.1.5.2", at 0x7f8ec741c37f, in ompl::base::GoalLazySamples::goalSamplingThread()
[move_group-4] #12   Object "/home/roboknecht/roboct_humble/moveit2_ws/install/moveit_planners_ompl/lib/libmoveit_ompl_interface.so.2.5.0", at 0x7f8eec385b5b, in ompl_interface::ConstrainedGoalSampler::sampleUsingConstraintSampler(ompl::base::GoalLazySamples const*, ompl::base::State*)
[move_group-4] #11   Object "/home/roboknecht/roboct_humble/moveit2_ws/install/moveit_core/lib/libmoveit_constraint_samplers.so.2.5.0", at 0x7f8ef40310cf, in constraint_samplers::UnionConstraintSampler::project(moveit::core::RobotState&, unsigned int)
[move_group-4] #10   Object "/home/roboknecht/roboct_humble/moveit2_ws/install/moveit_core/lib/libmoveit_constraint_samplers.so.2.5.0", at 0x7f8ef402ccd2, in constraint_samplers::IKConstraintSampler::sampleHelper(moveit::core::RobotState&, moveit::core::RobotState const&, unsigned int, bool)
[move_group-4] #9    Object "/home/roboknecht/roboct_humble/moveit2_ws/install/moveit_core/lib/libmoveit_constraint_samplers.so.2.5.0", at 0x7f8ef4026821, in constraint_samplers::IKConstraintSampler::callIK(geometry_msgs::msg::Pose_<std::allocator<void> > const&, boost::function<void (geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, moveit_msgs::msg::MoveItErrorCodes_<std::allocator<void> >&)> const&, double, moveit::core::RobotState&, bool)
[move_group-4] #8    Object "/home/roboknecht/roboct_humble/roboct_ws/install/trac_ik_kinematics/lib/libmoveit_trac_ik_kinematics_plugin.so", at 0x7f8ef4246e37, in trac_ik_kinematics_plugin::TRAC_IKKinematicsPlugin::searchPositionIK(geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, double, std::vector<double, std::allocator<double> >&, moveit_msgs::msg::MoveItErrorCodes_<std::allocator<void> >&, kinematics::KinematicsQueryOptions const&) const
[move_group-4] #7    Object "/home/roboknecht/roboct_humble/roboct_ws/install/trac_ik_kinematics/lib/libmoveit_trac_ik_kinematics_plugin.so", at 0x7f8ef42440bc, in trac_ik_kinematics_plugin::TRAC_IKKinematicsPlugin::searchPositionIK(geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, double, std::vector<double, std::allocator<double> >&, boost::function<void (geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::vector<double, std::allocator<double> > const&, moveit_msgs::msg::MoveItErrorCodes_<std::allocator<void> >&)> const&, moveit_msgs::msg::MoveItErrorCodes_<std::allocator<void> >&, std::vector<double, std::allocator<double> > const&, kinematics::KinematicsQueryOptions const&) const [clone .cold]
[move_group-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8efa208557, in __cxa_throw
[move_group-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8efa2082f6, in std::terminate()
[move_group-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8efa20828b, in
[move_group-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f8efa1fcbfd, in
[move_group-4] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f8ef9f3a7f2]
[move_group-4] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f8ef9f54475]
[move_group-4] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[move_group-4]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[move_group-4]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f8ef9fa8a7c]
[move_group-4] Aborted (Signal sent by tkill() 87991 1000)
[ERROR] [move_group-4]: process has died [pid 87991, exit code -6, cmd '/home/roboknecht/roboct_humble/moveit2_ws/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_cx4xx7wl --params-file /tmp/launch_params_lr73or5l --params-file /tmp/launch_params_jl3v_s2w --params-file /tmp/launch_params_ns8whfmm --params-file /tmp/launch_params_mk_3pxfu --params-file /tmp/launch_params_a6fqft5u --params-file /tmp/launch_params_xp2mkyua --params-file /tmp/launch_params_x9tffxau'].
[move_group-4]`

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

vatanaksoytezer, 2022-07-25:
Hello @ravnicas, thank you very much for reporting the issue. Which port of trac-ik are you using? I haven't seen something similar before with other IK solvers, so I'm questioning whether it might be related to trac-ik port you are using.

ravnicas, 2022-07-25:
I linked the port of trac-ik which I'm using. The lib was ported by another user, I mostly ported the plugin part.

It was functional on foxy and galactic.

ravnicas, 2022-07-26:
i didn't see the change from boost::bind to std::bind immediatly.. therefore my solution_callback was wrong in newer versions.

closed 2022-07-26 by ravnicas