moveit/moveit2 #1448: Moveit setup assistant for ros foxy

stevedanomodolor, 2022-07-26:
I am currently using foxy, having to use ros1 moveit setup assistant to then convert it back to ros2 gets pretty annoying.

mikeferguson, 2022-07-26:
I'm not sure how difficult this would be - I think there are a bunch of prereq's that were added to support the MSA in Humble and beyond - @DLu would probably have a better idea if this is actually feasible

DLu, 2022-07-27:
The launch files generated by MSA require use of [`moveit_configs_utils`](https://github.com/ros-planning/moveit2/tree/main/moveit_configs_utils) which has not been ported yet. See #1449

I started to do that port but I was distracted by porting Latched Strings instead (https://github.com/ros-planning/moveit2/pull/1414) which makes the launch files a bit simpler.

stevedanomodolor, 2022-07-27:
@DLu is there anything I can do to help?

DLu, 2022-07-28:
> @DLu is there anything I can do to help?

Not at this moment, but thanks

stevedanomodolor, 2022-07-31:
> > @DLu is there anything I can do to help?
> 
> Not at this moment, but thanks

Any update?

mikeferguson, 2022-07-31:
> Any update?

Another prerequisite PR for this is here: https://github.com/ros-planning/moveit2/pull/1450 - but also requires upstream changes to moveit_resources.

Given the complexity of the set of changes, I imagine that this backport is going to take some time.

closed 2022-12-13 by stevedanomodolor