moveit/moveit2 #1465: Humble: MoveIt2 Getting Started Tutorial - colcon build failed (moveit_ros_planning)

vetletj, 2022-08-01:
### Description
Not able to build (MoveIt tutorials source) colcon workspace correctly. Seems like there is a problem with _moveit_ros_planning_ package.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build (Debian)
* Desktop install

### Steps to reproduce
Follow [Moveit2 "Getting Started" Tutorial](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html#install-ros-2-and-colcon). 

### Expected behaviour
Build colcon workspace without error. 

### Actual behaviour
```
Failed   <<< moveit_ros_planning [51.2s, exited with code 2]

Summary: 32 packages finished [2min 29s]
  1 package failed: moveit_ros_planning
  1 package had stderr output: moveit_ros_planning
  30 packages not processed
```

### Console output
```
tht@tht-stationary02:~/ws_moveit2$ colcon build --mixin release
[0.186s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'hardware_interface' is in: /opt/ros/humble
	'controller_manager' is in: /opt/ros/humble
	'ros2_control_test_assets' is in: /opt/ros/humble
	'moveit_simple_controller_manager' is in: /opt/ros/humble
	'moveit_planners_ompl' is in: /opt/ros/humble
	'moveit_kinematics' is in: /opt/ros/humble
	'controller_manager_msgs' is in: /opt/ros/humble
	'moveit_ros_move_group' is in: /opt/ros/humble
	'moveit_ros_visualization' is in: /opt/ros/humble
	'moveit_servo' is in: /opt/ros/humble
	'controller_interface' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager controller_manager_msgs hardware_interface moveit_kinematics moveit_planners_ompl moveit_ros_move_group moveit_ros_visualization moveit_servo moveit_simple_controller_manager ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Starting >>> joint_limits
Starting >>> rviz_visual_tools
Starting >>> rosparam_shortcuts
Finished <<< moveit_common [0.32s]                                    
Finished <<< moveit_resources_panda_description [0.44s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_configs_utils [0.51s]                            
Finished <<< moveit_resources_fanuc_description [0.54s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< ros2_control_test_assets [0.58s]
Starting >>> hardware_interface
Finished <<< moveit_resources_pr2_description [0.60s]                         
Finished <<< moveit_resources_prbt_support [0.61s]
[... 66 lines cut ...]
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_display_random_state.dir/build.make:318: planning_components_tools/moveit_display_random_state] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:960: planning_components_tools/CMakeFiles/moveit_display_random_state.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_publish_scene_from_text.dir/build.make:318: planning_components_tools/moveit_publish_scene_from_text] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1050: planning_components_tools/CMakeFiles/moveit_publish_scene_from_text.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/build.make:319: planning_components_tools/moveit_visualize_robot_collision_volume] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:990: planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_planning [51.2s, exited with code 2]

Summary: 32 packages finished [2min 29s]
  1 package failed: moveit_ros_planning
  1 package had stderr output: moveit_ros_planning
  30 packages not processed
```

Thanks for considering this.

JafarAbdi, 2022-08-01:
Hi @vetletj,

Do you have moveit2 debians installed (run `apt policy ros-humble-moveit-ros-planning`)? Could you please try a clean build?

tylerjw, 2022-08-01:
It looks like you are missing `moveit_resources` for some reason. You should be able to get that using rosdep or you can put that in your workspace.

vetletj, 2022-08-01:
Hello @tylerjw, 

Thanks for the input! 

Tried to clone the [repository](https://github.com/ros-planning/moveit_resources) into my workspace but then I'm getting this error:
```
tht@tht-stationary02:~/ws_moveit2$ colcon build
[0.171s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'moveit_planners_ompl' is in: /opt/ros/humble
	'moveit_simple_controller_manager' is in: /opt/ros/humble
	'moveit_ros_move_group' is in: /opt/ros/humble
	'moveit_servo' is in: /opt/ros/humble
	'moveit_kinematics' is in: /opt/ros/humble
	'moveit_ros_visualization' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding moveit_kinematics moveit_planners_ompl moveit_ros_move_group moveit_ros_visualization moveit_servo moveit_simple_controller_manager

This may be promoted to an error in a future release of colcon-override-check.
[0.171s] ERROR:colcon:colcon build: Duplicate package names not supported:
- moveit_resources_prbt_ikfast_manipulator_plugin:
  - src/moveit2/moveit_planners/test_configs/prbt_ikfast_manipulator_plugin
  - src/moveit2/moveit_resources/prbt_ikfast_manipulator_plugin
- moveit_resources_prbt_moveit_config:
  - src/moveit2/moveit_planners/test_configs/prbt_moveit_config
  - src/moveit2/moveit_resources/prbt_moveit_config
- moveit_resources_prbt_pg70_support:
  - src/moveit2/moveit_planners/test_configs/prbt_pg70_support
  - src/moveit2/moveit_resources/prbt_pg70_support
- moveit_resources_prbt_support:
  - src/moveit2/moveit_planners/test_configs/prbt_support
  - src/moveit2/moveit_resources/prbt_support

```

JafarAbdi, 2022-08-01:
> Hello @tylerjw,
>
> Thanks for the input!
>
> Tried to clone the [repository](https://github.com/ros-planning/moveit_resources) into my workspace but then I'm getting this error:
>
> ``` [...]

You need to check out the ros2 branch

vetletj, 2022-08-01:
Aha! Went ahead a bit to fast there, thanks. 

Didn't see this before now so in the meanwhile I've deleted ROS 2 Humble and reinstalled it, now everything with the MoveIt2 source build went smoothly for some reason i don't know. Guess something was off with my underlay. 

Sadly I'm still not able to build the MoveIt2 tutorials workspace without error:
```
tht@tht-stationary02:~/ws_moveit2$ colcon build --mixin release
[0.182s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'hardware_interface' is in: /opt/ros/humble
	'ros2_control_test_assets' is in: /opt/ros/humble
	'controller_manager' is in: /opt/ros/humble
	'controller_interface' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Starting >>> joint_limits
Starting >>> rviz_visual_tools
Starting >>> rosparam_shortcuts
Finished <<< moveit_common [0.24s]                                  
Finished <<< moveit_resources_panda_description [0.24s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_prbt_support [0.33s]                   
Finished <<< moveit_resources_fanuc_description [0.35s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_pr2_description [0.36s]
Finished <<< ros2_control_test_assets [0.40s]                        
Starting >>> hardware_interface
Finished <<< moveit_resources_panda_moveit_config [0.22s]
Starting >>> moveit_core
Finished <<< moveit_configs_utils [0.50s]                            
Finished <<< moveit_resources_fanuc_moveit_config [0.20s]
Starting >>> moveit_resources
Finished <<< joint_limits [0.60s]                                            
Finished <<< moveit_resources [0.17s]                                         
Finished <<< rosparam_shortcuts [0.85s]                                       
Finished <<< rviz_marker_tools [1.05s] 
Finished <<< hardware_interface [0.71s]
Starting >>> controller_interface
Starting >>> transmission_interface
Finished <<< controller_manager_msgs [1.17s]
Finished <<< rviz_visual_tools [1.46s] 
Finished <<< moveit_task_constructor_msgs [1.86s]
Finished <<< controller_interface [0.90s]                             
Starting >>> controller_manager
Finished <<< transmission_interface [0.91s]
Finished <<< moveit_core [1.71s]                                       
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< controller_manager [1.29s]                                       
Starting >>> ros2controlcli
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [1.31s]
Finished <<< chomp_motion_planner [1.32s]
Starting >>> moveit_chomp_optimizer_adapter
[... 78 lines cut ...]
/home/tht/ws_moveit2/src/moveit_task_constructor/core/src/solvers/cartesian_path.cpp:106:60: error: template argument 1 is invalid
  106 |             props.get<kinematics::KinematicsBase::IKCostFn>("kinematics_cost_fn"));
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/tht/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/solvers/planner_interface.h:43,
                 from /home/tht/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/solvers/cartesian_path.h:41,
                 from /home/tht/ws_moveit2/src/moveit_task_constructor/core/src/solvers/cartesian_path.cpp:39:
/home/tht/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/properties.h:332:18: note: candidate: ‘template<class T> const T& moveit::task_constructor::PropertyMap::get(const string&, const T&) const’
  332 |         const T& get(const std::string& name, const T& fallback) const {
      |                  ^~~
/home/tht/ws_moveit2/src/moveit_task_constructor/core/include/moveit/task_constructor/properties.h:332:18: note:   template argument deduction/substitution failed:
/home/tht/ws_moveit2/src/moveit_task_constructor/core/src/solvers/cartesian_path.cpp:106:60: error: template argument 1 is invalid
  106 |             props.get<kinematics::KinematicsBase::IKCostFn>("kinematics_cost_fn"));
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
[... cut ...] k_constructor_core.dir/solvers/cartesian_path.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:254: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_core [14.2s, exited with code 2]

Summary: 53 packages finished [38.1s]
  1 package failed: moveit_task_constructor_core
  1 package had stderr output: moveit_task_constructor_core
  4 packages not processed
```

JafarAbdi, 2022-08-02:
Do you mind running `vcs export src/`? Looks like you have an outdated version of moveit2

shu-rt, 2022-08-04:
Hi. I ran into this same issue.  
For me, I was using the [ros-planning/moveit2](https://github.com/ros-planning/moveit2) `humble` branch instead of the `main` branch.  
This is because I ran into a build error from `moveit_ros_planning` pakcage.  

I looked into the `humble` branch and figured out that the [`kinematics_base`](https://github.com/ros-planning/moveit2/blob/main/moveit_core/kinematics_base) package didn't have the member `IKCostFn` as shown in the error log above.   

Is it possible to merge the current `main` branch into `humble` branch or will there be some other conflcts of some sort.

vetletj, 2022-08-04:
Thanks for still trying to help @JafarAbdi, much appreciated. Here is the output from `vcs export src`:
```
tht@tht-stationary02:~/ws_moveit2$ vcs export src
repositories:
  moveit2:
    type: git
    url: https://github.com/ros-planning/moveit2.git
    version: humble
  moveit2_tutorials:
    type: git
    url: https://github.com/ros-planning/moveit2_tutorials
    version: main
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources
    version: ros2
  moveit_task_constructor:
    type: git
    url: https://github.com/ros-planning/moveit_task_constructor.git
    version: ros2
  moveit_visual_tools:
    type: git
    url: https://github.com/ros-planning/moveit_visual_tools
    version: ros2
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control.git
    version: 300328516752c4020c561fd1a2dc96e9f52d3033
  rosparam_shortcuts:
    type: git
    url: https://github.com/PickNikRobotics/rosparam_shortcuts
    version: ros2
  rviz_visual_tools:
    type: git
    url: https://github.com/PickNikRobotics/rviz_visual_tools.git
    version: ros2
```

vatanaksoytezer, 2022-08-04:
I recommend using main branch for Humble for now, I will take a look on what needs to be done to properly support humble branch with tutorials.

vetletj, 2022-08-05:
Thanks for the input @vatanaksoytezer ! Really appreciate all you who try to help us starting out. 

I tried to change the moveit2 and moveit2_tutorials to main branch, but then getting (as @shu-rt pointed out) build error from `moveit_ros_planning` package.   

```
tht@tht-stationary02:~/ws_moveit2$ vcs export src
repositories:
  moveit2:
    type: git
    url: https://github.com/ros-planning/moveit2.git
    version: main
  moveit2_tutorials:
    type: git
    url: https://github.com/ros-planning/moveit2_tutorials.git
    version: main
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources
    version: ros2
  moveit_task_constructor:
    type: git
    url: https://github.com/ros-planning/moveit_task_constructor.git
    version: ros2
  moveit_visual_tools:
    type: git
    url: https://github.com/ros-planning/moveit_visual_tools
    version: ros2
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control.git
    version: 300328516752c4020c561fd1a2dc96e9f52d3033
  rosparam_shortcuts:
    type: git
    url: https://github.com/PickNikRobotics/rosparam_shortcuts
    version: ros2
  rviz_visual_tools:
    type: git
    url: https://github.com/PickNikRobotics/rviz_visual_tools.git
    version: ros2
```

```
tht@tht-stationary02:~/ws_moveit2/src$ git clone -b humble https://github.com/ros-planning/moveit2.git
Cloning into 'moveit2'...
remote: Enumerating objects: 112485, done.
remote: Counting objects: 100% (679/679), done.
remote: Compressing objects: 100% (405/405), done.
remote: Total 112485 (delta 338), reused 500 (delta 248), pack-reused 111806
Receiving objects: 100% (112485/112485), 81.07 MiB | 2.54 MiB/s, done.
Resolving deltas: 100% (67681/67681), done.
tht@tht-stationary02:~/ws_moveit2/src$ cd ..
tht@tht-stationary02:~/ws_moveit2$ colcon build --mixin release
[0.187s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_interface' is in: /opt/ros/humble, /home/tht/ws_moveit2/install/controller_interface
	'hardware_interface' is in: /home/tht/ws_moveit2/install/hardware_interface, /opt/ros/humble
	'controller_manager' is in: /home/tht/ws_moveit2/install/controller_manager, /opt/ros/humble
	'ros2_control_test_assets' is in: /home/tht/ws_moveit2/install/ros2_control_test_assets, /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
[... 83 lines cut ...]
gmake[1]: *** [CMakeFiles/Makefile2:820: planning_scene_monitor/CMakeFiles/trajectory_monitor_tests.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_display_random_state.dir/build.make:318: planning_components_tools/moveit_display_random_state] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:960: planning_components_tools/CMakeFiles/moveit_display_random_state.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_publish_scene_from_text.dir/build.make:318: planning_components_tools/moveit_publish_scene_from_text] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1050: planning_components_tools/CMakeFiles/moveit_publish_scene_from_text.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
[... cut ...]
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/build.make:319: planning_components_tools/moveit_visualize_robot_collision_volume] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:990: planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_planning [49.2s, exited with code 2]

Summary: 32 packages finished [2min 11s]
  1 package failed: moveit_ros_planning
  1 package had stderr output: moveit_ros_planning
  30 packages not processed
```

vatanaksoytezer, 2022-08-07:
@vetletj can you try a clean build by removing 'build` and 'install` folders and running the build command? Your branches look good to me.

vetletj, 2022-08-08:
Deleted `build` and `install` folder and tried a clean build, but got same output as above. Also tried to uninstall setuptools `pip uninstall setuptools` and reinstall the latest version, cause i had 58.2.0 version installed. Tried again with a clean build, but unfortunately still same error. 
```
tht@tht-stationary02:~/ws_moveit2$ colcon build --mixin release
[0.184s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_manager' is in: /opt/ros/humble
	'controller_interface' is in: /opt/ros/humble
	'hardware_interface' is in: /opt/ros/humble
	'ros2_control_test_assets' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Starting >>> joint_limits
Starting >>> rviz_visual_tools
Starting >>> rosparam_shortcuts
Finished <<< moveit_common [0.31s]                                    
Finished <<< moveit_resources_panda_description [0.45s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_configs_utils [0.49s]                            
Finished <<< moveit_resources_prbt_support [0.54s]
Finished <<< moveit_resources_fanuc_description [0.57s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< ros2_control_test_assets [0.61s]                                 
Starting >>> hardware_interface
Finished <<< moveit_resources_pr2_description [0.61s]
Finished <<< moveit_resources_panda_moveit_config [0.38s]                    
Starting >>> moveit_core
Finished <<< moveit_resources_fanuc_moveit_config [0.38s]                    
Starting >>> moveit_resources
Finished <<< moveit_resources [0.47s]                                        
Finished <<< rviz_marker_tools [7.41s] 
Finished <<< controller_manager_msgs [9.19s]
Finished <<< joint_limits [12.0s]       
Finished <<< moveit_task_constructor_msgs [12.1s]
Finished <<< rosparam_shortcuts [13.9s]                                       
Finished <<< hardware_interface [19.4s]                                       
Starting >>> controller_interface
Starting >>> transmission_interface
Finished <<< rviz_visual_tools [27.9s]                                         
Finished <<< controller_interface [9.27s]                               
Starting >>> controller_manager
Finished <<< transmission_interface [12.7s]                             
Finished <<< controller_manager [29.9s]                                 
Starting >>> ros2controlcli
Finished <<< ros2controlcli [0.44s]                                     
Starting >>> ros2_control
Finished <<< ros2_control [0.36s]                                           
Finished <<< moveit_core [1min 18s]                                
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_simple_controller_manager [7.80s]
Starting >>> moveit_plugins
Starting >>> moveit_ros_control_interface
Finished <<< moveit_plugins [0.45s]          
[... 32 lines cut ...]
gmake[1]: *** [CMakeFiles/Makefile2:990: planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_display_random_state.dir/build.make:318: planning_components_tools/moveit_display_random_state] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:960: planning_components_tools/CMakeFiles/moveit_display_random_state.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
[... cut ...]
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_evaluate_collision_checking_speed.dir/build.make:318: planning_components_tools/moveit_evaluate_collision_checking_speed] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1020: planning_components_tools/CMakeFiles/moveit_evaluate_collision_checking_speed.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_planning [51.1s, exited with code 2]

Summary: 32 packages finished [2min 25s]
  1 package failed: moveit_ros_planning
  1 package had stderr output: moveit_ros_planning
  30 packages not processed
```

vetletj, 2022-08-08:
Just because I was completely lost, I deleted and reinstalled ROS 2 Humble again, and then tried to build MoveIt2 from source. Now it seems like everything builds fine except three stdrr outputs (SetuptoolsDeprecationWarning). Guess it didn't build correctly before because I didn't have the latest version of setuptools?

For clarity, this is how I managed to build MoveIt2 and MoveIt2 Tutorials from source on Ubuntu 22.04 ROS 2 Humble:
* ROS 2 Humble - Ubuntu 22.04 (Debian) packages - Desktop install
* Check that you have the latest version of setuptools `pip install setuptools` (59.6.0)
* humble branch for moveit2 and main branch for moveit2_tutorials
* when building colcon workspace use `--parallel-workers 1`

At least this is how i managed to make it work.

vatanaksoytezer, 2022-08-08:
setuptools is just a warrning related to ROS2 itself and should not affected build Nevertheless, I'm happy it compiled you get stuff working, thank you for sticking around!

closed 2022-08-12 by vetletj

andy-Chien, 2022-08-17:
I had same issue as this.
It seems like compiler linked to a wrong library, so I removed ros-humble-moveit* installed by rosdep and rebuild it.
It's work.

B6Infinity, 2024-04-02:
Hey there... Just ran into the issue.
Executing
`rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`
did the trick for me.

Refer [here](https://github.com/ros-planning/moveit2_tutorials/issues/388#issuecomment-1137090802)