moveit/moveit2 #1475: termination_condition for OMPL has no effect

AndyZe, 2022-08-03:
### Description

My base `ompl_planning.yaml` is attached; I'm using the persistentPRMstar planner. I can't find any way to modify the planning time.

I've tried these parameters:

  PersistentPRMstar:
    type: geometric::PRMstar
    termination_condition: Iteration[50]   (or `Iteration[100]` or `CostConvergence[10,.1]`)

panda_arm:
  nr_attempts: 50
  
And I dropped in a high-resolution C++ timer to test planning duration:

```
    auto start_time = Clock::now();
    auto success = (move_group_interface.plan(plan) == moveit::core::MoveItErrorCode::SUCCESS);
    auto end_time = Clock::now();
    RCLCPP_INFO_STREAM(
        LOGGER,
        "Planning time: " << std::chrono::duration_cast<std::chrono::milliseconds>(end_time - start_time).count());
```

Planning time is always the same within +/-200ms or so. (Also, the roadmap which ought to be loaded seems to have no effect on planning time.)

mamoll, 2022-08-03:
Can you set a breakpoint [here](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L531) and check that the correct PlannerTerminationCondition object is constructed?

AndyZe, 2022-08-03:
> Can you set a breakpoint [here](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L531) and check that the correct PlannerTerminationCondition object is constructed?

Yep, that function is called

AndyZe, 2022-08-04:
I dug through the code a little. The misleading bit of code is here in `model_based_planning_context.cpp`:

```
  else if (termination_and_params[0] == "Iteration")
  {
    if (termination_and_params.size() > 1)
    {
      return ob::plannerOrTerminationCondition(
          ob::timedPlannerTerminationCondition(timeout - ompl::time::seconds(ompl::time::now() - start)),
          ob::IterationTerminationCondition(std::stoul(termination_and_params[1])));
    }
    else
    {
      RCLCPP_ERROR(LOGGER, "Missing argument to Iteration termination condition");
    }
  }
```

So there is a termination OR condition. This is more of a misleading API/documentation issue than an actual bug.

mamoll, 2022-08-04:
So if you set planning time to "infinity", the planner terminates after the specified number of iterations?

AndyZe, 2022-08-05:
I didn't test `infinity` but I did test both termination conditions separately and they both seem fine.

AndyZe, 2022-08-05:
I guess I'll just close this...

closed 2022-08-05 by AndyZe