moveit/moveit2 #1476: Merge / Fast forward main into humble

shu-rt, 2022-08-04:
### Description

Hi, I'm having trouble with building the `main` branch. I get an error from the `moveit_ros _planning` package.
However, when I switch to the `humble` branch, it then works.

On the other hand, I get a build error from `moveit_task_constructor_core` which apparently uses the library referenced from moveit2's `main` branch.

Thus, it would be nice if the `humble` branch gets fast forwarded or gets the `main` branch merged into it.

### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 20.04
* Source build
* If source, which branch? `main` branch

### Steps to reproduce
Here's my build command.
```sh
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --packages-up-to moveit
```

### Expected behaviour
A clean build.

### Actual behaviour
An error as shown below.
```sh
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_prbt_support
--- stderr: moveit_configs_utils                                                                                                              
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [0.68s]
Finished <<< moveit_common [0.71s]                                                                              
Starting >>> moveit_core
Finished <<< moveit_resources_prbt_support [0.78s]                               
Finished <<< moveit_core [1min 46s]                               
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> moveit_simple_controller_manager
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_simple_controller_manager [12.8s]                                                                                                                
Starting >>> moveit_plugins
Finished <<< moveit_plugins [0.73s]                                                                                                                                 
Finished <<< pilz_industrial_motion_planner_testutils [16.6s]                                                                                                       
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [20.9s]                                                                                             
Finished <<< moveit_ros_occupancy_map_monitor [22.7s] 
Starting >>> moveit_ros_planning                           
--- stderr: moveit_ros_planning                                          
/usr/bin/ld: libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `/usr/bin/ld: libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
/usr/bin/ld: libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
collect2: error: ld returned 1 exit status
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_scene_monitor/CMakeFiles/current_state_monitor_tests.dir/build.make:320: planning_scene_monitor/current_state_monitor_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:788: planning_scene_monitor/CMakeFiles/current_state_monitor_tests.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[2]: *** [planning_scene_monitor/CMakeFiles/trajectory_monitor_tests.dir/build.make:320: planning_scene_monitor/trajectory_monitor_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:820: planning_scene_monitor/CMakeFiles/trajectory_monitor_tests.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_evaluate_collision_checking_speed.dir/build.make:318: planning_components_tools/moveit_evaluate_collision_checking_speed] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1020: planning_components_tools/CMakeFiles/moveit_evaluate_collision_checking_speed.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/build.make:319: planning_components_tools/moveit_visualize_robot_collision_volume] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:990: planning_components_tools/CMakeFiles/moveit_visualize_robot_collision_volume.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_publish_scene_from_text.dir/build.make:318: planning_components_tools/moveit_publish_scene_from_text] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1050: planning_components_tools/CMakeFiles/moveit_publish_scene_from_text.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setCollisionObjectUpdateCallback(boost::function<void (std::shared_ptr<collision_detection::World::Object const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_print_planning_scene_info.dir/build.make:318: planning_components_tools/moveit_print_planning_scene_info] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:930: planning_components_tools/CMakeFiles/moveit_print_planning_scene_info.dir/all] Error 2
/usr/bin/ld: ../planning_scene_monitor/libmoveit_planning_scene_monitor.so.2.5.3: undefined reference to `planning_scene::PlanningScene::setAttachedBodyUpdateCallback(boost::function<void (moveit::core::AttachedBody*, bool)> const&)'
[... cut ...] const> const&, collision_detection::World::Action)> const&)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [planning_components_tools/CMakeFiles/moveit_display_random_state.dir/build.make:318: planning_components_tools/moveit_display_random_state] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:960: planning_components_tools/CMakeFiles/moveit_display_random_state.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_ros_planning [1min 13s, exited with code 2]
```

vatanaksoytezer, 2022-08-07:
Hello @shu-rt thank you for reporting your issue. Please try to use main branch for `moveit2` and `moveit2_tutorials` and ros2 branch for `moveit_task_constructor` for now. If that is the case, please issue clean build by removing `build` and `install` folders and running the build command?

vatanaksoytezer, 2022-08-07:
My bad, so I just noticed we have fast-forwarded humble into 2.5.3, so this issue should not be relevant for moveit2. However, we should take similar action for moveit2_tutorials which has a separate issue already. Thus, I'm closing this issue. Please feel free to  re-open or open a new issue if you still experience problems.

closed 2022-08-07 by vatanaksoytezer