moveit/moveit2 #1477: MoveIt2 for ros2 foxy python project

im-renpei, 2022-08-04:
Hi, how can I use MoveIt2 for my customized robot in the ROS2 Python project (as I noticed that currently MoveIt2 foxy only provides c++ interfaces).

(1) As for the MoveIt Setup Assistant, it seems that we need to first generate configuration files in MoveIt1 then manually modified them to adapt to MoveIt2. Is this the only way at present?
(2) In the ROS2 Foxy Python3 project, if we can only use the MoveIt2 Foxy C++ interface, do you have any suggestions?

tylerjw, 2022-08-04:
1. We just released a ported MSA on Rolling/Humble. There is a chance you could use a source build or one of the docker images on Rolling/Humble.
2. There is a couple of ongoing projects to add python support, but this is only for rolling. You can find information in other issues or on Discord. https://discord.gg/xXpGUjNA

im-renpei, 2022-08-04:
Hi @tylerjw , appreciate your timely reply! I can directly use Rolling/Humble MSA to generate MoveIt2 config files and then use them in the MoveIt2 Foxy version, right?

Does anyone of you have suggestions for using the MoveIt2 Python interface in the Foxy version? Such as recommended repos?

tylerjw, 2022-08-04:
I do not know of any projects to add python support for Foxy.

I do believe the configs for humble/rolling are much closer to what will need to Foxy than the ones generated by MSA on ROS 1.

mikeferguson, 2022-08-04:
> I do believe the configs for humble/rolling are much closer to what will need to Foxy than the ones generated by MSA on ROS 1.

Only possible challenge is a few changes in the moveit_config_builder stuff - which is being back ported to foxy here: https://github.com/ros-planning/moveit2/pull/1450