moveit/moveit2 #1478: Floating joint interpolation takes the long way around

werner291, 2022-08-05:
Hi, I believe I found a bug in the joint value interpolation code for floating joints.

See the snippet below:

```
const double jv1[] = {0,0,0, 0, 0, 0.862044, 0.506833};
const double jv2[] = { 0,0,0, 0, 0, -0.969825, 0.243803 };

double intp[7];

moveit::core::FloatingJointModel fjm("joint", 0, 0);

fjm.interpolate(jv1, jv2, 0.5, intp);

Eigen::Isometry3d tf;

fjm.computeTransform(jv1, tf);
Eigen::AngleAxisd rotA(tf.rotation());

fjm.computeTransform(jv2, tf);
Eigen::AngleAxisd rotB(tf.rotation());

fjm.computeTransform(intp, tf);
Eigen::AngleAxisd rotC(tf.rotation());

std::cout << "AA: " << rotA.angle() << " " << rotA.axis().transpose() << std::endl; // 2.07858 0 0 1
std::cout << "BB: " << rotB.angle() << " " << rotB.axis().transpose() << std::endl; // 2.64902 -0 -0 -1
std::cout << "CC: " << rotC.angle() << " " << rotC.axis().transpose() << std::endl; // 0.285223  0  0 -1
```

Note how the interpolated angle has value 0.285: it's taking the long way around, with the interpolation going through 0 instead of Pi/-Pi radians.

werner291, 2022-08-05:
I believe this line (https://github.com/ros-planning/moveit2/blob/main/moveit_core/robot_model/src/floating_joint_model.cpp#L146)  is supposed to prevent exactly this case, but evidently it does not appear to be working quite right.

werner291, 2022-08-05:
Wait a minute... how is `dq < 0` ever going to happen if you take the absolute value earlier?

werner291, 2022-08-05:
I think the formula got copied wrong... I'mma try moving that `fabs` around to see what happens.

werner291, 2022-08-05:
PR should fix the issue, I threw in a test as well.

henningkayser, 2022-08-09:
PR is merged, so I'm closing this

closed 2022-08-09 by henningkayser