moveit/moveit2 #1490: Stale bot for PRs

vatanaksoytezer, 2022-08-09:
Add a GH bot that would warn the authors and/or reviewers when a PR goes stale (2 weeks of response?). This should be annoying enough so that we prevent valuable PRs and work go stale. We may add an additional feature that would close the PR if it warned enough times and didn't get responses / updates.

v4hn, 2022-08-10:
Stale-PR/Issue notifications are a lot of junk in your notification list/inbox. More unread/marked-as-read content does not help.
I'm sorry, but I think it's the wrong response to the problem (which is a normal one in Open Source projects).

Instead, if PickNik can afford it, add dedicated well-defined working hours for some engineer(s) explicitly on PR handling (instead of specific grander projects) and decide a ranking for all open PRs/issues on a bi-weekly basis. Then have them spend time on the top of the list and enforce that people do not wait for feedback from PR authors but are allowed to push their own cleanup patches or create follow-up PRs.

henningkayser, 2023-03-17:
I'm proposing a version of this here https://github.com/ros-planning/moveit2/pull/2022.

@v4hn I agree that stale labels and notifications won't help with identifying inactivity (it's fairly easy to do this using the normal GitHub tools). The purpose of this action is really to require the authors of issues to be responsible for their proposal, and to not bloat the issue list with ill-considered wishes or thoughts just to get it out there and hope that someone else will magically spend a few days to produce a solution. Issues are a communication tool, and they should be seen as an invitation for an active discussion by the author. If no-one even bothers to discuss an issue for 3 months, it's obviously not worthy of discussion and it can be closed (same thing I'm doing manually all the time). Note that feature discussions and roadmap items can be exempt from this explicitly. With PRs it's also different, that is why I disabled automatic closing for them as well. Here the GHA acts as an automatic reminder for the author and for us to take care of this work item and to find a solution for moving forward. It obviously doesn't replace putting in the hours for doing the reviews.