moveit/moveit2 #1501: Optional "latency" parameter when checking trajectory tolerances

AndyZe, 2022-08-10:
Sometimes there is significant latency in a hardware control loop. Let's say it's 50ms.

When checking trajectory execution tolerances, it would be nice to artificially delay (commanded joints) by 50ms when compared against `/joint_states`. This would often allow smaller, more realistic tolerances to be used.

henningkayser, 2022-08-11:
This somewhat sounds like a workaround for a different problem, but I see that it would be useful for debugging. But wouldn't the latency vary a lot for each run in such a case? I think instead of using a fixed parameter, we can directly read the latency of each run from the goal future or the response callback [here](https://github.com/ros-planning/moveit2/blob/main/moveit_plugins/moveit_simple_controller_manager/src/follow_joint_trajectory_controller_handle.cpp#L82). The response duration can be added to the expected start time, or we could even warn/error if latency is way too high.

gavanderhoorn, 2022-08-11:
Isn't this about feedback delay @AndyZe? Not so much about how long it takes an action server to respond to a goal?

henningkayser, 2022-08-11:
When reading "trajectory execution", I'm thinking about actions. Andy, please clarify if it's about streamed commands (Servo?). In that case, I'd be curious how you measure latency against joint_states. Assuming you have a good matching method, you could also try to measure latency at runtime by comparing a subset of the commands with buffered joint_states. A fixed latency parameter could still be added, but still I would think that the actual delay would fluctuate too much for it to be very useful.

gavanderhoorn, 2022-08-11:
It's also applicable to actions.

Tolerances would always apply, and if the controller checks tracking error, latency between command and feedback may cause your controller to go into a fault state.

> Assuming you have a good matching method, you could also try to measure latency at runtime by comparing a subset of the commands with buffered joint_states

that sounds overly complicated for something which is most of the time a constant. On most robots I('ve) work(ed) with, there is a delay (due to filtering and/or buffering) of up to 300 ms between command and feedback.

Without any way to configure your controller (or execution manager) to take that into account, it will only 'see' a position deviation, which, depending on configured tolerances, will lead it to conclude there's too much tracking error.

henningkayser, 2022-08-11:
> It's also applicable to actions.

 Probably, but I don't think that this is currrently the case in MoveIt. Afaik, trajectory tolerance checking is only implemented in the ros2 controller, so any latencies should be considered in the control loop. The MoveIt controller handle (or trajectory execution manager) is only monitoring duration and doesn't send subsequent commands that could be matched to any feedback.

> Tolerances would always apply and be constant, and if the controller checks tracking error, latency between command and feedback may cause your controller to go into a fault state (due to feedback lagging too much behind commanded state).
True, but this also seems to only apply to the controller implementation in ros2_controllers.

> that sounds overly complicated for something which is most of the time a constant. On most robots I('ve) work(ed) with, there is a delay (due to filtering and/or buffering) of up to 300 ms between command and feedback.
300ms sounds extreme. I agree that there is always a constant delay which we mostly handled with tuning filters in the past. A constant latency param could help with that. But the bigger issue (to me) are spurious latency spikes in the message pipeline. 

> Without any way to configure your controller (or execution manager) to take that into account, it will only 'see' a position deviation, which, depending on configured tolerances, will lead it to conclude there's too much tracking error.
The execution manager doesn't track position deviation in the first place, all it does is duration monitoring. I think we are not even monitoring the action feedback at this time, [which could tell us about the actual tracking error](https://github.com/ros-controls/control_msgs/blob/galactic-devel/control_msgs/action/FollowJointTrajectory.action#L51).

AndyZe, 2022-08-11:
You guys are correct, this isn't about Servo. It's about trajectories. This was a good summary:

`Tolerances would always apply and be constant, and if the controller checks tracking error, latency between command and feedback may cause your controller to go into a fault state (due to feedback lagging too much behind commanded state).`

I guess Henning is on the right track, this issue belongs more in ros2_control. Will close this.

closed 2022-08-11 by AndyZe