moveit/moveit2 #1505: Build fail during moveit_setup_srdf_plugins

Cryoschrome, 2022-08-11:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source Build
* Humble branch; commit: [7dd0434](https://github.com/ros-planning/moveit2/commit/7dd0434f742d6586f32843802135ebcc2720c79c)

### Steps to reproduce
Hi I am trying to build MoveIt2 from source. I get the following error while building.
```
--- stderr: moveit_setup_srdf_plugins
/home/cryos/ws_moveit2/src/moveit2/moveit_setup_assistant/moveit_setup_srdf_plugins/include/moveit_setup_srdf_plugins/rotated_header_view.hpp:0: Note: No relevant classes found. No output generated.
---
```
### Expected behaviour
moveit2 builds successfully

### Actual behaviour
Build fails due to the above mentioned error

vatanaksoytezer, 2022-08-11:
Hello @Cryoschrome, thank you for reporting your issue. A clean build (by removing your build and install directories) should fix this.

Cryoschrome, 2022-08-11:
Hello @vatanaksoytezer . Thank for the reply. Cleaning and building did the trick. 
Thank you!

closed 2022-08-11 by Cryoschrome