moveit/moveit2 #1506: Failed to execute trajectory. No controllers found

Cryoschrome, 2022-08-12:
### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source Build
* Humble branch; commit: [7dd0434](https://github.com/ros-planning/moveit2/commit/7dd0434f742d6586f32843802135ebcc2720c79c)

Hi, 
I have created a URDF file consisting of only one joint test MoveIt2. I have generated the configuration and launch files using `moveit_setup_assistant`. Launching `demo.py`, I am able to plan the trajectory but get the following error while executing the trajectory.

```
[move_group-3] You can start planning now!
[move_group-3] 
[ros2_control_node-5] [ERROR] [1660287632.576754302] [controller_manager]: The 'type' param was not defined for 'joint_state_broadcaster'.
[spawner-6] [INFO] [1660287632.595321648] [spawner_rotate_joint_controller]: Waiting for '/controller_manager' node to exist
[ERROR] [spawner-7]: process has died [pid 20098, exit code 1, cmd '/home/cryos/ws_moveit2/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --ros-args'].
[rviz2-4] [INFO] [1660287632.756185625] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1660287632.756360386] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1660287632.787299277] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-5] [ERROR] [1660287632.802626856] [controller_manager]: The 'type' param was not defined for 'rotate_joint_controller'.
[rviz2-4] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-4]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[ERROR] [spawner-6]: process has died [pid 20096, exit code 1, cmd '/home/cryos/ws_moveit2/install/controller_manager/lib/controller_manager/spawner rotate_joint_controller --ros-args'].
[rviz2-4] [ERROR] [1660287635.978754799] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-4] [INFO] [1660287635.996635161] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-4] [INFO] [1660287636.101178252] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0553721 seconds
[rviz2-4] [INFO] [1660287636.101247149] [moveit_robot_model.robot_model]: Loading robot model 'one_joint'...
[rviz2-4] Link rotate_joint had 0 children
[rviz2-4] [INFO] [1660287636.129264888] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-4] [INFO] [1660287636.130505870] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-4] [INFO] [1660287636.142442876] [interactive_marker_display_94300198827264]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-4] Link rotate_joint had 0 children
[rviz2-4] [INFO] [1660287636.150739509] [moveit_ros_visualization.motion_planning_frame]: group rotate_joint
[rviz2-4] [INFO] [1660287636.150780825] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'rotate_joint' in namespace ''
[rviz2-4] [INFO] [1660287636.161254802] [move_group_interface]: Ready to take commands for planning group rotate_joint.
[rviz2-4] [INFO] [1660287636.162023355] [interactive_marker_display_94300198827264]: Sending request for interactive markers
[rviz2-4] [INFO] [1660287636.162718322] [moveit_ros_visualization.motion_planning_frame]: group rotate_joint
[rviz2-4] [INFO] [1660287636.193252689] [interactive_marker_display_94300198827264]: Service response received for initialization
[rviz2-4] [INFO] [1660287665.376076009] [move_group_interface]: MoveGroup action client/server ready
[move_group-3] [INFO] [1660287665.377775263] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-3] [INFO] [1660287665.378312961] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[rviz2-4] [INFO] [1660287665.378402145] [move_group_interface]: Planning request accepted
[move_group-3] [INFO] [1660287666.378570203] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[move_group-3] Check clock synchronization if your are running ROS across multiple machines!
[move_group-3] [WARN] [1660287666.378668894] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to fetch current robot state.
[move_group-3] [INFO] [1660287666.378768087] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-3] [INFO] [1660287666.378897114] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'ompl'
[move_group-3] [INFO] [1660287666.380389086] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'rotate_joint' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-3] [INFO] [1660287666.402330383] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was computed successfully.
[rviz2-4] [INFO] [1660287666.403234666] [move_group_interface]: Planning request complete!
[rviz2-4] [INFO] [1660287666.479199060] [move_group_interface]: time taken to generate plan: 0.0131463 seconds
[move_group-3] [INFO] [1660287667.969461226] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-3] [INFO] [1660287667.969752491] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-3] [INFO] [1660287667.969836746] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-3] [INFO] [1660287667.969863255] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-3] [INFO] [1660287667.969894683] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-3] [INFO] [1660287667.969905332] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-3] [ERROR] [1660287667.969930929] [moveit_ros.trajectory_execution_manager]: Unable to identify any set of controllers that can actuate the specified joints: [ base_to_rotate_joint ]
[move_group-3] [ERROR] [1660287667.969946207] [moveit_ros.trajectory_execution_manager]: Known controllers and their joints:
[move_group-3] 
[rviz2-4] [INFO] [1660287667.969831737] [move_group_interface]: Execute request accepted
[rviz2-4] [INFO] [1660287667.970409940] [move_group_interface]: Execute request aborted
[rviz2-4] [ERROR] [1660287668.069143893] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```

Here are my config files
### moveit_controllers.yaml
```
# MoveIt uses this configuration for controller management

moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - rotate_joint_controller

  rotate_joint_controller:
    type: FollowJointTrajectory
    joints:
      - base_to_rotate_joint
```
### ros2_control.xacro
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="one_joint_ros2_control" params="name initial_positions_file">
        <xacro:property name="initial_positions" value="${load_yaml(initial_positions_file)['initial_positions']}"/>

        <ros2_control name="${name}" type="system">
            <hardware>
                <!-- By default, set up controllers for simulation. This won't work on real hardware -->
                <plugin>mock_components/GenericSystem</plugin>
            </hardware>
            <joint name="base_to_rotate_joint">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['base_to_rotate_joint']}</param>
                </state_interface>
                <state_interface name="velocity"/>
            </joint>

        </ros2_control>
    </xacro:macro>
</robot>
```
<br>
Could anyone explain what is going on?<br>
Thank you!

AndyZe, 2022-08-12:
You also need a `ros2_controllers.yaml`. I bet that is where the error is. Example here: https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/ros2_controllers.yaml

Cryoschrome, 2022-08-15:
Thanks for the reply @AndyZe!
I have added ros2_controllers.yaml 
```
# This config file is used by ros2_control
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    rotate_joint_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

rotate_joint_controller:
  ros__parameters:
    command_interfaces:
      - position
      - velocity
    state_interfaces:
      - position
      - velocity
    joints:
      - base_to_rotate_joint
```
I am still facing the same issue. The complete terminal log is [here](https://gist.github.com/Cryoschrome/2e6a47a87e8ff89509822db4a7fb2585)

AndyZe, 2022-08-15:
@AltinayGrass @Cryoschrome did you guys copy that file from somewhere? Should we update it there to avoid more confusion in the future?

closed 2022-08-15 by AndyZe

AltinayGrass, 2022-08-15:
no, we create the file with setup_assistant @AndyZe. 

Humble and Moveit binary install on ubuntu 22.04

I cloned ros-planning/moveit2 main repositories and build only setup_assistant. We found line differences from moveit_resources/panda_config

AndyZe, 2022-08-15:
Created issue #1514 to track this. Thanks!

Hdriwn, 2024-02-05:
> Hello,
> 
> We had the same problem. We edited the moveit_controllers.yaml file with the text editor and added the two lines at the bottom.
> 
> ```
> action_ns: follow_joint_trajectory
> default: true
> ```
> 
> complete file like below;
> 
> ```
> # MoveIt uses this configuration for controller management
> 
> moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager
> 
> moveit_simple_controller_manager:
>   controller_names:
>     - arm_controller
> 
>   arm_controller:
>     type: FollowJointTrajectory
>     joints:
>       - L1
>       - L2
>       - L3
>       - L4
>       - L5
>       - L6
>     action_ns: follow_joint_trajectory
>     default: true
> ```

This didn't work for me, but the editing the **moveit_controllers.yaml** to the following worked.
```

# MoveIt uses this configuration for controller management
moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

controller_names:
  - arm_controller
arm_controller:
  action_ns: follow_joint_trajectory
  type: FollowJointTrajectory
  default: true
  joints:
    - robot_j1
    - robot_j2
    - robot_j3
    - robot_j4
    - robot_j5
    - robot_j6
```