moveit/moveit2 #1511: robot_description parameter syncing

tylerjw, 2022-08-15:
A single node should receive the robot_description config and all other nodes that share the need for that parameter should get it from that one node. The way this works should be made simpler and standardized.

If this is via a global parameter server, latched topics, or the ROS 2 parameter interface, should be investigated for the various tradeoffs to each approach.

mikeferguson, 2022-08-16:
I think the "standard" approach for ROS2 is to use the "robot_description" topic - usually published by the "robot_state_publisher". RVIZ supports this and so far in my experience, moveit2 is the only thing managing a second copy of the robot_description in it's parameters.

henningkayser, 2024-02-27:
While parameter syncing is a very valid approach, we have already had topic syncing implemented for a while.. I'm closing this for now, we can address this again once we are dealing with synchronization issues with things like dynamic robot models.

closed 2024-02-27 by henningkayser