moveit/moveit2 #1524: Custom low-level controller

Cryoschrome, 2022-08-19:
Hi, I am trying to develop the MoveIt2! pipeline for a custom manipulator. I would like to use a custom controller for my robot as ros2_control seems redundant as the motor driver has a control system built-in. So, all i need to do is send the trajectory generated by moveit to the micro-controller responsible for controlling the motor. I did read about FollowJointTrajectoryAction interface of moveit but I am confused on how to work with it. Is FollowJointTrajectoryAction interface the only way or is there any other method to send the trajectory.

I am using MoveIt2! with ROS Humble on Ubuntu 22.04

Please advice on how to implement it.
Any help is appreciated!
Thank you!

metanav, 2022-08-21:
You would need an action server of type FollowJointTrajectoryAction to receive the goals. You can implement it on Ubuntu and send the commands to MCU over serial.

Jason-Lee0, 2023-10-31:
Sorry to bother you guys, I have another problem. 

If I want to use moveit2 servo which using joystick to control the robot arm,  create the server of type FollowJointTrajectoryAction can also rechieve the work ?

- ROS2 version : Humble
- Ubuntu 22.04

Thanks for your reading.
Have a nice day!!

AndyZe, 2023-10-31:
It's no problem @Jason-Lee0. An action server won't work for Servo. It's too slow to respond and the code isn't compatible with it anyway.

Jason-Lee0, 2023-10-31:
@AndyZe Thanks for your quick reply.
An action server won't work for Servo, so how could I controll the real robot with moveit servo ? 

Could you give some advice to deal with it?

Thanks for your reading.
Sorry to bother you.

Best Regards.

AndyZe, 2023-10-31:
First check if the driver of the robot is capable of streaming commands. Then make a controller that accepts std_msgs/Float64MultiArray or trajectory_msgs/JointTrajectory message types.

The standard way to do this is with ros2_control. Examples here:  https://github.com/ros-controls/ros2_control_demos

mikeferguson, 2024-10-23:
This question appears to have been resolved by linking to appropriate demos

closed 2024-10-23 by mikeferguson