moveit/moveit2 #1526: Asking GitHub repo credentials while building MoveIt2 on MacOS

metanav, 2022-08-21:
I am trying to build MoveIt2 on MacOS Catalina (10.15.7). I have installed ROS2 Humble successfully.  While executing the command below it require credentials for GitHub repo,

```

rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y                                                        

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_ros_planning_interface: No definition of [python3] for OS [osx]
moveit_planners_ompl: No definition of [libomp-dev] for OS [osx]
moveit_ros_perception: No definition of [libomp-dev] for OS [osx]
moveit_ros_benchmarks: No definition of [libboost-filesystem] for OS [osx]
Continuing to install resolvable dependencies...
Error: No available formula or cask with the name "ros/humble/ros2_controllers".
Please tap it and then try again: brew tap ros/humble
Error: No available formula or cask with the name "ros/humble/realtime_tools".
Please tap it and then try again: brew tap ros/humble
Error: No available formula or cask with the name "ros/humble/joint_state_publisher".
Please tap it and then try again: brew tap ros/humble
Error: No available formula or cask with the name "ros/humble/joystick_drivers".
Please tap it and then try again: brew tap ros/humble
executing command [brew install qt5]
Warning: qt@5 5.15.5_1 is already installed and up-to-date.
To reinstall 5.15.5_1, run:
  brew reinstall qt@5
executing command [brew install ros/humble/ros2_controllers]
==> Tapping ros/humble
Cloning into '/usr/local/Homebrew/Library/Taps/ros/homebrew-humble'...
Username for 'https://github.com': 

```

tylerjw, 2022-08-22:
This has something to do with your git config. It looks like it is trying to clone a repo `ros2_controllers` and because of how you have your system configured it is trying to authenticate to github.com to do the clone.

metanav, 2022-08-22:
I think it is trying to clone from the repository https://github.com/ros/homebrew-humble which is either private or not available.

gavanderhoorn, 2022-08-22:
The default behaviour of Github is to ask for credentials for any repository which does not exist -- or is indeed private.

My guess is, it's the former.

metanav, 2022-08-22:
Has anyone installed MoveIt2 with ROS2 humble on MacOS suceesfully?

metanav, 2022-08-23:
I was able to manually clone the repositories which were failed while running rosdep rosdep. Now I get the errors while running the command:

```

$ colcon build   --mixin release

Starting >>> moveit_core
[Processing: moveit_core]                             
[Processing: moveit_core]                                  
[Processing: moveit_core]                                        
--- stderr: moveit_core                                          
In file included from /Users/naveen/ws_moveit2/src/moveit2/moveit_core/collision_detection_bullet/src/bullet_integration/ros_bullet_utils.cpp:34:
/Users/naveen/ws_moveit2/src/moveit2/moveit_core/collision_detection_bullet/include/moveit/collision_detection_bullet/bullet_integration/ros_bullet_utils.h:25:10: fatal error: 'octomap_msgs/conversions.h' file not found
#include <octomap_msgs/conversions.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
gmake[2]: *** [collision_detection_bullet/CMakeFiles/moveit_collision_detection_bullet.dir/build.make:160: collision_detection_bullet/CMakeFiles/moveit_collision_detection_bullet.dir/src/bullet_integration/ros_bullet_utils.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /Users/naveen/ws_moveit2/src/moveit2/moveit_core/collision_detection_bullet/src/collision_env_bullet.cpp:39:
/Users/naveen/ws_moveit2/src/moveit2/moveit_core/collision_detection_bullet/include/moveit/collision_detection_bullet/bullet_integration/ros_bullet_utils.h:25:10: fatal error: 'octomap_msgs/conversions.h' file not found
#include <octomap_msgs/conversions.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
gmake[2]: *** [collision_detection_bullet/CMakeFiles/moveit_collision_detection_bullet.dir/build.make:118: collision_detection_bullet/CMakeFiles/moveit_collision_detection_bullet.dir/src/collision_env_bullet.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2286: collision_detection_bullet/CMakeFiles/moveit_collision_detection_bullet.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_core [1min 49s, exited with code 2]

```

The `octomap_msgs` package is already installed.
```

install/octomap_msgs: 
├── include
│   └── octomap_msgs
│       ├── conversions.h
│       └── octomap_msgs
│           ├── msg
│           │   ├── detail
│           │   │   ├── dds_fastrtps
│           │   │   ├── octomap__builder.hpp
│           │   │   ├── octomap__functions.c
│           │   │   ├── octomap__functions.h
│           │   │   ├── octomap__rosidl_typesupport_fastrtps_c.h

```

mikeferguson, 2024-10-23:
Calling this a duplicate of #1725

closed 2024-10-23 by mikeferguson