moveit/moveit2 #1536: Hybrid Planner: Cannot find planning configuration for 'arm'

marc-wittwer, 2022-08-24:
### Description

I added the hybrid planner node-launch-section to the move_group.launch.py from the panda_moveit_config.

My launch file:
<details>
<summary>Toggle for my launch file</summary>

```
#!/usr/bin/env -S ros2 launch
"""Configure and setup move group for planning with MoveIt 2"""

from os import path
from typing import List

import yaml
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.conditions import IfCondition
from launch.substitutions import (
    Command,
    FindExecutable,
    LaunchConfiguration,
    PathJoinSubstitution,
    PythonExpression,
)
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

from launch_ros.descriptions import ComposableNode
from launch_ros.actions import ComposableNodeContainer

def generate_launch_description():

    # Declare all launch arguments
    declared_arguments = generate_declared_arguments()

    # Get substitution for all arguments
    description_package = LaunchConfiguration("description_package")
    description_filepath = LaunchConfiguration("description_filepath")
    moveit_config_package = "panda_moveit_config"
    name = LaunchConfiguration("name")
    prefix = LaunchConfiguration("prefix")
    gripper = LaunchConfiguration("gripper")
    collision_arm = LaunchConfiguration("collision_arm")
    collision_gripper = LaunchConfiguration("collision_gripper")
    safety_limits = LaunchConfiguration("safety_limits")
    safety_position_margin = LaunchConfiguration("safety_position_margin")
    safety_k_position = LaunchConfiguration("safety_k_position")
    safety_k_velocity = LaunchConfiguration("safety_k_velocity")
    ros2_control = LaunchConfiguration("ros2_control")
    ros2_control_plugin = LaunchConfiguration("ros2_control_plugin")
    ros2_control_command_interface = LaunchConfiguration(
        "ros2_control_command_interface"
    )
    gazebo_preserve_fixed_joint = LaunchConfiguration(
        "gazebo_preserve_fixed_joint")
    enable_servo = LaunchConfiguration("enable_servo")
    enable_rviz = LaunchConfiguration("enable_rviz")
    rviz_config = LaunchConfiguration("rviz_config")
    use_sim_time = LaunchConfiguration("use_sim_time")
    log_level = LaunchConfiguration("log_level")

    # URDF
    _robot_description_xml = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution(
[... 560 lines cut ...]
```
</details>
<details>
<summary>hybrid_planning_manager.yaml</summary>

```
planner_logic_plugin_name: "moveit_hybrid_planning/ReplanInvalidatedTrajectory"

```
</details>

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Source or Binary build?
* If binary, which release version? galactic

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
The hybrid planner should process the request like in the tutorial.

### Actual behaviour
The robot in RViz stands still and does not move. However, I can see the planned path in RViz as a ghost.

marc-wittwer, 2022-08-24:
<details>
<summary>Toggle for log output</summary>

```
[move_group-7] [WARN] [1661361821.552472570] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-7] [ERROR] [1661361821.552527370] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[move_group-7] [INFO] [1661361821.556557476] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[component_container-5] [INFO] [1661361821.561221384] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0 seconds
[component_container-5] [INFO] [1661361821.561326484] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[component_container-5] [INFO] [1661361821.561359484] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ign gazebo-1] WARNING: CPU random generator seem to be failing, disable hardware random number generation
[ign gazebo-1] WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
[component_container-5] Link panda_link0 had 1 children
[component_container-5] Link panda_link1 had 1 children
[component_container-5] Link panda_link2 had 1 children
[component_container-5] Link panda_link3 had 1 children
[component_container-5] Link panda_link4 had 1 children
[component_container-5] Link panda_link5 had 1 children
[component_container-5] Link panda_link6 had 1 children
[component_container-5] Link panda_link7 had 1 children
[component_container-5] Link panda_link8 had 1 children
[component_container-5] Link panda_hand had 3 children
[component_container-5] Link panda_leftfinger had 0 children
[component_container-5] Link panda_rightfinger had 0 children
[component_container-5] Link panda_hand_tcp had 0 children
[move_group-7] [INFO] [1661361821.632079098] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[component_container-5] [INFO] [1661361821.647701824] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[component_container-5] [INFO] [1661361821.649129426] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-7] [INFO] [1661361821.650104128] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-7] [INFO] [1661361821.650187928] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-7] [INFO] [1661361821.650215828] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-7] [INFO] [1661361821.650463528] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-7] [INFO] [1661361821.650514328] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.314160
[move_group-7] [INFO] [1661361821.650544128] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-7] [INFO] [1661361821.650583629] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-7] [INFO] [1661361821.650614929] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-7] [INFO] [1661361821.650681329] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[move_group-7] [ERROR] [1661361821.650982629] [moveit.ros_planning.planning_pipeline]: Exception while loading planning adapter plugin 'default_planner_request_adapters/SetMaxCartesianEndEffectorSpeed': According to the loaded plugin descriptions the class default_planner_request_adapters/SetMaxCartesianEndEffectorSpeed with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  chomp/OptimizerAdapter default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddRuckigTrajectorySmoothing default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds default_planner_request_adapters/ResolveConstraintFrames
[move_group-7] [INFO] [1661361821.651083029] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-7] [INFO] [1661361821.651125629] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[move_group-7] [INFO] [1661361821.651150329] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-7] [INFO] [1661361821.651312430] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[component_container-5] [INFO] [1661361821.650891129] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[... cut ...]
[component_container-5] [INFO] [1661361853.711672428] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'arm'
[component_container-5] [INFO] [1661361853.846691144] [local_planner_component]: The local planner has been stuck for several iterations. Aborting.
[component_container-5] [INFO] [1661361853.846991845] [global_planner_component]: Received global planning goal request
[component_container-5] [WARN] [1661361853.847184145] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'arm' using planner 'ompl'. Will use defaults instead.

```

</details>

sjahr, 2022-08-29:
Hi @marc-wittwer, I think your error might be related to 
`[component_container-5] [WARN] [1661361852.576617616] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'arm' using planner 'ompl'. Will use defaults instead.`

You should check that the planning group is consistently configured and exists in the panda_moveit_config. Why did you change it in `local_planner_config.yaml` from `panda_arm` to `arm`? Not sure, which srdf file you're using but the one in panda_moveit_config does not has a group called `arm`

marc-wittwer, 2022-08-29:
Thanks for your reply. @sjahr 

I just got the robot moving today.
The solution topic of the local_planner had to be changed to the following in the `local_planner.yaml.` (which is specific to my setup): 
`local_solution_topic: "/joint_trajectory_controller/joint_trajectory"`

This allows moving the robot. :tada: 

However, moveit still complains about the planning configuration. 
```
[component_container-5] [WARN] [1661764326.053499208] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'arm' using planner 'ompl'. Will use defaults instead.
```
I'm using the following SRDF file:
https://github.com/AndrejOrsula/panda_ign_moveit2/blob/master/panda_moveit_config/srdf/panda.srdf
I used this repo because Andrej provided a great example repo which combines panda+moveit+gazebo (https://github.com/AndrejOrsula/ign_moveit2_examples)

Since the SRDF that I'm using has `"arm"` as a group, I thought that I have to adapt the hybrid planner config to use `arm`.
Is this not the right thinking?

Thanks for any hints.

marc-wittwer, 2022-08-29:
I just noticed that the hybrid_planner_demo prints the same warning about the planning configuration.
I guess that's not normal and I need to freshly install moveit....? @sjahr 

```
ros2 launch moveit_hybrid_planning hybrid_planning_demo.launch.py 

[component_container_mt-1] [WARN] [1661779844.751936912] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'panda_arm' using planner 'ompl'. Will use defaults instead.
[component_container_mt-1] [WARN] [1661779844.752046213] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container_mt-1] [INFO] [1661779844.752187615] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.

```

<details>

<summary>Toggle for log output</summary>

```
INFO] [launch]: All log files can be found below /home/mwittwer/.ros/log/2022-08-29-15-30-36-447879-ubuntu-4098475
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [4098487]
[INFO] [static_transform_publisher-2]: process started with pid [4098489]
[INFO] [rviz2-3]: process started with pid [4098491]
[INFO] [robot_state_publisher-4]: process started with pid [4098493]
[INFO] [hybrid_planning_demo_node-5]: process started with pid [4098495]
[INFO] [ros2_control_node-6]: process started with pid [4098497]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-7]: process started with pid [4098499]
[INFO] [ros2 run controller_manager spawner panda_joint_group_position_controller-8]: process started with pid [4098501]
[static_transform_publisher-2] [INFO] [1661779837.147287596] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'panda_link0'
[robot_state_publisher-4] Link panda_link1 had 1 children
[robot_state_publisher-4] Link panda_link2 had 1 children
[robot_state_publisher-4] Link panda_link3 had 1 children
[robot_state_publisher-4] Link panda_link4 had 1 children
[robot_state_publisher-4] Link panda_link5 had 1 children
[robot_state_publisher-4] Link panda_link6 had 1 children
[robot_state_publisher-4] Link panda_link7 had 1 children
[robot_state_publisher-4] Link panda_link8 had 1 children
[robot_state_publisher-4] Link panda_hand had 2 children
[robot_state_publisher-4] Link panda_leftfinger had 0 children
[robot_state_publisher-4] Link panda_rightfinger had 0 children
[robot_state_publisher-4] [INFO] [1661779837.155772065] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-4] [INFO] [1661779837.155902766] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-4] [INFO] [1661779837.155934367] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-4] [INFO] [1661779837.155962467] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-4] [INFO] [1661779837.155990567] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-4] [INFO] [1661779837.156018467] [robot_state_publisher]: got segment panda_link3
[robot_state_publisher-4] [INFO] [1661779837.156046668] [robot_state_publisher]: got segment panda_link4
[robot_state_publisher-4] [INFO] [1661779837.156299370] [robot_state_publisher]: got segment panda_link5
[robot_state_publisher-4] [INFO] [1661779837.156334470] [robot_state_publisher]: got segment panda_link6
[robot_state_publisher-4] [INFO] [1661779837.156366170] [robot_state_publisher]: got segment panda_link7
[robot_state_publisher-4] [INFO] [1661779837.156394871] [robot_state_publisher]: got segment panda_link8
[robot_state_publisher-4] [INFO] [1661779837.156423070] [robot_state_publisher]: got segment panda_rightfinger
[rviz2-3] WARNING: CPU random generator seem to be failing, disable hardware random number generation
[rviz2-3] WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
[ros2_control_node-6] [INFO] [1661779837.203526255] [controller_manager]: update rate is 500 Hz
[ros2_control_node-6] [INFO] [1661779837.204104060] [controller_manager]: RT kernel is recommended for better performance
[component_container_mt-1] [INFO] [1661779837.269111789] [hybrid_planning_container]: Load Library: /home/mwittwer/ws_moveit2/install/moveit_hybrid_planning/lib/libmoveit_global_planner_component.so.2.3.2
[component_container_mt-1] [INFO] [1661779837.271331507] [hybrid_planning_container]: Found class: rclcpp_components::NodeFactoryTemplate<moveit::hybrid_planning::GlobalPlannerComponent>
[component_container_mt-1] [INFO] [1661779837.271373107] [hybrid_planning_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<moveit::hybrid_planning::GlobalPlannerComponent>
[component_container_mt-1] [INFO] [1661779837.372263530] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00502074 seconds
[component_container_mt-1] [INFO] [1661779837.373005136] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[component_container_mt-1] Link panda_link1 had 1 children
[component_container_mt-1] Link panda_link2 had 1 children
[component_container_mt-1] Link panda_link3 had 1 children
[component_container_mt-1] Link panda_link4 had 1 children
[component_container_mt-1] Link panda_link5 had 1 children
[component_container_mt-1] Link panda_link6 had 1 children
[component_container_mt-1] Link panda_link7 had 1 children
[... cut ...]
[component_container_mt-1] [INFO] [1661779844.760958886] [local_planner_component]: Collision ahead, hold current position
[component_container_mt-1] [INFO] [1661779844.766516531] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'panda_arm'
[hybrid_planning_demo_node-5] [WARN] [1661779844.766887334] [moveit_planning_scene.planning_scene]: Tried to remove world object 'box1', but it does not exist in this scene.
[hybrid_planning_demo_node-5] [INFO] [1661779845.211702638] [test_hybrid_planning_client]: Hybrid planning goal succeeded

```

</details>

marc-wittwer, 2022-08-29:
I verified the output in a docker container. The error still remains.
```
docker run -it -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix moveit/moveit2:rolling-source
root@0458b44b162e:~/ws_moveit# ls
build  install  log  src
root@0458b44b162e:~/ws_moveit# . install/setup.bash 
root@0458b44b162e:~/ws_moveit# ros2 launch moveit_hybrid_planning hybrid_planning_demo.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-08-29-14-01-40-775715-0458b44b162e-68
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [81]
[INFO] [static_transform_publisher-2]: process started with pid [83]
[INFO] [rviz2-3]: process started with pid [85]
[INFO] [robot_state_publisher-4]: process started with pid [87]
[INFO] [ros2_control_node-5]: process started with pid [89]
[INFO] [spawner-6]: process started with pid [91]
[INFO] [spawner-7]: process started with pid [93]
[INFO] [hybrid_planning_demo_node-8]: process started with pid [95]
[static_transform_publisher-2] [WARN] [1661781701.541573925] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-2] [INFO] [1661781701.622404427] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'panda_link0'
[ros2_control_node-5] [INFO] [1661781701.635917594] [resource_manager]: Loading hardware 'PandaFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.638748208] [resource_manager]: Initialize hardware 'PandaFakeSystem' 
[ros2_control_node-5] [WARN] [1661781701.638841609] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1661781701.638874709] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'
[ros2_control_node-5] [INFO] [1661781701.638942609] [resource_manager]: Loading hardware 'PandaHandFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.638978710] [resource_manager]: Initialize hardware 'PandaHandFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.639015910] [resource_manager]: Successful initialization of hardware 'PandaHandFakeSystem'
[ros2_control_node-5] [INFO] [1661781701.639085410] [resource_manager]: 'configure' hardware 'PandaFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.639111210] [resource_manager]: Successful 'configure' of hardware 'PandaFakeSystem'
[ros2_control_node-5] [INFO] [1661781701.639135710] [resource_manager]: 'activate' hardware 'PandaFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.639157711] [resource_manager]: Successful 'activate' of hardware 'PandaFakeSystem'
[ros2_control_node-5] [INFO] [1661781701.639181011] [resource_manager]: 'configure' hardware 'PandaHandFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.639205811] [resource_manager]: Successful 'configure' of hardware 'PandaHandFakeSystem'
[ros2_control_node-5] [INFO] [1661781701.639230911] [resource_manager]: 'activate' hardware 'PandaHandFakeSystem' 
[ros2_control_node-5] [INFO] [1661781701.639255411] [resource_manager]: Successful 'activate' of hardware 'PandaHandFakeSystem'
[component_container_mt-1] [INFO] [1661781701.656587597] [hybrid_planning_container]: Load Library: /root/ws_moveit/install/moveit_hybrid_planning/lib/libmoveit_global_planner_component.so.2.5.3
[component_container_mt-1] [INFO] [1661781701.659869313] [hybrid_planning_container]: Found class: rclcpp_components::NodeFactoryTemplate<moveit::hybrid_planning::GlobalPlannerComponent>
[component_container_mt-1] [INFO] [1661781701.659988414] [hybrid_planning_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<moveit::hybrid_planning::GlobalPlannerComponent>
[ros2_control_node-5] [INFO] [1661781701.671909774] [controller_manager]: update rate is 500 Hz
[ros2_control_node-5] [INFO] [1661781701.672465177] [controller_manager]: RT kernel is recommended for better performance
[rviz2-3] WARNING: CPU random generator seem to be failing, disabling hardware random number generation
[rviz2-3] WARNING: RDRND generated: 0xffffffff 0xffffffff 0xffffffff 0xffffffff
[robot_state_publisher-4] [INFO] [1661781701.684207635] [kdl_parser]: Link panda_link1 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685609442] [kdl_parser]: Link panda_link2 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685691642] [kdl_parser]: Link panda_link3 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685730042] [kdl_parser]: Link panda_link4 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685759742] [kdl_parser]: Link panda_link5 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685786042] [kdl_parser]: Link panda_link6 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685813843] [kdl_parser]: Link panda_link7 had 1 children.
[robot_state_publisher-4] [INFO] [1661781701.685840943] [kdl_parser]: Link panda_link8 had 1 children.
[... cut ...]
[component_container_mt-1] [WARN] [1661781709.050602480] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[hybrid_planning_demo_node-8] [INFO] [1661781709.050792380] [test_hybrid_planning_client]: Local goal accepted by server
[component_container_mt-1] [INFO] [1661781709.061008531] [local_planner_component]: The local planner is solving...
[hybrid_planning_demo_node-8] [INFO] [1661781709.511651872] [test_hybrid_planning_client]: Hybrid planning goal succeeded

```

marc-wittwer, 2022-08-29:
This is the behaviour I get when using the hybrid planner. It stutters a lot.
What could be the cause of this?

https://user-images.githubusercontent.com/71826111/187243715-cd768924-4d7b-47e5-a0b2-16fa1e0fa926.mp4

sjahr, 2022-08-30:
> I just noticed that the hybrid_planner_demo prints the same warning about the planning configuration. I guess that's not normal and I need to freshly install moveit....? @sjahr
> 
> ```
> ros2 launch moveit_hybrid_planning hybrid_planning_demo.launch.py 
> 
> 
> [component_container_mt-1] [WARN] [1661779844.751936912] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'panda_arm' using planner 'ompl'. Will use defaults instead.
> [component_container_mt-1] [WARN] [1661779844.752046213] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
> [component_container_mt-1] [INFO] [1661779844.752187615] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
> ```
> 
> Toggle for log output

I think you can ignore this warning for now except you want to use a non-default OMPL config. Regarding your second video, it is hard to tell what is happening just based on the video. The default logic plugin replans when the global solution is invalidated  for example when a new collision object appears in the planning scene

marc-wittwer, 2022-08-30:
Lowering the local planner frequency improved the stuttering:
`local_planning_frequency: 50.0`

marc-wittwer, 2022-08-30:
Sometimes the hybrid planner crashes after executing on a goal:
```
[component_container-5] [INFO] [1661886665.659271714] [hybrid_planning_manager]: Received goal request
[component_container-5] [INFO] [1661886665.659650915] [global_planner_component]: Received global planning goal request
[component_container-5] [WARN] [1661886665.659884616] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'arm' using planner 'ompl'. Will use defaults instead.
[component_container-5] [WARN] [1661886665.659987517] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container-5] [INFO] [1661886665.660153718] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container-5] [INFO] [1661886665.712393457] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'arm'
[component_container-5] [INFO] [1661886665.716054774] [local_planner_component]: Received local planning goal request
[component_container-5] [INFO] [1661886665.733745954] [local_planner_component]: The local planner is solving...
[component_container-5] [INFO] [1661886667.867061011] [local_planner_component]: The local planner is solving...
[component_container-5] [INFO] [1661886669.949992536] [local_planner_component]: The local planner is solving...
[component_container-5] [INFO] [1661886670.666867613] [hybrid_planning_manager]: Received goal request
[component_container-5] [INFO] [1661886670.667303815] [global_planner_component]: Received global planning goal request
[component_container-5] [WARN] [1661886670.667617917] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'arm' using planner 'ompl'. Will use defaults instead.
[component_container-5] [WARN] [1661886670.667739518] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container-5] [INFO] [1661886670.667966719] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container-5] [INFO] [1661886670.692076829] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'arm'
[component_container-5] [INFO] [1661886670.694709441] [local_planner_component]: Received local planning goal request
[component_container-5] [ERROR] [1661886670.695699045] [hybrid_planning_manager]: Hybrid Planning Manager failed to react to  'Local planning action canceled'
[component_container-5] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[component_container-5]   what():  goal_handle attempted invalid transition from state ABORTED with event SUCCEED, at /tmp/binarydeb/ros-galactic-rcl-action-3.1.3/src/rcl_action/goal_handle.c:95
[ERROR] [component_container-5]: process has died [pid 2035456, exit code -6, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container --ros-args -r __node:=hybrid_planning_container -r __ns:=/'].

```

sjahr, 2022-10-05:
Hi @marc-wittwer, it took me so long to respond. Does this issue still exist? If you fixed something in MoveIt, it would be great if you could open a PR with your fix. If it still exists, can you open a separate issue, so we can track and fix it eventually? I'll close this issue for now, since the original question is solved.

closed 2022-10-05 by sjahr

skaeringur97, 2022-11-07:
@marc-wittwer did you find a solution to this? I'm having similar problems

marc-wittwer, 2022-11-07:
@sjahr @skaeringur97 Unfortunately I was not able to get it working using the built-in hybrid planner.
I somehow modelled my own hybrid framework:
[image]

I used JointJog messages to track a certain joint configuration.

marc-wittwer, 2022-11-07:
I did not use the hybrid plugin since my C++ understanding is limited and the whole MoveIt architecture is quite complex. I simply spawned 3 python nodes as seen in the illustration above.
I manually called the MoveGroup action server to plan a trajectory with the `plan_only = True` parameter and then sent JointJog messages via the `delta_joint_cmds` topic to match this topic.

@sjahr I just finished my master thesis project. It would not have been possible without your hybrid_planning framework contribution. since it got me on the right track. Thank you for that. Here is the final video of my project if you are interested:
https://youtu.be/jY5_l84q21U

sjahr, 2022-11-08:
Congratulations! Glad to hear that it helped you. This video looks cool, is / will your thesis be published and/ or is your source code open source? I'd be interested to learn more about the actual implementation and architecture of your project