moveit/moveit2 #1548: `slots=True` decorator on MoveItConfigs breaks non-Jammy deployments.

danzimmerman, 2022-09-02:
### Description

#1308 introduces a Python 3.10 `@dataclass(slots=True)` decorator on the `MoveItConfigs` class which breaks MSA2 MoveIt Configs and some newer examples (MoveIt Servo [Pose Tracking](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/launch/pose_tracking_example.launch.py#L11), for one) on [platforms other than native installation on 22.04 Jammy Jellyfish.](https://github.com/ros-planning/moveit2/pull/1308#issuecomment-1145172779)

This includes Windows 10 Humble/Rolling, which seems like it'll be using Python 3.8.3: 

https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027 

I'm also regularly using MoveIt2 on both Ubuntu Focal and Windows 11 via [Robostack](https://robostack.github.io/), which is also affected. The Robostack project may be on a shorter timeline to bump the Python version to 3.10, but currently MSA and anything using `MoveItConfigs` is broken out of the box:

<details>

```python
>>> from moveit_configs_utils import MoveItConfigsBuilder
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/moveit_configs_utils/__init__.py", line 1, in <module>
    from .moveit_configs_builder import MoveItConfigsBuilder, MoveItConfigs
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 87, in <module>
    @dataclass(slots=True)
TypeError: dataclass() got an unexpected keyword argument 'slots'
>>> from moveit_configs_utils import MoveItConfigsBuilder
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/moveit_configs_utils/__init__.py", line 1, in <module>
    from .moveit_configs_builder import MoveItConfigsBuilder, MoveItConfigs
  File "/home/<user>/.local/opt/mambaforge/envs/humble/lib/python3.9/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 87, in <module>
    @dataclass(slots=True)
TypeError: dataclass() got an unexpected keyword argument 'slots'
```

</details>

It's an easy enough patch for non-Jammy platforms at this point, but I wanted to open this to discuss the intended support for platforms that will use older Python versions for a while, given that [`moveit_configs_utils`](https://github.com/ros-planning/moveit2/tree/main/moveit_configs_utils/moveit_configs_utils) seems to be the future of MSA and MoveIt launch examples and will be hard to avoid.

jdlangs, 2023-03-02:
I just hit this as well. Everything still works fine without the `slots` keyword so it does feel like unnecessary breakage.

danzimmerman, 2023-03-02:
@jdlangs What platform/install are you using? 

Updating my experience since the original post: 

The Robostack team migrated their Humble build from Python 3.9 to 3.10 so my latest fresh Robostack install is working. My test of this is on Windows, but I think Humble is at Python 3.10 across platforms now.

Looks like the Robostack instructions still have Noetic and Galactic on Python 3.9:

https://robostack.github.io/GettingStarted.html

Windows native/Chocolatey binary install still assumes Python 3.8.3.

[https://docs.ros.org/en/rolling/Installation/Windows-Install-Binary.html#install-python
](https://docs.ros.org/en/rolling/Installation/Windows-Install-Binary.html#install-python
)

jdlangs, 2023-03-02:
I'm building ROS2 Humble and MoveIt2 from source on Ubuntu 20.04

JafarAbdi, 2023-04-12:
Foxy's EOL date is in less than a month, and Ubuntu 20.04 will no longer be a tier 1 platform for any of the releases.

closed 2023-04-12 by JafarAbdi