moveit/moveit2 #1554: Moveit2 reports that the execution has finished but the real robot has not reached the goal pose

Juancams, 2022-09-06:
### Environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source, foxy branch

### Description
I have my own robot arm, in which I have created all the part of ros2_control (hardware_interface) so that it can read the values of the motors (speed, position) and added the plugin to the urdf. I have also generated the moveit_config files from the moveit_setup_assistant and move_group launchers following the ones used by the panda arm.

The problem is that the arm stops without having finished its trajectory. You can see how in the controller topic "/robot_controller/state" the desired position and the current position are different, so it is not using a fake for the position. When I execute the trajectory again, it stops again without finishing it and each time with less speed and duration.

As I mentioned, I don't know if it's a problem with moveit or with ros2_control, since apparently it's the controller that tells me that it has arrived.

```
[moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to robot_controller
[robot_controller]: Received new action goal
[robot_controller]: Accepted new action goal
[moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: robot_controller started execution
[moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[move_group_interface]: Plan and Execute request accepted
[robot_controller]: Goal reached, success!
[moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller robot_controller successfully finished
[moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
```

henningkayser, 2023-01-26:
@Juancams what types of joints do you use for your hardware and how far off is the current state from the target? Most likely, it's really the ros2_controller hitting some conditions that render the trajectory as successful, MoveIt is just receiving the action result. Without more data (target trajectory, plot of hardware states) it's hard to tell what's happening. However, you can try changing some of the controller parameters to see if they have any effect https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/userdoc.html#details-about-parameters. I would try reducing the `stopped_velocity_tolerance` and separately see if the robot is following the trajectory better when setting `open_loop_control: true`.

tylerjw, 2023-03-10:
Have you tried upgrading to Humble? Foxy is now deprecated.

closed 2023-03-10 by tylerjw