moveit/moveit2 #1558: Calculate set of IK solutions for a given pose

tylerjw, 2022-09-07:
Given a pose, it would be nice if we could calculate a set of IK solutions for that pose. This would be useful for MTC tasks.

With the current interface, the approach is to sample IK many times and keep the results.

This could be useful in cases where the robot arm has >6 dof or the IK problem is underconstrained (allow approximate or other similar relaxing of the constraints).

Robotawi, 2022-09-16:
Hi @tylerjw. 
I want to work on this issue. I would appreciate your guidance about where to start and will try to proceed from there.

SalahSoliman, 2022-09-23:
@tylerjw 
If I'm not mistaken:
`constructGoal()` calls -> `ConstrainedGoalSampler` calls -> sampleUsingConstraintSampler calls -> `sample` (in `default_constaint_sampler` which calls -> `sample_helper` which calls -> `callIK` which eventually calls -> `searchPositionIK`

It's quite complicated, what are your thoughts about tackling this issue?

tylerjw, 2022-09-23:
I honestly don't know. I think the whole interface for IK is way too convoluted. This is an example of a valid use case of IK solvers that is not supported by the current architecture.

SalahSoliman, 2022-09-24:
@tylerjw are you considering refactoring? 
Please let me know, I would be interested in working on this.

SalahSoliman, 2022-09-28:
@tylerjw would you like to setup a meeting to discuss possible architectural modifications?

tylerjw, 2022-09-28:
Not yet, I think this is part of a much larger discussion we are having internally. If you have any ideas for how you might accomplish this please write about them here.

closed 2025-02-19 by github-actions[bot]