moveit/moveit2 #1565: Servo control in gazebo

AltinayGrass, 2022-09-08:
If I put the 6 DOF robot urdf file with real weights and inertia values  of links to the Gazebo and then servo control on, is it possible to see realistic values ​​in the effort controller or speed controller?

AndyZe, 2022-09-08:
I would think the effort controller definitely won't work. @vatanaksoytezer is Gazebo compatible with a JointGroupVelocity controller?

vatanaksoytezer, 2022-09-08:
I think position, velocity and effort should be working as of now in both Gazebo Classic and Ignition Gazebo (Gazebo Sim), see https://github.com/ros-controls/gz_ros2_control for reference.

For the realistic part of the question: I don't think it would be quite realistic in terms of how close the values are to the real life, but can give you a nice hint of what to expect in general and might help identify low and high effort works.

closed 2022-09-09 by vatanaksoytezer