moveit/moveit2 #1566: return to zero speed in servo_jog mode

AltinayGrass, 2022-09-09:
### Description

After canceling to send joint command, velocity of robot_states does not return to zero, but after cancelling cartesian command, velocities of robot_states is returning to zero.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? ROS binary / moveit Source
* If binary, which release version? latest
* If source, which branch? main

### Expected behaviour
if there is no vel command actuators should smoothly return to zero speed

### Actual behaviour

[Screencast from 09-09-2022 09:22:03 AM.webm](https://user-images.githubusercontent.com/97592357/189284908-bee9abde-a62d-4114-80c6-b58235456e1c.webm)

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

vatanaksoytezer, 2022-09-09:
I think what you desire makes sense to me but I am unsure if this should be on MoveIt side or the robot side, tagging @AndyZe for more opinions.

AndyZe, 2022-09-09:
Nice GUI! I think it's a good idea to fix this. Can you make a PR?

AltinayGrass, 2022-09-09:
maybe instead of writing command only for the axes that will move, it is necessary to write values ​​for all axes even if their values ​​are zero.

closed 2022-09-09 by AltinayGrass