moveit/moveit2 #1570: Failed to find exported target names in   'ws_moveit2/install/moveit_core/share/moveit_core/cmake/export_moveit_coreExport.cmake'

atuldivekar, 2022-09-14:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* ROS binary ros-humble-desktop
* MoveIt2 build from source Humble branch

### Steps to reproduce
Install ROS2 as described at 
https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

source /opt/ros/humble/setup.bash

Install MoveIt2 as described at
https://moveit.ros.org/install-moveit2/source/

export COLCON_WS=~/ws_moveit2/
mkdir -p $COLCON_WS/src
cd $COLCON_WS/src

echo  $ROS_DISTRO   
humble

git clone https://github.com/ros-planning/moveit2.git -b $ROS_DISTRO
for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

### Expected behaviour
MoveIt should build successfully

### Actual behaviour
Fails as shown:

[0.923s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_interface' is in: /home/adivekar/ws_moveit2/install/controller_interface, /opt/ros/humble
	'controller_manager' is in: /home/adivekar/ws_moveit2/install/controller_manager
	'hardware_interface' is in: /home/adivekar/ws_moveit2/install/hardware_interface, /opt/ros/humble
	'ros2_control_test_assets' is in: /home/adivekar/ws_moveit2/install/ros2_control_test_assets
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> ros2_control_test_assets
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Starting >>> joint_limits
Starting >>> rviz_visual_tools
Finished <<< moveit_common [0.89s]
Starting >>> rosparam_shortcuts
Finished <<< moveit_resources_panda_description [1.00s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [1.20s]
Finished <<< ros2_control_test_assets [1.16s]
Starting >>> hardware_interface
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_prbt_support [1.20s]
Finished <<< moveit_resources_pr2_description [1.29s]
--- stderr: moveit_configs_utils
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [1.54s]
Finished <<< moveit_resources_panda_moveit_config [0.69s]
Starting >>> moveit_core
Finished <<< joint_limits [1.59s]
Finished <<< moveit_resources_fanuc_moveit_config [0.65s]
Starting >>> moveit_resources
Finished <<< moveit_resources [0.35s]
Finished <<< controller_manager_msgs [2.75s]
Finished <<< moveit_core [9.94s]
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< rviz_marker_tools [12.1s]
--- stderr: moveit_resources_prbt_ikfast_manipulator_plugin
CMake Error at /home/adivekar/ws_moveit2/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/adivekar/ws_moveit2/install/moveit_core/share/moveit_core/cmake/export_moveit_coreExport.cmake'
Call Stack (most recent call first):
  /home/adivekar/ws_moveit2/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:41 (include)
  CMakeLists.txt:22 (find_package)

---
Failed   <<< moveit_resources_prbt_ikfast_manipulator_plugin [1.11s, exited with code 1]
Aborted  <<< moveit_simple_controller_manager [1.23s]
Aborted  <<< moveit_ros_occupancy_map_monitor [1.38s]
Aborted  <<< chomp_motion_planner [1.10s]
Aborted  <<< pilz_industrial_motion_planner_testutils [2.56s]
Aborted  <<< hardware_interface [18.7s]
Aborted  <<< moveit_task_constructor_msgs [21.7s]
Aborted  <<< rosparam_shortcuts [21.5s]
Aborted  <<< rviz_visual_tools [40.3s]

Summary: 14 packages finished [41.4s]
  1 package failed: moveit_resources_prbt_ikfast_manipulator_plugin
  8 packages aborted: chomp_motion_planner hardware_interface moveit_ros_occupancy_map_monitor moveit_simple_controller_manager moveit_task_constructor_msgs pilz_industrial_motion_planner_testutils rosparam_shortcuts rviz_visual_tools
  6 packages had stderr output: chomp_motion_planner moveit_configs_utils moveit_resources_prbt_ikfast_manipulator_plugin moveit_ros_occupancy_map_monitor moveit_simple_controller_manager pilz_industrial_motion_planner_testutils
  40 packages not processed

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

gaokai15, 2022-09-24:
I got a similar error when building "moveit2_tutorials". The command I run is 
`colcon build --mixin release`
The error I got:
```
Starting >>> moveit2_tutorials
--- stderr: moveit2_tutorials                          
CMake Error at /home/kai/ws_moveit2/install/moveit_visual_tools/share/moveit_visual_tools/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/kai/ws_moveit2/install/moveit_visual_tools/share/moveit_visual_tools/cmake/export_moveit_visual_toolsExport.cmake'
Call Stack (most recent call first):
  /home/kai/ws_moveit2/install/moveit_visual_tools/share/moveit_visual_tools/cmake/moveit_visual_toolsConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

---
Failed   <<< moveit2_tutorials [9.43s, exited with code 1]
```

GabrielReVa, 2022-10-05:
I get the same error but with my own interfaces. The error is:
CMake Error at C:/QuixMind/QuixSmart/ros2_ws/install/inventory_interfaces/share/inventory_interfaces/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  'C:/QuixMind/QuixSmart/ros2_ws/install/inventory_interfaces/share/inventory_interfaces/cmake/inventory_interfaces__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  C:/QuixMind/QuixSmart/ros2_ws/install/inventory_interfaces/share/inventory_interfaces/cmake/inventory_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:16 (find_package)
---
Failed   <<< inventory_storage

AbdoullahNdoye, 2022-10-07:
I got the same error using colcon build with realsense2 package : 

CMake Error at /home/ndoyea/ros2_ws/install/realsense2_camera_msgs/share/realsense2_camera_msgs/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/ndoyea/ros2_ws/install/realsense2_camera_msgs/share/realsense2_camera_msgs/cmake/export_realsense2_camera_msgs__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/ndoyea/ros2_ws/install/realsense2_camera_msgs/share/realsense2_camera_msgs/cmake/realsense2_camera_msgsConfig.cmake:41 (include)
  CMakeLists.txt:96 (find_package)

AbdoullahNdoye, 2022-10-08:
This helped me solve it for me:  https://github.com/ament/ament_cmake/pull/395/commits/11c44dbd646846a2e8ca912155f0704e9b0c3c57. 
I changed two lines in the file "ament_cmake_export_targets-extras.cmake.in"  located in  /opt/ros/humble/share/ament_cmake_export_targets/cmake.

henningkayser, 2023-01-25:
@atuldivekar has your problem been solved with the ament_cmake fix?

atuldivekar, 2023-01-25:
I have not worked with MoveIt recently. Will comment here if I need help. Otherwise the issue can be closed.

closed 2023-01-25 by henningkayser