moveit/moveit2 #1576: Eigen issue when compiling moveit2

GiladRobee, 2022-09-20:
### Description

when compiling the repo several errors occure related to Eigen lib.
those can be fixed manualy but i feel like something is missing

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? main

### Steps to reproduce
following this instructions: https://moveit.ros.org/install-moveit2/source/
when i compile i get the following error:
```
fatal error: Eigen/Eigen: No such file or directory
   43 | #include <Eigen/Eigen>
```
i have Eigen installed:
```
dpkg -l | grep libeigen3-dev
ii  libeigen3-dev                                     3.4.0-2ubuntu2                          all          lightweight C++ template library for linear algebra
```
quick google tells me to change the header to # include <eigen3/Eigen/Core>
but then i have to change all apearnces of it in the entire project.
after that i have to change other Eigen related headers in order for it to compile.

i was using a script to change all headers at once:
the main part is: 
```
change_dic["# include <Eigen/Core>"] = "# include <eigen3/Eigen/Core>"
change_dic["#include <Eigen/Eigen>"] = "#include <Eigen/Dense>"
change_dic["#include <eigen3/Eigen/Eigen>"] = "#include <Eigen/Dense>"
change_dic["#include \"eigen3/Eigen/Eigen\""] = "#include <Eigen/Dense>"
```

SalahSoliman, 2022-09-24:
You can link it as follows: 
```
cd ~/usr/include
sudo ln -sf eigen3/Eigen Eigen
sudo ln -sf eigen3/unsupported unsupported
```

henningkayser, 2023-01-25:
@GiladRobee does this still happen to you with the current main branch? Also, what package is failing for you? It would be great if you could post a full build log and a workspace print so that we know what's going on. This issue hasn't occurred on any of our CI build jobs for any distro, afaik.

GiladRobee, 2023-01-26:
I would not be able to reproduce it since I have moved to the binary build.
This can be closed, and thank you for your answers.
Best regards.

closed 2023-01-26 by GiladRobee