moveit/moveit2 #1584: Is there a way to add complex geometry as a Planning Scene object?

sampreets3, 2022-09-28:
Hi, 
I am using MoveIt2 (Foxy) for my project. I see that I am able to add collision objects of the type `moveit_msgs::msg::AttachedCollisionObject` to the PlanningScene and the robot shall try to plan around the objects. I am interested to know if there exists a method to add more complex geometry _(an STL mesh maybe)_ to the planning scene and manipulate its position and orientation dynamically. 

I know that there used to be a [tutorial](https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html) on adding collision objects from a `PointCloud` object in MoveIt1, it would be great to know if that feature has been already ported to MoveIt2, more importantly in the Foxy Release. 

If not, I would like to hear your advice on how to accomplish a similar result on Foxy.

Thanks in advance for the help !

AndyZe, 2022-09-28:
Yep, that's definitely possible. Here's an example. It's a little old but this part of the API hasn't changed much at all.

https://answers.ros.org/question/209994/publishing-a-collision-object-to-moveit/

Or a tutorial PR that pretty much does the same thing, except with a box, not a mesh. Changing it to a mesh should be pretty easy:

https://github.com/ros-planning/moveit2_tutorials/pull/492/files#diff-9ed90fb827f20edb9797d631678346dc93268dcf61f2bc6d5de287628d063d7c

sampreets3, 2022-09-28:
Hey @AndyZe, 

Thanks for the quick response ! I'm going to try out both approaches and will update my answer if I succeed.

sjahr, 2022-12-09:
@sampreets3 I hope you`ve been successful. I`ll close this for now. Feel free to reopen if the answer did not work

closed 2022-12-09 by sjahr

sampreets3, 2022-12-12:
Hey @sjahr , yeah sorry for not keeping it up to date. I have indeed solved my problem.

newtonjeri, 2024-05-04:
> Hey @sjahr , yeah sorry for not keeping it up to date. I have indeed solved my problem.
> 

Hi @sampreets3, I have been able to load meshes to the planning scene however they are not collision enabled, kindly if you can, share your solution. Thanks