moveit/moveit2 #1594: Moveit build error - Ubuntu 22.04

lonebots, 2022-10-02:
### Description

The terminal crashed when trying to build Moveit using the source. All the dependencies where correctly installed. and further details about the crash and logs are provided below.

### Your environment
* ROS Distro: Humble (ros2)
* OS Version: Ubuntu 22.04
* Source build, (binary not available)
* Branch : stable and unstable (the issue prevails)

### Steps to reproduce
Follow the source build instruction provided [here.](https://moveit.ros.org/install-moveit2/source/) The error occurred after even after successfully installing all the dependencies given in the installation (exception used a different version of clang-format, because clang-format-10 is not available anymore). It occurs when running the build command. 
> cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

### Expected behaviour
The expected behaviour is to successfully complete the build with minimum warning about deprecation. 

### Actual behaviour
The source build exited with 2 errors.
> --- stderr: moveit_ros_robot_interaction
/usr/bin/ld: libmoveit_robot_interaction.so.2.5.3: undefined reference to `moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, std::function<bool (moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*)> const&, kinematics::KinematicsQueryOptions const&, std::function<double (geometry_msgs::msg::Pose_<std::allocator<void> > const&, moveit::core::RobotState const&, moveit::core::JointModelGroup*, std::vector<double, std::allocator<double> > const&)>)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/locked_robot_state_test.dir/build.make:329: locked_robot_state_test] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:185: CMakeFiles/locked_robot_state_test.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

### Backtrace or Console output

> Starting >>> moveit_ros_perception
> [Processing: moveit_kinematics, moveit_planners_ompl, moveit_ros_perception, moveit_ros_robot_interaction, moveit_ros_warehouse]
[Processing: moveit_kinematics, moveit_planners_ompl, moveit_ros_perception, moveit_ros_robot_interaction, moveit_ros_warehouse]
--- stderr: moveit_ros_robot_interaction
/usr/bin/ld: libmoveit_robot_interaction.so.2.5.3: undefined reference to `moveit::core::RobotState::setFromIK(moveit::core::JointModelGroup const*, geometry_msgs::msg::Pose_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, std::function<bool (moveit::core::RobotState*, moveit::core::JointModelGroup const*, double const*)> const&, kinematics::KinematicsQueryOptions const&, std::function<double (geometry_msgs::msg::Pose_<std::allocator<void> > const&, moveit::core::RobotState const&, moveit::core::JointModelGroup*, std::vector<double, std::allocator<double> > const&)>)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/locked_robot_state_test.dir/build.make:329: locked_robot_state_test] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:185: CMakeFiles/locked_robot_state_test.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
> Failed   <<< moveit_ros_robot_interaction [1min 9s, exited with code 2]
Aborted  <<< moveit_kinematics [1min 17s]
Aborted  <<< moveit_ros_perception [1min 30s]
Aborted  <<< moveit_ros_warehouse [2min 21s]
Aborted  <<< moveit_planners_ompl [2min 48s]

> Summary: 29 packages finished [10min 47s]
  1 package failed: moveit_ros_robot_interaction
  4 packages aborted: moveit_kinematics moveit_planners_ompl moveit_ros_perception moveit_ros_warehouse
  4 packages had stderr output: controller_manager moveit_configs_utils moveit_ros_robot_interaction ros2controlcli
  19 packages not processed

Gleis1234, 2022-10-02:
got the same issue with binary install from repo ros-humble-moveit.

Nils-ChristianIseke, 2022-10-03:
I had the same issue, two different approaches worked for me:

1. Replace `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release` by: 
```
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 1

```
2. If the first approach does not work you can also try to follow those [instructions ](https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html). After the command:` vcs import < moveit2_tutorials/moveit2_tutorials.repos` i had to switch to the humble branch of moveit2 otherwise the build would fail:
```
cd ~/ws_moveit2/src/moveit2
git switch humble
cd ~/ws_moveit2/src/

```
And as above instead of `colcon build --mixin release` i used:
```
colcon build --mixin release  --parallel-workers 1
```

Gleis1234, 2022-10-03:
The second way helped. I built from ~/ws_movit2. 
Result 
/home/guenther/ws_moveit2/src/moveit2/moveit_setup_assistant/moveit_setup_srdf_plugins/include/moveit_setup_srdf_plugins/rotated_header_view.hpp:0: Note: No relevant classes found. No output generated.
---
Summary: 63 packages finished [37min 36s]
  4 packages had stderr output: controller_manager moveit_configs_utils moveit_setup_srdf_plugins ros2controlcli
I am not shure about the error in moveit_setup_srdf_plugins/include/moveit_setup_srdf_plugins/rotated_header_view.hpp .

AndyZe, 2022-10-03:
If `--parallel-workers 1` helped, it probably means you were short on RAM.

I think you could use more threads than `1`. You could try `--parallel-workers 4` or so... Probably the full 8 or 12 threads is too much.

Gleis1234, 2022-10-03:
I'm using --parallel-workers 2. That's ok. The main reason seams to be the git switch humble.

sjahr, 2022-10-05:
Another solution (weaker) solution is to disable building multiple packages in parallel with `--executor sequential`
I'll close this issue since it seems to be solved :+1:

closed 2022-10-05 by sjahr

sea-bass, 2022-10-17:
So running this build without parallelism happens to fix things, but I think I've fixed the root cause here: https://github.com/ros-planning/moveit2/pull/1617