moveit/moveit2 #1597: move_group segfault on shutdown

tylerjw, 2022-10-03:
### Description

```
 [move_group-1] #15   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x558f087aa794, in _start
    [move_group-1] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb8614ace3f, in __libc_start_main
    [move_group-1] #13   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb8614acd8f, in 
    [move_group-1] #12 | Source "./src/move_group.cpp", line 317, in ~shared_ptr
    [move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr.h", line 122, in ~__shared_ptr
    [move_group-1]     |   120:   */
    [move_group-1]     |   121:   template<typename _Tp>
    [move_group-1]     | > 122:     class shared_ptr : public __shared_ptr<_Tp>
    [move_group-1]     |   123:     {
    [move_group-1]     |   124:       template<typename... _Args>
    [move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 1154, in ~__shared_count
    [move_group-1]     |  1152:       __shared_ptr(const __shared_ptr&) noexcept = default;
    [move_group-1]     |  1153:       __shared_ptr& operator=(const __shared_ptr&) noexcept = default;
    [move_group-1]     | >1154:       ~__shared_ptr() = default;
    [move_group-1]     |  1155: 
    [move_group-1]     |  1156:       template<typename _Yp, typename = _Compatible<_Yp>>
    [move_group-1]       Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 705, in main [0x558f087a96cb]
    [move_group-1]         702:       ~__shared_count() noexcept
    [move_group-1]         703:       {
    [move_group-1]         704: 	if (_M_pi != nullptr)
    [move_group-1]       > 705: 	  _M_pi->_M_release();
    [move_group-1]         706:       }
    [move_group-1]         707: 
    [move_group-1]         708:       __shared_count(const __shared_count& __r) noexcept
    [move_group-1] #11   Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 168, in _M_release [0x558f087ab779]
    [move_group-1]         165: 	if (__gnu_cxx::__exchange_and_add_dispatch(&_M_use_count, -1) == 1)
    [move_group-1]         166: 	  {
    [move_group-1]         167:             _GLIBCXX_SYNCHRONIZATION_HAPPENS_AFTER(&_M_use_count);
    [move_group-1]       > 168: 	    _M_dispose();
    [move_group-1]         169: 	    // There must be a memory barrier between dispose() and destroy()
    [move_group-1]         170: 	    // to ensure that the effects of dispose() are observed in the
    [move_group-1]     |   681: 		      _M_get_Tp_allocator());
    [move_group-1]     |   682: 	_GLIBCXX_ASAN_ANNOTATE_BEFORE_DEALLOC;
    [move_group-1]     | Source "/usr/include/c++/11/bits/alloc_traits.h", line 854, in _Destroy<std::weak_ptr<rclcpp::ClientBase>*>
    [move_group-1]     |   852: 	     allocator<_Tp>&)
    [move_group-1]     |   853:     {
    [move_group-1]     | > 854:       _Destroy(__first, __last);
    [move_group-1]     |   855:     }
    [move_group-1]     | Source "/usr/include/c++/11/bits/stl_construct.h", line 196, in __destroy<std::weak_ptr<rclcpp::ClientBase>*>
    [move_group-1]     |   194: #endif
    [move_group-1]     |   195:       std::_Destroy_aux<__has_trivial_destructor(_Value_type)>::
    [move_group-1]     | > 196: 	__destroy(__first, __last);
    [move_group-1]     |   197:     }
    [move_group-1]     | Source "/usr/include/c++/11/bits/stl_construct.h", line 163, in _Destroy<std::weak_ptr<rclcpp::ClientBase> >
    [move_group-1]     |   161: 	{
    [move_group-1]     |   162: 	  for (; __first != __last; ++__first)
    [move_group-1]     | > 163: 	    std::_Destroy(std::__addressof(*__first));
    [move_group-1]     |   164: 	}
    [move_group-1]     |   165:     };
    [move_group-1]     | Source "/usr/include/c++/11/bits/stl_construct.h", line 151, in ~weak_ptr
    [move_group-1]     |   149:       std::destroy_at(__pointer);
    [move_group-1]     |   150: #else
    [move_group-1]     | > 151:       __pointer->~_Tp();
    [move_group-1]     |   152: #endif
    [move_group-1]     |   153:     }
    [move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr.h", line 686, in ~__weak_ptr
    [move_group-1]     |   684:    */
    [move_group-1]     |   685:   template<typename _Tp>
    [move_group-1]     | > 686:     class weak_ptr : public __weak_ptr<_Tp>
    [move_group-1]     |   687:     {
    [move_group-1]     |   688:       template<typename _Arg>
    [move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 1610, in ~__weak_count
    [move_group-1]     |  1608:       __weak_ptr(const __weak_ptr&) noexcept = default;
    [move_group-1]     |  1609: 
    [move_group-1]     | >1610:       ~__weak_ptr() = default;
    [move_group-1]     |  1611: 
    [move_group-1]     |  1612:       // The "obvious" converting constructor implementation:
    [move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 798, in _M_weak_release
    [move_group-1]     |   796:       {
    [move_group-1]     |   797: 	if (_M_pi != nullptr)
    [move_group-1]     | > 798: 	  _M_pi->_M_weak_release();
    [move_group-1]     |   799:       }
    [move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 161, in _M_release
    [move_group-1]     |   160:       void
    [move_group-1]     | > 161:       _M_release() noexcept
    [move_group-1]     |   162:       {
    [move_group-1]     |   163:         // Be race-detector-friendly.  For more info see bits/c++config.
    [move_group-1]       Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 184, in _M_dispose [0x7fb861a4fb21]
    [move_group-1]         181: 						       -1) == 1)
    [move_group-1]         182:               {
    [move_group-1]         183:                 _GLIBCXX_SYNCHRONIZATION_HAPPENS_AFTER(&_M_weak_count);
    [move_group-1]       > 184: 	        _M_destroy();
    [move_group-1]         185:               }
    [move_group-1]         186: 	  }
    [move_group-1]         187:       }
    [move_group-1] Segmentation fault (Address not mapped to object [0x7fb844621c90])
Error: ROR] [component_container_mt-5]: process has died [pid 7949, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args --log-level error --ros-args -r __node:=moveit_studio_container -r __ns:=/'].
```

timonegk, 2022-10-19:
I'm facing another segmentation fault that seems to be only in version `2.5.3-1jammy.20220914.xxxxx`. The previous version (`2.5.3-1jammy.20220815.xxxxx` works fine). Maybe a rebuild of the packages could fix the issue?

```
[move_group-1] Stack trace (most recent call last):
[move_group-1] #20   Object "", at 0xffffffffffffffff, in 
[move_group-1] #19   Object "/opt/ros/rolling/lib/moveit_ros_move_group/move_group", at 0x560929291794, in 
[move_group-1] #18   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7ffa37827e3f]
[move_group-1] #17   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7ffa37827d8f]
[move_group-1] #16   Object "/opt/ros/rolling/lib/moveit_ros_move_group/move_group", at 0x560929290332, in 
[move_group-1] #15   Object "/opt/ros/rolling/lib/libmoveit_cpp.so.2.5.3", at 0x7ffa38171014, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-1] #14   Object "/opt/ros/rolling/lib/libmoveit_cpp.so.2.5.3", at 0x7ffa3816e9cb, in moveit_cpp::MoveItCpp::loadPlanningSceneMonitor(moveit_cpp::MoveItCpp::PlanningSceneMonitorOptions const&)
[move_group-1] #13   Object "/opt/ros/rolling/lib/libmoveit_planning_scene_monitor.so.2.5.3", at 0x7ffa3808fcaa, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #12   Object "/opt/ros/rolling/lib/libmoveit_planning_scene_monitor.so.2.5.3", at 0x7ffa3808fc1a, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::shared_ptr<planning_scene::PlanningScene> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #11   Object "/opt/ros/rolling/lib/libmoveit_planning_scene_monitor.so.2.5.3", at 0x7ffa3808f7ac, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::shared_ptr<planning_scene::PlanningScene> const&, std::shared_ptr<robot_model_loader::RobotModelLoader> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #10   Object "/opt/ros/rolling/lib/libtf2_ros.so", at 0x7ffa37f4790e, in tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, bool)
[move_group-1] #9    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7ffa37dc9e67, in rclcpp::executors::EventsExecutor::add_callback_group(std::shared_ptr<rclcpp::CallbackGroup>, std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, bool)
[move_group-1] #8    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7ffa37dc9d84, in rclcpp::executors::EventsExecutorEntitiesCollector::add_callback_group(std::shared_ptr<rclcpp::CallbackGroup>, std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
[move_group-1] #7    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7ffa37dc9917, in rclcpp::executors::EventsExecutorEntitiesCollector::add_callback_group(std::shared_ptr<rclcpp::CallbackGroup>, std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::map<std::weak_ptr<rclcpp::CallbackGroup>, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::owner_less<std::weak_ptr<rclcpp::CallbackGroup> >, std::allocator<std::pair<std::weak_ptr<rclcpp::CallbackGroup> const, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface> > > >&)
[move_group-1] #6    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7ffa37dc8f07, in rclcpp::executors::EventsExecutorEntitiesCollector::set_callback_group_entities_callbacks(std::shared_ptr<rclcpp::CallbackGroup>)
[move_group-1] #5    Object "/opt/ros/rolling/lib/libtf2_ros.so", at 0x7ffa37f4170e, in 
[move_group-1] #4    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7ffa37e30f33, in rclcpp::QOSEventHandlerBase::set_on_new_event_callback(void (*)(void const*, unsigned long), void const*)
[move_group-1] #3    Object "/opt/ros/rolling/lib/librmw_fastrtps_shared_cpp.so", at 0x7ffa342cb21c, in rmw_fastrtps_shared_cpp::__rmw_event_set_callback(rmw_event_s*, void (*)(void const*, unsigned long), void const*)
[move_group-1] #2    Object "/opt/ros/rolling/lib/librmw_fastrtps_shared_cpp.so", at 0x7ffa342c6f1b, in SubListener::set_on_new_event_callback(rmw_event_type_e, void const*, void (*)(void const*, unsigned long))
[move_group-1] #1    Object "/opt/ros/rolling/lib/libmoveit_planning_scene_monitor.so.2.5.3", at 0x7ffa380a1198, in void rclcpp::detail::cpp_callback_trampoline<void const*, unsigned long, void>(void const*, unsigned long)
[move_group-1] #0    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7ffa37dc35c8, in 
[move_group-1] Segmentation fault (Address not mapped to object [(nil)])
```

timonegk, 2022-10-21:
I am unfortunately no longer able to reproduce my issue, I'm getting a different error now. I will report back when I know more.

timonegk, 2023-02-07:
The issue is back, here is the stack trace:
```
[move_group-1] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[move_group-1]          at line 127 in ./src/class_loader.cpp
[move_group-1] Stack trace (most recent call last):
[move_group-1] #11   Object "", at 0xffffffffffffffff, in 
[move_group-1] #10   Object "/opt/ros/rolling/lib/moveit_ros_move_group/move_group", at 0x55c6da1697a4, in _start
[move_group-1] #9    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2d3d19be3f, in __libc_start_main
[move_group-1] #8    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2d3d19bd8f, in 
[move_group-1] #7  | Source "./src/move_group.cpp", line 317, in ~shared_ptr
[move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr.h", line 122, in ~__shared_ptr
[move_group-1]     |   120:   */
[move_group-1]     |   121:   template<typename _Tp>
[move_group-1]     | > 122:     class shared_ptr : public __shared_ptr<_Tp>
[move_group-1]     |   123:     {
[move_group-1]     |   124:       template<typename... _Args>
[move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 1154, in ~__shared_count
[move_group-1]     |  1152:       __shared_ptr(const __shared_ptr&) noexcept = default;
[move_group-1]     |  1153:       __shared_ptr& operator=(const __shared_ptr&) noexcept = default;
[move_group-1]     | >1154:       ~__shared_ptr() = default;
[move_group-1]     |  1155: 
[move_group-1]     |  1156:       template<typename _Yp, typename = _Compatible<_Yp>>
[move_group-1]       Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 705, in main [0x55c6da168711]
[move_group-1]         702:       ~__shared_count() noexcept
[move_group-1]         703:       {
[move_group-1]         704: 	if (_M_pi != nullptr)
[move_group-1]       > 705: 	  _M_pi->_M_release();
[move_group-1]         706:       }
[move_group-1]         707: 
[move_group-1]         708:       __shared_count(const __shared_count& __r) noexcept
[move_group-1] #6    Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 168, in _M_release [0x55c6da16a739]
[move_group-1]         165: 	if (__gnu_cxx::__exchange_and_add_dispatch(&_M_use_count, -1) == 1)
[move_group-1]         166: 	  {
[move_group-1]         167:             _GLIBCXX_SYNCHRONIZATION_HAPPENS_AFTER(&_M_use_count);
[move_group-1]       > 168: 	    _M_dispose();
[move_group-1]         169: 	    // There must be a memory barrier between dispose() and destroy()
[move_group-1]         170: 	    // to ensure that the effects of dispose() are observed in the
[move_group-1]         171: 	    // thread that runs destroy().
[move_group-1] #5  | Source "./src/rclcpp/node.cpp", line 280, in ~shared_ptr
[move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr.h", line 122, in ~__shared_ptr
[move_group-1]     |   120:   */
[move_group-1]     |   121:   template<typename _Tp>
[move_group-1]     | > 122:     class shared_ptr : public __shared_ptr<_Tp>
[move_group-1]     |   123:     {
[move_group-1]     |   124:       template<typename... _Args>
[move_group-1]     | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 1154, in ~__shared_count
[move_group-1]     |  1152:       __shared_ptr(const __shared_ptr&) noexcept = default;
[move_group-1]     |  1153:       __shared_ptr& operator=(const __shared_ptr&) noexcept = default;
[move_group-1]     | >1154:       ~__shared_ptr() = default;
[move_group-1]     |  1155: 
[move_group-1]     |  1156:       template<typename _Yp, typename = _Compatible<_Yp>>
[move_group-1]       Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 705, in ~Node [0x7f2d3d74868e]
[move_group-1]         702:       ~__shared_count() noexcept
[move_group-1]         703:       {
[move_group-1]         704: 	if (_M_pi != nullptr)
[move_group-1]       > 705: 	  _M_pi->_M_release();
[move_group-1]         706:       }
[move_group-1]         707: 
[move_group-1]         708:       __shared_count(const __shared_count& __r) noexcept
[move_group-1] #4    Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 168, in _M_release [0x7f2d3d723cc9]
[move_group-1]         165: 	if (__gnu_cxx::__exchange_and_add_dispatch(&_M_use_count, -1) == 1)
[move_group-1]         166: 	  {
[move_group-1]         167:             _GLIBCXX_SYNCHRONIZATION_HAPPENS_AFTER(&_M_use_count);
[move_group-1]       > 168: 	    _M_dispose();
[move_group-1]         169: 	    // There must be a memory barrier between dispose() and destroy()
[move_group-1]         170: 	    // to ensure that the effects of dispose() are observed in the
[move_group-1]         171: 	    // thread that runs destroy().
[move_group-1] #3  | Source "/usr/include/c++/11/bits/shared_ptr_base.h", line 348, in ~NodeBase
[move_group-1]     |   346:       virtual void
[move_group-1]     |   347:       _M_dispose() noexcept
[... 113 lines cut ...]

def launch_setup(context, *args, **kwargs):
    robot_description = load_file("moveit_resources_pr2_description", "urdf/robot.xml")
    robot_description_semantic_config = load_file("moveit_resources_pr2_description", "srdf/robot.xml")

    # Trajectory Execution Functionality
    moveit_controllers = {
        "moveit_controller_manager": "moveit_simple_controller_manager/MoveItSimpleControllerManager",
    }

[... cut ...]
                               'robot_description': robot_description,
                               'robot_description_semantic': robot_description_semantic_config,
                           },
                           moveit_controllers,
                           ],
                           arguments=['--ros-args', '--log-level', 'WARN'],
                           )
    return [move_group_node]

def generate_launch_description():
    return LaunchDescription([OpaqueFunction(function=launch_setup)])
```

henningkayser, 2023-03-10:
isn't this fixed with #2001

samialperen, 2023-04-05:
@henningkayser  I have the same issue
```
[move_group-7]          at line 127 in ./src/class_loader.cpp
[move_group-7] Stack trace (most recent call last):
[move_group-7] #14   Object "", at 0xffffffffffffffff, in 
[move_group-7] #13   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55de4baf8784, in 
[move_group-7] #12   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f5c83206e3f]
[move_group-7] #11   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f5c83206d8f]
[move_group-7] #10   Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55de4baf76bb, in 
[move_group-7] #9    Object "/opt/ros/humble/lib/moveit_ros_move_group/move_group", at 0x55de4baf9769, in 
[move_group-7] #8    Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.4", at 0x7f5c83c27a46, in moveit_cpp::MoveItCpp::~MoveItCpp()
[move_group-7] #7    Object "/opt/ros/humble/lib/libmoveit_cpp.so.2.5.4", at 0x7f5c83c25999, in 
[move_group-7] #6    Object "/opt/ros/humble/lib/libmoveit_trajectory_execution_manager.so.2.5.4", at 0x7f5c82f305b5, in trajectory_execution_manager::TrajectoryExecutionManager::~TrajectoryExecutionManager()
[move_group-7] #5    Object "/opt/ros/humble/lib/libmoveit_trajectory_execution_manager.so.2.5.4", at 0x7f5c82f431a9, in 
[move_group-7] #4    Object "/robot_control_platform_dependencies_ws/install/rclcpp/lib/librclcpp.so", at 0x7f5c83847328, in rclcpp::Node::~Node()
[move_group-7] #3    Object "/robot_control_platform_dependencies_ws/install/rclcpp/lib/librclcpp.so", at 0x7f5c838470e1, in rclcpp::Node::~Node()
[move_group-7] #2    Object "/robot_control_platform_dependencies_ws/install/rclcpp/lib/librclcpp.so", at 0x7f5c8384cec8, in rclcpp::node_interfaces::NodeBase::~NodeBase()
[move_group-7] #1    Object "/robot_control_platform_dependencies_ws/install/rclcpp/lib/librclcpp.so", at 0x7f5c8384ccf5, in rclcpp::node_interfaces::NodeBase::~NodeBase()
[move_group-7] #0    Object "/robot_control_platform_dependencies_ws/install/rclcpp/lib/librclcpp.so", at 0x7f5c8384ddd2, in rclcpp::CallbackGroup::~CallbackGroup()
[move_group-7] Segmentation fault (Address not mapped to object [0x7f5c47c957a8])
```
moveit version: 2.5.4-1jammy.20230302.235511

JafarAbdi, 2023-04-05:
If I recall correctly https://github.com/ros-planning/moveit2/pull/1918 fixed this issue, @samialperen do you mind testing it?

samialperen, 2023-04-05:
@JafarAbdi tested, does not work. Same error exists.

samialperen, 2023-04-05:
I also tried https://github.com/ros-planning/moveit2/pull/2001/files with humble branch and it did not work either.

samialperen, 2023-04-06:
@JafarAbdi tested [#2001](https://github.com/ros-planning/moveit2/pull/2001/files) and all good. Thanks.

closed 2023-04-09 by JafarAbdi

timonegk, 2024-06-20:
This issue still exists for me with version 2.8.0

adi0x90, 2026-10-12:
I captured this `move_group` shutdown crash while running a physical FR3 with ROS 2 Humble and MoveIt 2.5.9. `ros2_control_node` first aborted after `franka::NetworkException: UDP receive: Timeout`. ROS launch then sent SIGINT to `move_group`, which crashed during teardown:

```text
moveit_cpp::MoveItCpp::~MoveItCpp()
trajectory_execution_manager::TrajectoryExecutionManager::~TrajectoryExecutionManager()
rclcpp::Node::~Node()
rclcpp::node_interfaces::NodeBase::~NodeBase()
rclcpp::CallbackGroup::~CallbackGroup()
Segmentation fault (exit -11)
```

This is a shortened stack from one shutdown; it omits paths, process IDs, and addresses. I have not reproduced the crash on a clean shutdown or proved the memory-lifetime mechanism. The same 2.5.9 destructor chain also appears in [franka_ros2#131](https://github.com/frankarobotics/franka_ros2/issues/131) and [moveit2#3681](https://github.com/moveit/moveit2/issues/3681).

In [MoveIt 2.5.9](https://github.com/moveit/moveit2/blob/2.5.9/moveit_ros/planning/moveit_cpp/src/moveit_cpp.cpp#L86-L90), `MoveItCpp::~MoveItCpp()` still calls `clearContents()`, which resets `planning_scene_monitor_` without first resetting `trajectory_execution_manager_` ([source](https://github.com/moveit/moveit2/blob/2.5.9/moveit_ros/planning/moveit_cpp/src/moveit_cpp.cpp#L282-L288)). [#2001](https://github.com/moveit/moveit2/pull/2001) removed that explicit clearing on main, after which this issue was closed. Does Humble 2.5.9 need a backport of #2001 for this destructor stack?