moveit/moveit2 #1600: Crashes/ Freezes while building MoveIt2 from source with limited HW resources

sjahr, 2022-10-05:
### Description

You experience crashes or your system freezes when you try to build MoveIt2 on a system with limited/ weaker hardware resources.

### Your environment
* ROS Distro: Any ROS2 distro
* Source or Binary build? Source
* If source, which branch? All

### Steps to reproduce
Build MoveIt2 from source on your machine with correct dependencies

### Expected behaviour
MoveIt2 should require a longer build time but build without problems

### Actual behaviour
You experience crashes/ freezes

### Related Issues
#1594

sjahr, 2022-10-05:
The problem is related to the potential high RAM usage of colcon. So you can fix it by reducing the RAM load:

:arrow_right: [Configuring colcon's executor](https://colcon.readthedocs.io/en/released/reference/executor-arguments.html) with the `--executor` flag:
```
colcon build --event-handlers desktop_notification- status- --executor sequential
```
or the `--parallel-workers` flag
```
colcon build --event-handlers desktop_notification- status- --parallel-workers 1
```

closed 2022-10-05 by sjahr

yigitboracagiran, 2024-09-20:
`export MAKEFLAGS="-j 1"`

`colcon build --executor sequential`

The -j option allows you to specify the number of jobs which run in parallel. Each job can be run on a separate CPU core, which can significantly speed up the build process on multi-core systems.

Note: You can change the variable, it doesn't need to be 1. 

Be aware that you will increase the chance of freezing every time you increase the variable but in the other hands, building time will be increase every time you decrease the variable. You can see the effect of building the project on your system with `htop` command in another terminal.