moveit/moveit2 #1611: Moveit2 Source Build Linux - Colcon Build Error

CanBayraktaroglu, 2022-10-13:
**Description**
Attempting to build MoveIt2 with ROS2 Humble installed from binary. I have tried every single possible solution I could find in the previously closed issues, unfortunately to no avail.

**Your Environment**
ROS Distro: Humble
OS Version: Ubuntu 22.04
Source or Binary build: Binary

**Steps to reproduce**
Follow instructions at https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html and https://moveit.ros.org/install-moveit2/source/

**Expected behaviour**
The build should succeed.

**Actual behaviour**
Build Errors with controller_manager relating to absent export targets. Failed moveit2 build.

**Backtrace or Console Output**

```

 bayraktaroglu@for1583lnx:~/ws_moveit2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --allow-overriding controller_interface controller_manager hardware_interface ros2_control_test_assets

Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> controller_manager_msgs
Starting >>> moveit_configs_utils
Starting >>> ros2_control_test_assets
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_common [1.01s]
Starting >>> joint_limits
Finished <<< moveit_resources_fanuc_description [1.05s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_description [1.19s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [1.18s]
Finished <<< moveit_resources_prbt_support [1.22s]
Finished <<< ros2_control_test_assets [1.30s]
Starting >>> hardware_interface
Finished <<< moveit_resources_fanuc_moveit_config [0.66s]
Finished <<< joint_limits [0.83s]
Finished <<< moveit_resources_panda_moveit_config [0.67s]
Starting >>> moveit_core
Starting >>> moveit_resources
--- stderr: moveit_configs_utils
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
   warnings.warn(
---
Finished <<< moveit_configs_utils [2.22s]
Finished <<< moveit_resources [0.47s]
Finished <<< hardware_interface [1.66s]
Starting >>> controller_interface
Starting >>> transmission_interface
Finished <<< controller_interface [2.91s]
Finished <<< transmission_interface [3.10s]
Finished <<< controller_manager_msgs [6.88s]
Starting >>> controller_manager
--- stderr: controller_manager
CMake Error at /home/bayraktaroglu/ws_moveit2/install/controller_manager_msgs/share/controller_manager_msgs/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/home/bayraktaroglu/ws_moveit2/install/controller_manager_msgs/share/controller_manager_msgs/cmake/export_controller_manager_msgs__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /home/bayraktaroglu/ws_moveit2/install/controller_manager_msgs/share/controller_manager_msgs/cmake/controller_manager_msgsConfig.cmake:41 (include)
  CMakeLists.txt:21 (find_package)
---
Failed   <<< controller_manager [3.92s, exited with code 1]
Aborted  <<< moveit_core [56.9s]
Summary: 15 packages finished [59.6s]
  1 package failed: controller_manager
  1 package aborted: moveit_core
  2 packages had stderr output: controller_manager moveit_configs_utils
  36 packages not processed
bayraktaroglu@for1583lnx:~/ws_moveit2$ 

```

tylerjw, 2022-10-13:
Do you mind running `vcs export` in your source directory and posting it here?

CanBayraktaroglu, 2022-10-13:
Of course! Thank you for the quick response. Here is the output of the command

```
repositories:
  moveit2:
    type: git
    url: https://github.com/ros-planning/moveit2.git
    version: humble
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources.git
    version: ros2
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control.git
    version: 601292093be6d309fc6bbb5bc879ecb80cf5a3bf
```

samcts2309, 2022-10-14:
what's your cmake version? I had similar issue with cmake 3.24.1, and my solution was drop back cmake version to 3.23.2.

CanBayraktaroglu, 2022-10-14:
when i run the cmake --version command i get 3.24.2, but when i run apt policy cmake i get the following:

```
cmake:
  Installed: 3.22.1-1ubuntu1.22.04.1
  Candidate: 3.22.1-1ubuntu1.22.04.1
  Version table:
 *** 3.22.1-1ubuntu1.22.04.1 500
        500 http://de.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     3.22.1-1ubuntu1 500
        500 http://de.archive.ubuntu.com/ubuntu jammy/main amd64 Packages
```
which version are you asking for and how do i downgrade the one affecting the moveit2 build?

tylerjw, 2022-10-17:
I'm not sure how you downgrade as it seems 3.22.1 is the one you have installed via apt. Maybe you installed it manually somehow?

When I run cmake --version on my machine (ubuntu 22) I see this response:
```
cmake version 3.22.1

CMake suite maintained and supported by Kitware (kitware.com/cmake)
```

Running `which cmake` might help you figure out where your newer version of cmake is and that might help you understand how to remove the newer version.

CanBayraktaroglu, 2022-10-17:
Allright, I managed to install cmake 3.23.2 separately and changed the paths for cmake to point to it. Moveit2 including the tutorials built succesfully. Yet for future developers, I request you to add some clear and detailed instruction to the moveit2 source build page regarding the specific cmake version necessary for build. Closing the issue.

closed 2022-10-17 by CanBayraktaroglu

tylerjw, 2022-10-17:
Thank you for following up this fixed your issue.