moveit/moveit2 #1625: 'std::length_error' on load Planning scene in foxy

SmarkSauer, 2022-10-19:
### Description
* ROS Distro: Foxy
* OS Version:Ubuntu 20.04
* Binary

Rviz Crashes on loading a planning scene:

[rviz2-1] [INFO] [1666187620.931024455] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1666187620.934407915] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] terminate called after throwing an instance of 'std::length_error'
[rviz2-1]   what():  Exceeded upper bound

henningkayser, 2023-01-26:
@SmarkSauer have you found out a bit more about this in the meantime? It's hard to debug this without more context 
 like a backtrace log. Would switching to a new distro like Humble work for you?

henningkayser, 2023-03-10:
Foxy is deprecated

closed 2023-03-10 by henningkayser