moveit/moveit2 #1632: Allow specifying controllers in a move group plan

sea-bass, 2022-10-21:
## Background
This is motivated by MoveIt Studio, but has a more general use case. Suppose there are several controllers that can be toggled for a specific robot.

When executing a trajectory, either through `MoveItCpp::execute()` or `MoveGroupInterface::execute()`, there is currently no option to explicitly specify controller names. 

`TrajectoryExecutionManager` has its own logic to enable controllers based on the joints that need to be controller and/or the currently active controller, but sometimes it's preferred to be explicit about this. Additionally, in a separate Gitissue (https://github.com/ros-planning/moveit2/issues/1631), we additionally found a logic bug in which this logic is not always selecting the active controller that we manually switched to by directly calling the controller switching service.

## Desired Solution
What seems plausible is to expose a way to explicitly pass in controller names to override the logic that auto-assigns controllers based on joint names. This could happen in two ways I can think of, both of which unfortunately require some finessing of the interfaces.

### 1. Modify the function interface itself
The following are just examples of adding an optional vector of strings for the controller names to specify, at the interface level.

Before:
```cpp
# MoveItCpp
MoveItCpp::execute(const std::string& group_name, const robot_trajectory::RobotTrajectoryPtr& robot_trajectory,
                   bool blocking)
# MoveGroupInterface
moveit::core::MoveItErrorCode execute(const moveit_msgs::msg::RobotTrajectory& trajectory, bool wait)
```

After:
```cpp
# MoveItCpp
MoveItCpp::execute(const std::string& group_name, const robot_trajectory::RobotTrajectoryPtr& robot_trajectory,
                   const std::optional<std::vector<std::string>>& controller_names, bool blocking)
# MoveGroupInterface
moveit::core::MoveItErrorCode execute(const moveit_msgs::msg::RobotTrajectory& trajectory, 
                  const std::optional<std::vector<std::string>>& controller_names, bool wait)
```

### 2. Modify the robot trajectory representation itself
Alternatively, we could keep the same arguments and modify the trajectory information so that the trajectory itself, or maybe each trajectory portion, has its set of controllers. We've done this with MTC in [this branch](https://github.com/JafarAbdi/moveit_task_constructor/blob/pr-ms_humble/msgs/msg/SubTrajectory.msg), where we add a new `[TrajectoryExecutionInfo` message](https://github.com/JafarAbdi/moveit_task_constructor/blob/pr-ms_humble/msgs/msg/TrajectoryExecutionInfo.msg) that contains controller names.

A similar thing could be done here for the `moveit_msgs::msg::RobotTrajectory` message definition in `moveit_msgs` (in the case of `MoveGroupInterface`), and in the `RobotTrajectory` class definition of `moveit_core`.

AndyZe, 2022-10-22:
I'm cool with this idea. I like Option 1 better than Option 2.

If we start doing this type of thing, would we even need ControllerManagers anymore? I can't think of a reason. Maybe @mikeferguson has one.

schornakj, 2022-10-23:
> If we start doing this type of thing, would we even need ControllerManagers anymore?

In the situation described, the MoveIt controller manager is the thing that's ultimately consuming the list of controllers and acting on it to set the required controller states, so I think we'd still be using it.

AndyZe, 2022-10-23:
OK, that's good to know. But, do you think the extra complexity of the controller managers is worth it? It requires every MoveIt2 user to set up a controller manager. 95% of them will never know what it does and they'll just use the same controller for every motion.

I remember the specifics of your (MoveIt Studio) case now -- you want to execute some trajectories with a Joint Trajectory Controller, some with an Admittance Controller. I think this is pretty unusual and anybody who is involved in such a situation can handle controller switching on their own.

Anyway, I think we should simplify and delete the Controller Managers.

AndyZe, 2022-10-24:
I guess we still need a ControllerManager for executing a combined (gripper + arm) trajectory, for example. Something where the joints are shared across 2 controllers. Sigh.

PR #2257 (2023-07-25): Specify controller name in MGI execution
Resolve #1632 

This PR adds the ability to execute via MGI while specifying a controller name. This can be useful for more fine control over controller-switching applications for which multiple controllers apply, and it is undesirable to let TrajectoryExecutionManager selet the controller for you.

Related and required change in moveit_msgs: https://github.com/ros-planning/moveit_msgs/pull/160
closed 2023-07-25 by sjahr