moveit/moveit2 #1633: Group X is not a chain

swiz23, 2022-10-21:
### Description
There seems to be a few unexpected warning messages that appear in the terminal if there is a planning group defined in an SRDF file which does not have an associated kinematics plugin. Reasons to do this might be to have a group of links that you want to disable collisions for in a given application. This did not used to happen, and it's unclear why it's happening now.

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Source
* If source, which branch? main

### Steps to reproduce
Add the following text to the SRDF in the MoveIt Fanuc Config package [here](https://github.com/ros-planning/moveit_resources/blob/ros2/fanuc_moveit_config/config/fanuc.srdf). This will create an 'eoat' group with 'tool0' as the single link.
```
    <group name="eoat">
        <link name="tool0"/>
    </group>
```

Build and Launch the package by typing `ros2 launch moveit_resources_fanuc_moveit_config demo.launch.py `

### Expected behaviour
The warnings shown below should not appear.

### Actual behaviour
These warning messages appear in the terminal. It doesn't affect usage, but it's unclear why this is happening now when it never happened before.
```
[move_group-4] [ERROR] [1666380869.436552686] [moveit_kinematics_base.kinematics_base]: Group 'eoat' is not a chain
[move_group-4] [ERROR] [1666380869.436566050] [kinematics_plugin_loader]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'eoat'
[move_group-4] [ERROR] [1666380869.436577384] [moveit_kinematics_base.kinematics_base]: Group 'eoat' is not a chain
[move_group-4] [ERROR] [1666380869.436580639] [kinematics_plugin_loader]: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'eoat'
[move_group-4] [ERROR] [1666380869.436584469] [moveit_ros.robot_model_loader]: Kinematics solver could not be instantiated for joint group eoat.
```

@JafarAbdi - any suggestions?

swiz23, 2023-01-03:
Thanks @sjahr, I'll try the latest changes and let you know how it goes.

swiz23, 2023-01-05:
Unfortunately, I'm still seeing this error. I confirmed that I didn't have an older version of moveit_ros_planning on the system that maybe moveit was linking against. Also, I confirmed that the files modified in the PR were also modified on my local system. FWIW, if I change the kinematics solver type to LMA, it prints out the same error but with the LMA plugin. Any suggestions?

henningkayser, 2023-01-20:
This behavior should have been fixed with https://github.com/ros-planning/moveit2/pull/1816 afaik. Is it possible, that you have some old build artifacts in your workspace or maybe old debian packages? Please try a clean install and report if it still happens for you.

swiz23, 2023-01-25:
I've tried to rebuild my docker image without cache a couple times, but did not see any change to this.

swiz23, 2023-02-02:
Closing since issue was resolved with PR #1903.

closed 2023-02-02 by swiz23