moveit/moveit2 #1637: Add ST-RRT* Planner to MoveIt

swiz23, 2022-10-24:
**Is your feature request related to a problem? Please describe.**
Currently, the ST-RRT* planner (added into ompl 9 months ago) is not yet integrated into  the [moveit_planners_ompl](https://github.com/ros-planning/moveit2/tree/main/moveit_planners/ompl) package. According to https://vhartmann.com/strrt/, this planner outperforms both RRT-Connect and RRT* on initial solution time and attained final solution cost. As a result, I think it would be advantageous to have this planner integrated with MoveIt.

**Describe the solution you'd like**
[ST-RRT* planner](https://github.com/ompl/ompl/blob/main/src/ompl/geometric/planners/rrt/src/STRRTstar.cpp) integrated into moveit_planners_ompl.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

henningkayser, 2022-11-10:
Enabling an OMPL planner should be as easy as adding an entry to [this function](https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L280) and providing a [planner config entry](https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/ompl_planning.yaml#L10). Have you tried this yet? Looks promising.

swiz23, 2022-11-11:
@henningkayser,

Yes, I've tried that, but I get an error message printed out from [here](https://github.com/ompl/ompl/blob/d8375d842a7f016d64f27a81d6f95eaa83fbe595/src/ompl/geometric/planners/rrt/src/STRRTstar.cpp#L44) that state space needs to be of type SpaceTimeStateSpace.

closed 2025-02-18 by github-actions[bot]