moveit/moveit2 #1640: Loading servo parameters defaulting to Panda

skaeringur97, 2022-10-28:
When loading servo parameters with `moveit_servo::ServoParameters::makeServoParameters(node, "servo")` it loads the default configurations for the Panda robot, declared in [servo_parameters.h](https://github.com/ros-planning/moveit2/blob/0ec600d1257b9198808f48598c499491cabd8eb1/moveit_ros/moveit_servo/include/moveit_servo/servo_parameters.h). If I use  `moveit_servo::ServoParameters::makeServoParameters(node)` instead, it loads the configurations from my .yaml file

I would like to have it always load your own .yaml file instead and give an error if something is missing from the .yaml file. Having the default values can have hidden errors.

henningkayser, 2022-11-10:
You're right, the default values are really not ideal here. We're currently working on refactoring all of MoveIt's parameters to use proper declaration and defaults using the [generate_parameter_library](https://github.com/PickNikRobotics/generate_parameter_library). See https://github.com/ros-planning/moveit2/pulls?q=is%3Apr+generate_parameter_library for examples. Contributions are always welcome ;)

sea-bass, 2024-11-03:
Since this is now in place, I'll close the issue.

closed 2024-11-03 by sea-bass