moveit/moveit2 #1682: Use psm->requestPlanningSceneState() instead of psm->getPlanningScene()

skaeringur97, 2022-11-01:
I've had some trouble with collisions if I have published collision objects before my own node. That is because in some places, like [MoveItCpp](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/moveit_cpp/src/moveit_cpp.cpp#L98) and the [local planner](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/hybrid_planning/local_planner/local_planner_component/src/local_planner_component.cpp#L89) in the hybrid planning example, the current scene is gotten with the function getPlanningScene(). Basically, if that function is used, the planner doesn't know about any collision objects published before its own node and will plan twards a collision. If I change it to `psm->requestPlanningSceneState("/get_planning_scene")` then it sees the collision and doesn't plan to it.

henningkayser, 2022-11-10:
Hi @skaeringur97, apparently you are running a remote node (MoveGroup?) that is maintaining your planning scene. In that case, proper synchronization is indeed necessary, which can be done with `requestPlanningSceneState()` in the user code. But this is somewhat specific to your setup and doesn't account for all use cases, especially if it's not clear which one is the "main" planning scene. This is why remote sync it's not implemented by default. The IMO more elegant way is to only call `requestPlanningSceneState()` initially to sync up the scene with the remote node and then use `startWorldGeometryMonitor()` with the right topic settings to listen for remote scene updates.

sjahr, 2022-12-09:
@skaeringur97 It seems like this issue is resolved by Hennings answer. Please re-open if you have further questions in this matter

closed 2022-12-09 by sjahr