moveit/moveit2 #1686: Local planner gets stuck, causing jittery motion

skaeringur97, 2022-11-02:
### Description
When I send positions to a hybrid planner, using either UR10e or Kinova Kortex, the robot jitters a lot before either succeeding or crashing. Weirdly, I don't have this issue when running the Panda robot. However, another user had the same problem explained in #1536 which makes me think this is a MoveIt issue and not a robot specific problem.

I followed the advice in #1536 and #819, lowering the local planners frequency and changing how the psm updates the scene, which mostly solved this issue. I also had to increase the `STUCK_ITERATIONS_THRESHOLD` in forward_trajectory.cpp to 50. However, I feel like there should be a better solution to this.

Here is an example of the Kinova robot showing the problem:

https://user-images.githubusercontent.com/31903354/199453104-3b762b7f-6562-4750-98c2-590755b7bf2d.mp4

And here is the same scenario after my fix:

https://user-images.githubusercontent.com/31903354/199453140-d483f255-630b-48f4-8a9f-d91aac6e837d.mp4

### Your environment
* ROS 2: [Humble]
* OS Version: Ubuntu 22.04
* ROS2 Source build and MoveIt2 Binary build

### Steps to reproduce
Clone the [hybrid planning example](https://moveit.picknik.ai/humble/doc/examples/hybrid_planning/hybrid_planning_tutorial.html) and change the test/hybrid_planning_demo_node.cpp to [this](https://gist.github.com/skaeringur97/f8165f47531047494b58f4aed394e12e) and run it with the [UR10e](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) or [Kinova robot](https://github.com/Kinovarobotics/kortex). If you manage to reproduce the error, you can try the fix by lowering the frequency in test/config/local_planner.yaml, changing `STUCK_ITERATIONS_THRESHOLD` to 50 and change `planning_scene_monitor_->updateFrameTransforms()` to `planning_scene_monitor_->updateSceneWithCurrentState()` in local_planner/local_constraint_solver_plugins/src/forward_trajectory.cpp 

### Expected behaviour
The robot moves to desired position.

### Actual behaviour
The robot gets stuck in the local planner, re-plans before getting stuck again causing a jittery movement.

### Backtrace or Console output
[Console output](https://gist.github.com/skaeringur97/fd1ca9b2a51fd5b62c9adebcf5f14fdf)

sjahr, 2022-11-08:
Hi @skaeringur97, thank you for opening this issue. The currently available plugins of the architecture are example implementations and might be unsuitable for your specific application. My suspension is that this might by caused by latencies between the robot and the hybrid planning architecture and/or that as you suggested, the STUCK_ITERATIONS_THRESHOLD is not configured correctly. I'd recommend you refactor the local planner plugin to fit your needs. Feel free to ping me if you have any questions :)  We implemented for example custom plugins for the a welding application in a collaboration with the Fraunhofer IPA. Here is the code https://github.com/PickNikRobotics/UR10e_welding_demo (custom plugin implementations can be found [here](https://github.com/PickNikRobotics/UR10e_welding_demo/tree/master/hybrid_planning_demo)) and a blog post about it https://picknik.ai/white-papers/2022-01-13-Hybrid-planning-Welding-with-the-UR10e.html.

skaeringur97, 2022-11-09:
Hi @sjahr, thank you. I will check it out and see if I can modify the local planner plugin.

I did find out that if I replace the robot command msg in forward_trajectory.cpp to be the whole local_trajectory and not just the closest point at each time, this problem disappeared. However, instead the robot moves super slowly. Here it is before this change:

https://user-images.githubusercontent.com/31903354/200806930-288b56b4-da71-4883-af80-a0f96397e509.mp4

Here it is after the change:

https://user-images.githubusercontent.com/31903354/200806940-3d27eff4-b5e1-4348-916a-6c3cab1da4f2.mp4

The only change I made was replacing the line `robot_command.getRobotTrajectoryMsg(robot_command_msg);` with `local_trajectory.getRobotTrajectoryMsg(robot_command_msg);` and commented out the local planner getting stuck detection. 

I also forgot to mention that I get this error, when the local planner gets stuck, in the terminal where I launch my driver:
`[ros2_control_node-1] [ERROR] [1667989446.262952668] [joint_trajectory_controller]: Holding position due to state tolerance violation`

skaeringur97, 2022-11-09:
The above fixes don't take the collision check into account, I had to change a bit more. It now looks like [this](https://gist.github.com/skaeringur97/f4085fd862618285618f447a20494696). This works, but it's not the best solution.. Do you have any ideas @sjahr for a better solution?

skaeringur97, 2022-11-09:
I also tried the servo implementation that the custom plugin implementation example used, with a few modifications because of the end-effector in that example, but I then get this warning msg in the driver terminal: `Joints on incoming trajectory don't match the controller joints.` I have verified the joint names in robot_command. The robot does follow the trajectory correctly, very slowly however, only after I increased the singularity thresholds to 170000 and 300000.

skaeringur97, 2022-11-10:
I found the issue! In the driver for the robot, there are position constraints set for each joint in the controllers.yaml file:
```
shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
elbow_joint: { trajectory: 0.2, goal: 0.1 }
wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
wrist_3_joint: { trajectory: 0.2, goal: 0.1 }
``` 
If I increase the limit or remove the constraints, the issue disappears.

skaeringur97, 2022-11-10:
Limiting the joints more also works!

bnbhat, 2023-09-03:
I'm facing a similar issue while trying to implement this on a UR5e robot. In my case, the robot is not moving at all, and the local planner is behaving just like in the video above. I don't see anything logging in the ur ros2 driver.

I have tried changing `STUCK_ITERATIONS_THRESHOLD`, `planning_scene_monitor_->updateFrameTransforms()` and local planners' frequency, but it did not help. 

This is how I launch 
- launch ur driver `ros2 launch ur_robot_driver ur5e.launch.py robot_ip:=192.168.56.101 launch_rviz:=false`
- launch hybrid planning  ` ros2 launch arc_hybrid_planner ur_moveit_hybrid_planner.launch.py`  launch file can be found [here](https://gist.github.com/bnbhat/d6c326b3ad3a1b3bc9eeba37c6153d35)

config:
[local_planner.yaml](https://gist.github.com/bnbhat/fb8b0d122c5af174388ec1b15d390cff)
[global_planner.yaml](https://gist.github.com/bnbhat/58da45631bcb9803169784cdd7a88668)

Here is the [error msg](https://gist.github.com/bnbhat/251b44a179dc2a67dd975151766e1e6c)

My Env:
ROS 2 Humble
Ubuntu 22.04
ursim simulation

@skaeringur97  Were you able to get it running perfectly? if yes, would you mind sharing your configurations?

Any help would be greatly appreciated. Thank you!

skaeringur97, 2023-09-04:
Have you tried limiting the joints more? In my case, when I allowed each joint to move from -360° to 360° the local planner had issues. But in most cases you don't need that kind of movement capabilities. Try limiting the joints such that it can still move within your workspace, but not more than that.

bnbhat, 2023-09-18:
Thank you so much for the reply! @skaeringur97 

I tried limiting the joints, but the local planner is still getting stuck.
When I compare it with the videos above, in my case robot is not moving at all.
[terminal output](https://gist.github.com/bnbhat/8ce61bbd2af2d9e2d839c7fead741fd4)

https://github.com/ros-planning/moveit2/assets/66437415/c1736c82-7d88-429a-b0d1-197867dbdb8f

I don't see anything getting logged in ros2-ur-driver. I suspect the local planner is not able to talk to the robot driver at all. But I can move the robot with MoveIt rviz GUI. 

What could be the reason for such behaviour?

skaeringur97, 2023-09-18:
I´m assuming you´re using fake hardware and not running it on actual hardware? If true, try using use_fake_hardware=true. You can also try use_sim_time=true and setting the ur_type if it's not ur5. Hope that helps :)

bnbhat, 2023-09-18:
I am using URSim simulator with Docker, which acts almost identically to a real robot connected over the network.

skaeringur97, 2023-09-18:
Do you then maybe need to set use_sim_time and/or use_fake_hardware to true?

bnbhat, 2023-09-18:
Thank you for the reply. I will try it now.