moveit/moveit2 #1689: Remove _ suffix from struct member variables

sjahr, 2022-11-03:
The _ suffix is a naming convention for private member variables of classes, not public struct members. Therefore, the _ suffix of members of [MotionPlanResponse](https://github.com/ros-planning/moveit2/blob/8c29ad9fa3e8a1fdd8e0c8d232b9273bc5754997/moveit_core/planning_interface/include/moveit/planning_interface/planning_response.h#L46) and [MotionPlanDetailedResponse](https://github.com/ros-planning/moveit2/blob/8c29ad9fa3e8a1fdd8e0c8d232b9273bc5754997/moveit_core/planning_interface/include/moveit/planning_interface/planning_response.h#L59) should be removed.

JoaoVitorMenezesMagalhaes, 2022-11-03:
Hello @sjahr!! Can you assign me to this issue, please ?

JoaoVitorMenezesMagalhaes, 2022-11-04:
I believe i have fixed it with my PR #1693

AlexWebb03, 2023-01-26:
Hiya, I'd like to work on this. Should I finish @JoaoVitorMenezesMagalhaes MR or open a new one?

sjahr, 2023-01-31:
@AlexWebb03 Great thank you!

> Should I finish @JoaoVitorMenezesMagalhaes PR or open a new one?

Whatever you prefer :+1:

closed 2023-02-14 by sjahr