moveit/moveit2 #1702: No kinematics plugins defined. Fill and load kinematics.yaml!

jwson97, 2022-11-09:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? humble

### Steps to reproduce
On Your First C++ Moveit Project tutorials, (https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html)

ros2 launch moveit2_tutorials demo.launch.py (shell 1)
ros2 run hello_moveit hello_moveit (shell 2)

### Expected behaviour
Everything is working fine

### Actual behaviour
Everything is working fine.
But I got error with this when running ros2 run hello_moveit hello_moveit

"[moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!"

An error occurs, but the action is performed.

To solve this error, I changed code 

    ###################### original code
    # planning_context
    moveit_config = (
        MoveItConfigsBuilder("moveit_resources_panda")
        .robot_description(file_path="config/panda.urdf.xacro")
        .trajectory_execution(file_path="config/gripper_moveit_controllers.yaml")
        .planning_scene_monitor(
            publish_robot_description=True, publish_robot_description_semantic=True
        )
        .to_moveit_configs()
    )

     ###################### modified code
    # planning_context
    moveit_config = (
        MoveItConfigsBuilder("moveit_resources_panda")
        .robot_description(file_path="config/panda.urdf.xacro")
        .robot_description_kinematics(file_path="config/kinematics.yaml")
        .trajectory_execution(file_path="config/gripper_moveit_controllers.yaml")
        .planning_scene_monitor(
            publish_robot_description=True, publish_robot_description_semantic=True
        )
        .to_moveit_configs()
    )

But still I got error.

### Backtrace or Console output

"[moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!"

henningkayser, 2022-11-10:
Interesting... That's the exact path that `robot_description_kinematics` defaults to. What happens if you use below?

```
moveit_config = (
    MoveItConfigsBuilder("moveit_resources_panda")
    .robot_description(file_path="config/panda.urdf.xacro")
    .robot_description_kinematics()
    .trajectory_execution(file_path="config/gripper_moveit_controllers.yaml")
    .planning_scene_monitor(
        publish_robot_description=True, publish_robot_description_semantic=True
    )
    .to_moveit_configs()
)
```
@JafarAbdi any thoughts?

jwson97, 2022-11-11:
Hi, 

I tried your code. But same result with warn.

henningkayser, 2023-01-20:
This might come in late, but it just occurred to me that this issue is very much the same problem as described here https://github.com/ros-planning/moveit2/issues/1843#issuecomment-1396881288. Running `hello_moveit` as executable means that parameters like the IK config are not initialized, so MoveIt is complaining about a missing config file. The fix is to launch your MGI node with proper parameters including the `kinematics.yaml`. Please close this issue if you think this solves your problem.

closed 2023-01-25 by jwson97

Nielk7991, 2023-11-17:
Hello @jwson97 and @henningkayser, 

I have exactly the same problem. I also use Humble and the same Ubuntu system.
When I try to run the demo and try to approach a pose via the hello moveit node, the node does not find any kinematics. yaml
Can you please help me with this?
This is my launch file:
```
from launch import LaunchDescription
from launch_ros.actions import Node
from moveit_configs_utils import MoveItConfigsBuilder
from moveit_configs_utils.launches import generate_demo_launch
from launch.substitutions import PathJoinSubstitution
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():

moveit_config = (MoveItConfigsBuilder("threedof_manipulator", package_name="moveit_endeffektor")
    .robot_description(file_path="config/threedof_manipulator.urdf.xacro")
    .robot_description_semantic(file_path="config/threedof_manipulator.srdf")
   .robot_description_kinematics(file_path="config/kinematics.yaml")
    .planning_pipelines(
            pipelines=["ompl", "pilz_industrial_motion_planner", "chomp"],
            default_planning_pipeline="ompl",
        )
    .to_moveit_configs()
)

move_group_demo = Node(
    name="hello_moveit",
    package="path_planner",
    executable="DummyNode",
    output="screen",
    parameters=[
        moveit_config.robot_description_kinematics,
    ],
)

return LaunchDescription([generate_demo_launch(moveit_config), move_group_demo])

OneOneLiu, 2024-08-19:
> This might come in late, but it just occurred to me that this issue is very much the same problem as described here [#1843 (comment)](https://github.com/moveit/moveit2/issues/1843#issuecomment-1396881288). Running `hello_moveit` as executable means that parameters like the IK config are not initialized, so MoveIt is complaining about a missing config file. The fix is to launch your MGI node with proper parameters including the `kinematics.yaml`. Please close this issue if you think this solves your problem.

I got the same warning and solved it by starting the `hello_moveit` node using `ros2 launch` rather than `ros2 run`. The launch.py file is as follows:
```python
from launch import LaunchDescription
from launch_ros.actions import Node
from moveit_configs_utils import MoveItConfigsBuilder

def generate_launch_description():
    # Set MoveIt configuration
    moveit_config = MoveItConfigsBuilder("moveit_resources_panda").to_moveit_configs()

    # Define hello_moveit node
    hello_moveit_node = Node(
        package="hello_moveit",
        executable="hello_moveit",
        output="screen",
        parameters=[
            moveit_config.robot_description,  # Load URDF
            moveit_config.robot_description_semantic,  # Load SRDF
            moveit_config.robot_description_kinematics,  # Load kinematics.yaml
        ],
    )

    return LaunchDescription([hello_moveit_node])
```

zhuxiyu24, 2024-11-08:
> > This might come in late, but it just occurred to me that this issue is very much the same problem as described here [#1843 (comment)](https://github.com/moveit/moveit2/issues/1843#issuecomment-1396881288). Running `hello_moveit` as executable means that parameters like the IK config are not initialized, so MoveIt is complaining about a missing config file. The fix is to launch your MGI node with proper parameters including the `kinematics.yaml`. Please close this issue if you think this solves your problem.
> 
> I got the same warning. Following the quoted instruction, I resolved it by starting the `hello_moveit` node using `ros2 launch` rather than `ros2 run`. The launch.py file is as follows:
> 
> from launch import LaunchDescription
> from launch_ros.actions import Node
> from moveit_configs_utils import MoveItConfigsBuilder
> 
> def generate_launch_description():
>     # Set MoveIt configuration
>     moveit_config = MoveItConfigsBuilder("moveit_resources_panda").to_moveit_configs()
> 
>     # Define hello_moveit node
>     hello_moveit_node = Node(
>         package="hello_moveit",
>         executable="hello_moveit",
>         output="screen",
>         parameters=[
>             moveit_config.robot_description,  # Load URDF
>             moveit_config.robot_description_semantic,  # Load SRDF
>             moveit_config.robot_description_kinematics,  # Load kinematics.yaml
>         ],
>     )
> 
>     return LaunchDescription([hello_moveit_node])

Thanks! Your solution helps me.