moveit/moveit2 #1708: MotionPlanResponse and MotionPlanDetailedResponse are redundant

AndyZe, 2022-11-09:
`[MotionPlanResponse](https://github.com/ros-planning/moveit2/blob/72e29b4f738c6a8d1eef3f46ac631523027660da/moveit_core/planning_interface/include/moveit/planning_interface/planning_response.h#L48)` and `[MotionPlanDetailedResponse](https://github.com/ros-planning/moveit2/blob/72e29b4f738c6a8d1eef3f46ac631523027660da/moveit_core/planning_interface/include/moveit/planning_interface/planning_response.h#L75)` are very similar. `MotionPlanDetailedResponse` seems to simply be a vectorized version of the other. Why can't we get rid of `MotionPlanResponse` and always use the vectorized version?

AndyZe, 2022-11-09:
I'm curious what @sjahr would say about this b/c he's been poking around in this code recently.

sjahr, 2022-11-10:
@henningkayser and I discussed that when I was working on that part of the code [here](https://github.com/ros-planning/moveit2/pull/1542#discussion_r982391876) but changing it was beyond the scope of the PR (Should have opened an issue like this though). I think this would be a good idea although, as Henning pointed out, the detailed response includes a vector of trajectories. We need decide if we want to keep it. What do you think?

henningkayser, 2022-11-10:
The detailed version is for debugging and may produce some overhead. I wouldn't really remove any of those, since we would have to pass a debug flag with the request to either get the vectors filled or not. However, we could argue that a better debugging interface would report from the individual planning steps directly, e.g. with callbacks returning individual MotionPlanResponse messages from the planners or adapters.

closed 2025-02-18 by github-actions[bot]