moveit/moveit2 #1738: Completed trajectory execution with status ABORTED

marioesteban11, 2022-11-18:
Hi everyone, 

I'm trying to move the ur3e robot with the URSIM, with moveit. but a have this error: 

> Completed trajectory execution with status ABORTED

I'm in:
ROS2 HUMBLE
UBUNTU 22.04

As I said before, Im trying to emulate some moves in my ur3e robot with moveit, rviz and URSIM. My problem comes when I try to move the robot with the moveit interface. Because the following error shows up:

`[move_group-1] [INFO] [1668761715.451816518] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[rviz2-2] [INFO] [1668761715.451979040] [move_group_interface]: Plan and Execute request accepted
[move_group-1] [INFO] [1668761715.452000062] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-1] [INFO] [1668761715.453266881] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-1] [INFO] [1668761715.453406408] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-1] [INFO] [1668761715.453418121] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1668761715.454242965] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [WARN] [1668761715.547564862] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-1] [INFO] [1668761715.549095042] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549114561] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549182105] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-1] [INFO] [1668761715.549547939] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1668761715.549578320] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549593870] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1668761715.549728407] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to scaled_joint_trajectory_controller
[move_group-1] [INFO] [1668761715.550150032] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: scaled_joint_trajectory_controller started execution
[move_group-1] [WARN] [1668761715.550168869] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-1] [ERROR] [1668761715.550151505] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-1] [ERROR] [1668761715.550216303] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller scaled_joint_trajectory_controller
[move_group-1] [INFO] [1668761715.550224960] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1668761715.559667313] [moveit_move_group_default_capabilities.move_action_capability]: Solution found but controller failed during execution
[rviz2-2] [INFO] [1668761715.560031615] [move_group_interface]: Plan and Execute request aborted
[rviz2-2] [ERROR] [1668761715.651703694] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached`

And I dont really know what is the problem, if its something that I code wrongly or I just cannot move the robot with Moveit and URSIM.

PD: I'm using scaled_joint_trajectory_controller.

Thanks in advance

gavanderhoorn, 2022-11-18:
> Goal was rejected by server

you'll want to check the terminal where you started the UR driver for information on *why* the goal was rejected.

It'd be nice if `moveit.simple_controller_manager` also logged the rejection reason (both an `error_code` as well as an `error_string` are part of the rejection result), but it doesn't seem like it does that any more (the ROS 1 version does I believe).

marioesteban11, 2022-11-18:
I've just seen that in the other terminal, this error shows up:

`[ur_ros2_control_node-1] [INFO] [1668761715.550004163] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [ERROR] [1668761715.550047689] [scaled_joint_trajectory_controller]: Can't accept new action goals. Controller is not running.
`

But I dont really know how to solve it

gavanderhoorn, 2022-11-18:
That's something you should ask the `Universal_Robots_ROS2_Driver` maintainers.

closed 2022-11-29 by marioesteban11

Ravik17, 2023-01-19:
@marioesteban11 Did you find a solution? I am facing the same issue.

LucaMIMOUNI, 2024-10-30:
@marioesteban11 Same here, would be nice to tell us what was the solution !

RGarewal, 2025-04-08:
The robot moved!(UR20 with polyscope 5.14 in my case)

Restarting the robot(moving to and fro from manual mode often results in robot not receiving moving command ) + freshly activating scaled_joint_trajectory_controller(which joint controller to use is crucial, just use one at a time, scaled_joint_trajectory_controlled is more robust option ) + running an external control urcap on teach pendant(so simply write a program with only external_control URcap command. Save this program and switch to automatic mode. In automatic mode, play this program and while its running switch to remote control mode using teach pendant.) made it happen!