moveit/moveit2 #1761: MoveIt stops/crashes after newest Humble update

skaeringur97, 2022-11-24:
### Description
This morning, I did apt-update and apt-upgrade. There was a big update (took about 8 minutes) and after that all my demos "crash". A college did the same update but uses Nav2 and not MoveIt and he has no problems. On each demo, the robot first moves for about one second before stopping. There are no error msgs or warnings, the robot simply stops moving.

### Your environment
* ROS2: [Humble]
* OS Version: e.g. Ubuntu 22.04
* Binary build for both ROS2 and MoveIt2

### Steps to reproduce
install ROS2 Humble and Moveit with binary install and run any MoveIt demo

sjahr, 2022-11-24:
Can you provide the terminal output when you run Moveit2?

skaeringur97, 2022-11-25:
I have verified that it is the update by running the same code on a computer that hasn't updated for 2 weeks. The code runs fine there.

gavanderhoorn, 2022-11-25:
Just to get it out of the way: have you rebuilt your workspace?

(there is no ABI compatibility guarantee in ROS (neither 1 nor 2). After updating with `apt`, workspaces *must* be rebuilt, or crashes are likely to occur)

skaeringur97, 2022-11-25:
@gavanderhoorn thanks for the reply. Yes, I have deleted my build, log and install folders and re-build. Unfortunately that didn't solve the problem

sjahr, 2022-11-25:
What about this error? 
```
[minimal_example-4] [ERROR] [1669287040.119821555] [moveit_ros.trajectory_execution_manager]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 2.665742 seconds). Stopping trajectory.
```
Which version of ros2_control are you using and does this message always appear when you run the simple examples?

skaeringur97, 2022-11-25:
@sjahr this error only shows after the robot stops before finishing the trajectory. I meant that there is no error about why it stops. 

> Which version of ros2_control are you using

I'm using version 2.16.0-1
> does this message always appear when you run the simple examples?

The above error only shows up in the example where I use MoveItCpp to execute (with `planning_components->execute();`). In the hybrid planning example no error msg shows. In both examples the robot moves a little before stopping.

>  it might get stuck in an infinite loop since there is no timeout check like this.

I'm not using hybrid planning in both examples. Does the same apply to MoveItCpp?

sjahr, 2022-11-25:
MoveItCpp uses the trajectory execution manager. Based on the error, it could be that for some reason, the execution doesn't finish successfully, that's what the error implies. Does the robot execute the whole trajectory before stopping?

skaeringur97, 2022-11-25:
@sjahr Sometimes it usually stops close to the end of the trajectory. When it manages to finish, it stops in the beginning of the next trajectory. I have only tested on fake hardware.

sjahr, 2022-11-28:
Can you verify that the trajectory execution action finishes successful and the next trajectory gets send to the robot?

skaeringur97, 2022-11-28:
The next line after `planning_components->execute();` is not executed, if that answers your question. I tried installing the docker but I was unable to get it working.. I can try again later when I have more time

vatanaksoytezer, 2022-11-28:
Seems like there are some ros2_control nodes dying, I would recommend checking the names. From your log:

```
[ros2_control_node-5] [INFO] [1669287031.628325609] [controller_manager]: Configuring controller 'panda_arm_controller'
[ros2_control_node-5] [ERROR] [1669287031.628376619] [controller_manager]: Could not configure controller with name 'panda_arm_controller' because no controller with this name exists
[ros2 run controller_manager spawner panda_arm_controller-6] [INFO] [1669287031.629186462] [spawner_panda_arm_controller]: Failed to configure controller
[ros2_control_node-5] [INFO] [1669287031.673832541] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node-5] [INFO] [1669287031.697335338] [controller_manager]: Loading controller 'panda_hand_controller'
[ros2 run controller_manager spawner panda_hand_controller-7] [INFO] [1669287031.724148949] [spawner_panda_hand_controller]: Loaded panda_hand_controller
[ros2_control_node-5] [INFO] [1669287031.725051029] [controller_manager]: Configuring controller 'panda_hand_controller'
[ros2_control_node-5] [INFO] [1669287031.725182489] [panda_hand_controller]: Action status changes will be monitored at 20Hz.
[ros2 run controller_manager spawner panda_hand_controller-7] [INFO] [1669287031.743958839] [spawner_panda_hand_controller]: Configured and activated panda_hand_controller
[ros2 run controller_manager spawner panda_arm_controller-6] [ros2run]: Process exited with failure 1
[ERROR] [ros2 run controller_manager spawner panda_arm_controller-6]: process has died [pid 15799, exit code 1, cmd 'ros2 run controller_manager spawner panda_arm_controller'].
[ros2 run controller_manager spawner joint_state_broadcaster-8] [ros2run]: Process exited with failure 1
[ERROR] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process has died [pid 15819, exit code 1, cmd 'ros2 run controller_manager spawner joint_state_broadcaster'].
```

sjahr, 2022-11-29:
Ok, the planning_components->execute() calls moveit_cpp->execute() [here](https://github.com/ros-planning/moveit2/blob/8dcbe40e1e169225b935f7c9fcc552768d7c86b0/moveit_ros/planning/moveit_cpp/src/planning_component.cpp#L399). Can you take a look into moveit_cpp->execute() and check if this returns? [this is the code](https://github.com/ros-planning/moveit2/blob/8dcbe40e1e169225b935f7c9fcc552768d7c86b0/moveit_ros/planning/moveit_cpp/src/moveit_cpp.cpp#L236).

vatanaksoytezer, 2022-11-29:
I just tested this with a fresh container. Seems like MoveGroupInterface works but I can reproduce what @skaeringur97 sees here with MoveItCpp. It fails in the execution step with the same error. I will go deeper into this, in the meantime, I recommend using a source build of the latest MoveIt2 main branch (main supports both humble and rolling).

skaeringur97, 2022-12-05:
Issue seems to be fixed for me. Thanks for the help @vatanaksoytezer and @sjahr ! I'll close this for now, feel free to re-open it if it isn't fixed for you

ronniethehood, 2022-12-06:
Hi @skaeringur97, I'm encountering the same issue since updating to the latest version of Humble. I've tried running with a binary build of Moveit2 and also the source build of the Moveit2 main branch. Mind explaining how/what fixed your issue?

skaeringur97, 2022-12-06:
@ronniethehood I just ran :
```
sudo apt update
sudo apt dist-upgrade
sudo apt autoremove
```
every morning and tested my demo. One morning it just worked after some update. I also have a binary install of ROS2 Humble btw

gordon-n-stevenson, 2022-12-07:
+1 Same issue as @ronniethehood @skaeringur97 -- built in linux on WSL using humble binaries and moveit main. Rolling seems to work

pinorobotics, 2022-12-07:
We also encountered it here https://github.com/ros-planning/moveit_resources/issues/157
Possibly duplicate issue.

gordon-n-stevenson, 2022-12-08:
Seems that moving to rolling binaries and moveit2 binaries (not source build) has solved the issue for us.

skaeringur97, 2022-12-09:
After the update, I now found that servo doesn't work for me. It either doesn't move at all, or moves a little bit before stopping. This is similar to the original issue, so I'll open this issue again instead of making a new one.

yeyanlei, 2022-12-12:
i met the same problem in the latest version of humble. Panda arm never finishes.

AndyZe, 2022-12-12:
If anybody is still building MoveIt2 from source, this would be some helpful information:

Bisect the issue by rolling back one commit at a time until the issue goes away. `git reset --hard`

Let us know what that commit is.

yeyanlei, 2022-12-13:
> Seems that moving to rolling binaries and moveit2 binaries (not source build) has solved the issue for us.

rolling binary is ok, thanks,  and i still consider ros2 version of humble.

yeyanlei, 2022-12-13:
> 
ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
moveit2 humble binary version, system stuck all the time. can you give me some help.
[image]

henningkayser, 2022-12-19:
I've just tried running humble on a fresh system, both source build and the binary succeeded for me without a crash whatsoever. The only thing I'm experiencing is a certain flakiness with the launch bringup, where sometimes controllers or the RViz panel fail to initialize. If that's good, planning+execution just work. I'm leaning towards a workspace issue. Are you sure you don't have any build artifacts from before the update? Do you have any upstream dependencies of MoveIt in your workspace (moveit_msgs, srdfdom, geometric_shapes, etc...)?

skaeringur97, 2022-12-20:
@henningkayser did you get MoveIt Servo running? Because that's the reason I re-opened this issue.

henningkayser, 2022-12-20:
@skaeringur97 I just tried it, Servo just works for me without issues. Would you mind sharing the output of `vcs branch` from your workspace? Also, are you running CycloneDDS or FastDDS?

skaeringur97, 2022-12-29:
@henningkayser I just updated again and still the Servo doesn't work for me. Here is the output of `vcs branch`
```
vcs branch
=== . (git) ===
master
```
It looks like I'm using FastDDS: `ros-humble-fastrtps/jammy,now 2.6.3-1jammy.20221202.001034 amd64 [installed,automatic]`

AndyZe, 2022-12-29:
Can you try switching to Cyclone? It should be a very easy switch:

`sudo apt install ros-rolling-rmw-cyclonedds-cpp`
# You may want to add this to ~/.bashrc to source it automatically
`export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`

We just updated the Rolling tutorial to recommend this, but it hasn't made it to the Humble tutorial yet. Sorry about that. I'll try to update the tutorials today.

skaeringur97, 2022-12-29:
I get this error when switching to Cyclone:
```
[INFO] [pick_and_place_demo-2]: process started with pid [52840]
[pick_and_place_demo-2] 1672328023.799129 [0] pick_and_p: selected interface "lo" is not multicast-capable: disabling multicast
[pick_and_place_demo-2] 1672328023.799260 [0] pick_and_p: Failed to find a free participant index for domain 0
[pick_and_place_demo-2] [ERROR] [1672328023.799324925] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error
[pick_and_place_demo-2] 
[pick_and_place_demo-2] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[pick_and_place_demo-2] This error state is being overwritten:
[pick_and_place_demo-2] 
[pick_and_place_demo-2]   'error not set, at ./src/rcl/node.c:263'
[pick_and_place_demo-2] 
[pick_and_place_demo-2] with this new error message:
[pick_and_place_demo-2] 
[pick_and_place_demo-2]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[pick_and_place_demo-2] 
[pick_and_place_demo-2] rcutils_reset_error() should be called after error handling to avoid this.
[pick_and_place_demo-2] <<<
[pick_and_place_demo-2] [ERROR] [1672328023.799365666] [rcl]: Failed to fini publisher for node: 1
[pick_and_place_demo-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[pick_and_place_demo-2]   what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at ./src/rcl/node.c:415
[pick_and_place_demo-2] Stack trace (most recent call last):
[pick_and_place_demo-2] #15   Object "/usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2", at 0xffffffffffffffff, in 
[pick_and_place_demo-2] #14   Object "/home/elias/riwo/hybrid-planning/install/demos/lib/demos/pick_and_place_demo", at 0x55b9db4172a4, in _start
[pick_and_place_demo-2] #13   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f9ce23f4e3f]
[pick_and_place_demo-2] #12   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f9ce23f4d8f]
[pick_and_place_demo-2] #11   Object "/home/elias/riwo/hybrid-planning/install/demos/lib/demos/pick_and_place_demo", at 0x55b9db416d98, in main
[pick_and_place_demo-2] #10   Object "/home/elias/riwo/hybrid-planning/install/demos/lib/demos/pick_and_place_demo", at 0x55b9db4253c8, in std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, char const (&) [25], char const (&) [1], rclcpp::NodeOptions&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, char const (&) [25], char const (&) [1], rclcpp::NodeOptions&)
[pick_and_place_demo-2] #9    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f9ce2b51c26, in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&)
[pick_and_place_demo-2] #8    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f9ce2b51530, in rclcpp::node_interfaces::NodeBase::NodeBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::Context>, rcl_node_options_s const&, bool, bool)
[pick_and_place_demo-2] #7    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7f9ce2b31278, in rclcpp::exceptions::throw_from_rcl_error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_error_state_s const*, void (*)())
[pick_and_place_demo-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f9ce26c323d, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
[pick_and_place_demo-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f9ce26c32b6, in std::terminate()
[pick_and_place_demo-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f9ce26c324b, in 
[pick_and_place_demo-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f9ce26b7bbd, in 
[pick_and_place_demo-2] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f9ce23f37f2]
[pick_and_place_demo-2] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f9ce240d475]
[pick_and_place_demo-2] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[pick_and_place_demo-2]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[pick_and_place_demo-2]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f9ce2461a7c]
[pick_and_place_demo-2] Aborted (Signal sent by tkill() 52840 1000)
```
I installed `sudo apt install ros-humble-rmw-cyclonedds-cpp` instead of the rolling version you commented with btw.

AndyZe, 2022-12-29:
OK. This is definitely a RMW issue. Aka networking nightmare that not too many people understand.

Are you working from home with your personal wifi or somewhere public like a coffee shop?

I'll attach my RMW config file shortly for you to try.

skaeringur97, 2022-12-29:
I'm working from home

AndyZe, 2022-12-29:
OK, here are some instructions I just tested, albeit on Rolling.

Create this Cyclone config file in /home/${USER}/cyclonedds.xml:

```
  <?xml version="1.0" encoding="UTF-8" ?>
  <CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <NetworkInterfaceAddress>wifi0</NetworkInterfaceAddress>
        </General>
    </Domain>
</CycloneDDS>
```

Replace wifi0 with your wifi interface name. You can find that with `ip link show`. Mine is `wlp0s20f3` for example.

Now add this to your bashrc to source it:

export CYCLONEDDS_URI=file:///home/${USER}/cyclonedds.xml

Of course, open a new terminal after this.

Any luck?

skaeringur97, 2022-12-29:
Now the driver node won't even start:
```
[ros2_control_node-1] 1672330856.014875 [0] ros2_contr: config: //CycloneDDS/Domain/General: 'NetworkInterfaceAddress': deprecated element (file:///home/elias/cyclonedds.xml line 5)
[ros2_control_node-1] 1672330856.014936 [0] ros2_contr: config: General/Interfaces: do not pass deprecated configuration General/{NetworkAddressString,MulticastRecvNetworkInterfaceAddresses,AssumeMulticastCapable}
[ros2_control_node-1] [ERROR] [1672330856.014961640] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error
[ros2_control_node-1] 
[ros2_control_node-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[ros2_control_node-1] This error state is being overwritten:
[ros2_control_node-1] 
[ros2_control_node-1]   'error not set, at ./src/rcl/node.c:263'
[ros2_control_node-1] 
[ros2_control_node-1] with this new error message:
[ros2_control_node-1] 
[ros2_control_node-1]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[ros2_control_node-1] 
[ros2_control_node-1] rcutils_reset_error() should be called after error handling to avoid this.
[ros2_control_node-1] <<<
[ros2_control_node-1] [ERROR] [1672330856.015009572] [rcl]: Failed to fini publisher for node: 1
[ros2_control_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[ros2_control_node-1]   what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at ./src/rcl/node.c:415
[robot_state_publisher-2] 1672330856.019453 [0] robot_stat: config: //CycloneDDS/Domain/General: 'NetworkInterfaceAddress': deprecated element (file:///home/elias/cyclonedds.xml line 5)
[robot_state_publisher-2] 1672330856.019552 [0] robot_stat: config: General/Interfaces: do not pass deprecated configuration General/{NetworkAddressString,MulticastRecvNetworkInterfaceAddresses,AssumeMulticastCapable}
[robot_state_publisher-2] [ERROR] [1672330856.019588468] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error
[robot_state_publisher-2] 
[robot_state_publisher-2] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[robot_state_publisher-2] This error state is being overwritten:
[robot_state_publisher-2] 
[robot_state_publisher-2]   'error not set, at ./src/rcl/node.c:263'
[robot_state_publisher-2] 
[robot_state_publisher-2] with this new error message:
[robot_state_publisher-2] 
[robot_state_publisher-2]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[robot_state_publisher-2] 
[robot_state_publisher-2] rcutils_reset_error() should be called after error handling to avoid this.
[robot_state_publisher-2] <<<
[robot_state_publisher-2] [ERROR] [1672330856.019662510] [rcl]: Failed to fini publisher for node: 1
[robot_state_publisher-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[robot_state_publisher-2]   what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at ./src/rcl/node.c:415
[rviz2-3] 1672330856.191790 [0]      rviz2: config: //CycloneDDS/Domain/General: 'NetworkInterfaceAddress': deprecated element (file:///home/elias/cyclonedds.xml line 5)
[rviz2-3] 1672330856.191930 [0]      rviz2: config: General/Interfaces: do not pass deprecated configuration General/{NetworkAddressString,MulticastRecvNetworkInterfaceAddresses,AssumeMulticastCapable}
[rviz2-3] [ERROR] [1672330856.191988301] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error
[rviz2-3] 
[rviz2-3] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[rviz2-3] This error state is being overwritten:
[rviz2-3] 
[rviz2-3]   'error not set, at ./src/rcl/node.c:263'
[rviz2-3] 
[rviz2-3] with this new error message:
[rviz2-3] 
[rviz2-3]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[rviz2-3] 
[rviz2-3] rcutils_reset_error() should be called after error handling to avoid this.
[rviz2-3] <<<
[rviz2-3] [ERROR] [1672330856.192082789] [rcl]: Failed to fini publisher for node: 1
[rviz2-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rviz2-3]   what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at ./src/rcl/node.c:415
[spawner-9] 1672330856.192873 [0]    spawner: config: //CycloneDDS/Domain/General: 'NetworkInterfaceAddress': deprecated element (file:///home/elias/cyclonedds.xml line 5)
[spawner-9] 1672330856.192947 [0]    spawner: config: General/Interfaces: do not pass deprecated configuration General/{NetworkAddressString,MulticastRecvNetworkInterfaceAddresses,AssumeMulticastCapable}
[spawner-9] [ERROR] [1672330856.192986601] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error
[spawner-9] 
[spawner-9] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[spawner-9] This error state is being overwritten:
[spawner-9] 
[spawner-9]   'error not set, at ./src/rcl/node.c:263'
[spawner-9] 
[spawner-9] with this new error message:
[spawner-9] 
[spawner-9]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[spawner-9] 
[spawner-9] rcutils_reset_error() should be called after error handling to avoid this.
[... 115 lines cut ...]
[ERROR] [spawner-6]: process has died [pid 55965, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner speed_scaling_state_broadcaster --controller-manager /controller_manager --ros-args'].
[... cut ...] 10e_update_rate.yaml --params-file /opt/ros/humble/share/ur_robot_driver/config/ur_controllers.yaml'].
[ERROR] [robot_state_publisher-2]: process has died [pid 55957, exit code -6, cmd '/opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args --params-file /tmp/launch_params_xh9ckkv8'].
[ERROR] [rviz2-3]: process has died [pid 55959, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/ur_description/rviz/view_robot.rviz --ros-args -r __node:=rviz2'].
```

AndyZe, 2022-12-29:
Hmm. I'm basically out of ideas. I'd post your Cyclone config and that^ output to an issue on the Cyclone repo and see what they say. Sorry.

https://github.com/eclipse-cyclonedds/cyclonedds

AndyZe, 2023-01-03:
[skaeringur97](https://github.com/skaeringur97) i must ask, have you tried to update then do a clean build?

- Delete the build/install/log folders from your workspace.  `rm -rf ./build ./install ./log`
- `sudo apt update`
- `sudo apt upgrade`
- `colcon build`

skaeringur97, 2023-01-03:
@AndyZe yes, this was also pointed out by @gavanderhoorn 

I´ve done these steps after each update to see if it has been fixed.

peterdavidfagan, 2023-01-04:
Hi @skaeringur97, have you also tried updating all ROS dependencies? I was facing the errors you mentioned above but following @gavanderhoorn's suggestion helped resolve the issues in my workspace. 

I am working on MoveIt today, if you feel it would be worthwhile I'd be happy to look to recreate the errors you mention in Docker.

skaeringur97, 2023-01-04:
@peterdavidfagan I update the dependencies in my workspace with `rosdep install -r --from-paths . --ignore-src --rosdistro humble -y` and tried again now to do the steps @AndyZe suggested. The only packages not upgraded are:
```
The following packages have been kept back:
  gnome-remote-desktop grub-common grub-pc grub-pc-bin grub2-common
  libnautilus-extension1a nautilus nautilus-data python3-update-manager
  update-manager update-manager-core
```
I still get the issue, similar to the original issue, that servo starts but is not able to finish the movement. This has been proven to work on another laptop with an older binary install btw.

sjahr, 2023-06-01:
@skaeringur97 Is this still an issue? If yes, feel free to re-open the issue

closed 2023-06-01 by sjahr