moveit/moveit2 #1762: Segmentation fault in kdl_kinematics_plugin.cpp

SmarkSauer, 2022-11-24:
### Description

Randomly when setting from IK the kdl_kinematics_plugin goes into a segemetation fault. 

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Moveit2Docker Binary
* 
### Backtrace or Console output
kdl_kinematics_plugin
[app-2] [DEBUG] [1669304583.954376192] [moveit_kinematics_base.kinematics_base]: searchPositionIK: Position request pose is 2.3466 -0.26803 1.252 0 0 -0.71934 0.694658
[app-2] [DEBUG] [1669304583.954396097] [moveit_kinematics_base.kinematics_base]: Input: [       2.817       0.812    -3.47321     8.18559           0]
[app-2] [DEBUG] [1669304583.954413836] [moveit_kinematics_base.kinematics_base]: [  0] delta_twist: [           0,           0,           0,           0,          -0,  -0.0349066]
[app-2] Stack trace (most recent call last) in thread 386373:
...
[app-2] #7    Source "./robot_state/src/robot_state.cpp", line 1479, in setFromIK [0x7fd6b0147971]
[app-2] #6    Source "./robot_state/src/robot_state.cpp", line 1490, in setFromIK [0x7fd6b014778a]
[app-2] #5    Source "./robot_state/src/robot_state.cpp", line 1573, in setFromIK [0x7fd6b0147399]
[app-2] #4    Source "./robot_state/src/robot_state.cpp", line 1809, in setFromIK [0x7fd6b0146ca2]
[app-2] #3    Source "./kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp", line 310, in searchPositionIK [0x7fd6a6fc7f38]
[app-2] #2    Source "./kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp", line 400, in searchPositionIK [0x7fd6a6fd2776]
[app-2] #1    Source "./kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp", line 483, in CartToJnt [0x7fd6a6fd036f]
[app-2] #0    Source "./kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp", line 527, in clipToJointLimits [0x7fd6a6fc985a]
[app-2] Segmentation fault (Address not mapped to object [0x7fda17a81c38])

Even if there is a programming error by user, vectors etc. should be checked for size. Debug output didnt help much either.

tylerjw, 2022-11-29:
Would you mind installing the debug symbols and reproducing the bug so we can get a stack trace?

henningkayser, 2023-01-25:
Closing due to inactivity. @SmarkSauer please reopen and respond if you are still running into this issue. Side-note: if this happens randomly, this seems to be a (possibly unrelated) memory issue, not a typical vector bound bug.

closed 2023-01-25 by henningkayser