moveit/moveit2 #1768: Hybrid planner crashes after succeeding

skaeringur97, 2022-11-30:
### Description

Sometimes when running the hybrid planner, a ABORT signal is sent just before the plan succeeds. This makes the goal handle send SUCCEEDED just after sending ABBORTED. This results in the following error msg:
```
[component_container_mt-1]   what():  goal_handle attempted invalid transition from state ABORTED with event SUCCEED, at ./src/rcl_action/goal_handle.c:95
``` 

Maybe some lock might be good to make sure that only one state is sent.

### Your environment
* ROS2: [Humble]
* OS Version: Ubuntu 22.04
* Binary build

### Steps to reproduce
Maybe difficult to reproduce without running a hybrid planning demo often and for a long time.

### Expected behaviour
Hybrid planning aborts.

### Actual behaviour
Hybrid planning succeeds and node crashes.

### Backtrace or Console output

<details>
<summary>Console output (only last few lines since it was running for a while before this happened)</summary>
<pre>
[component_container_mt-1] [INFO] [1669802585.721241181] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'ur_manipulator'
[component_container_mt-1] [INFO] [1669802585.721559211] [local_planner_component]: Received local planning goal request
[component_container_mt-1] [INFO] [1669802585.921812343] [local_planner_component]: The local planner is solving...
[component_container_mt-1] [INFO] [1669802586.522100202] [local_planner_component]: Collision ahead, holding current position
[component_container_mt-1] [INFO] [1669802586.522241134] [global_planner_component]: Received global planning goal request
[component_container_mt-1] [WARN] [1669802586.522747311] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'ur_manipulator' using planner 'ompl'. Will use defaults instead.
[component_container_mt-1] [WARN] [1669802586.522868646] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container_mt-1] [INFO] [1669802586.522955731] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container_mt-1] [INFO] [1669802586.622116203] [local_planner_component]: Collision ahead, holding current position
[pick_and_place_inside_ur10e_cage-2] [ERROR] [1669802586.721490928] [test_hybrid_planning_client]: Service timed out.
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802586.721565900] [test_hybrid_planning_client]: Local goal accepted by server
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802586.721582911] [test_hybrid_planning_client]: Global goal accepted by server
[component_container_mt-1] [INFO] [1669802588.021805685] [local_planner_component]: The local planner is solving...
[component_container_mt-1] [INFO] [1669802588.028746503] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'ur_manipulator'
[pick_and_place_inside_ur10e_cage-2] [ERROR] [1669802589.029018311] [test_hybrid_planning_client]: Service timed out.
[component_container_mt-1] [INFO] [1669802590.121789968] [local_planner_component]: The local planner is solving...
[component_container_mt-1] [INFO] [1669802592.221800022] [local_planner_component]: The local planner is solving...
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802592.522098211] [test_hybrid_planning_client]: Hybrid planning goal succeeded
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802592.522110598] [test_hybrid_planning_client]: Pre-grasp stage succeeded
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802592.522144688] [test_hybrid_planning_client]: Sending hybrid planning goal
[component_container_mt-1] [INFO] [1669802592.522226937] [hybrid_planning_manager]: Received goal request
[component_container_mt-1] [INFO] [1669802592.522526942] [global_planner_component]: Received global planning goal request
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802592.522636074] [test_hybrid_planning_client]: Global goal accepted by server
[component_container_mt-1] [WARN] [1669802592.522979777] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'ur_manipulator' using planner 'ompl'. Will use defaults instead.
[component_container_mt-1] [WARN] [1669802592.523139441] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container_mt-1] [INFO] [1669802592.523231453] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container_mt-1] [INFO] [1669802592.865112287] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'ur_manipulator'
[component_container_mt-1] [INFO] [1669802592.865378965] [local_planner_component]: Received local planning goal request
[pick_and_place_inside_ur10e_cage-2] [ERROR] [1669802593.865406781] [test_hybrid_planning_client]: Service timed out.
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802593.865480845] [test_hybrid_planning_client]: Local goal accepted by server
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802593.865523960] [test_hybrid_planning_client]: Hybrid planning goal succeeded
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802593.865577130] [test_hybrid_planning_client]: Place position: 0.3, 1.1, 0.85
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802593.865593026] [test_hybrid_planning_client]: Sending hybrid planning goal
[component_container_mt-1] [INFO] [1669802593.865674716] [hybrid_planning_manager]: Received goal request
[component_container_mt-1] [INFO] [1669802593.865903293] [global_planner_component]: Received global planning goal request
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802593.866015083] [test_hybrid_planning_client]: Global goal accepted by server
[component_container_mt-1] [WARN] [1669802593.866437996] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'ur_manipulator' using planner 'ompl'. Will use defaults instead.
[component_container_mt-1] [WARN] [1669802593.866577581] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[component_container_mt-1] [INFO] [1669802593.866693362] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[component_container_mt-1] [INFO] [1669802594.729966981] [moveit.ros_planning_interface.planning_component]: Deleting PlanningComponent 'ur_manipulator'
[component_container_mt-1] [INFO] [1669802594.730352966] [local_planner_component]: Received local planning goal request
[component_container_mt-1] [INFO] [1669802594.830641666] [local_planner_component]: The local planner is solving...
[... 39 lines cut ...]
[pick_and_place_inside_ur10e_cage-2] [ERROR] [1669802601.952360156] [test_hybrid_planning_client]: Hybrid planning goal was aborted
[pick_and_place_inside_ur10e_cage-2] [ERROR] [1669802601.952381369] [test_hybrid_planning_client]: Pre-grasp stage failed for apple39
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802601.952398235] [test_hybrid_planning_client]: removing apple39 from environment
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802601.953844286] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00138769 seconds
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802601.953857651] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802601.953863772] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[pick_and_place_inside_ur10e_cage-2] [WARN] [1669802601.963020740] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[pick_and_place_inside_ur10e_cage-2] [INFO] [1669802601.963113910] [test_hybrid_planning_client]: Sending hybrid planning goal
[component_container_mt-1] [INFO] [1669802601.963232304] [hybrid_planning_manager]: Received goal request
[component_container_mt-1] [INFO] [1669802601.963420229] [local_planner_component]: Received request to cancel local planning goal
[... cut ...]
[component_container_mt-1] [INFO] [1669802602.954712899] [local_planner_component]: The local planner is solving...
[pick_and_place_inside_ur10e_cage-2] [ERROR] [1669802603.254333073] [test_hybrid_planning_client]: Service timed out.
[component_container_mt-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[component_container_mt-1]   what():  goal_handle attempted invalid transition from state ABORTED with event SUCCEED, at ./src/rcl_action/goal_handle.c:95
</pre>
</details>

frankchan12138, 2024-10-31:
> ### Description
> Sometimes when running the hybrid planner, a ABORT signal is sent just before the plan succeeds. This makes the goal handle send SUCCEEDED just after sending ABBORTED. This results in the following error msg:
> 
> ```
> [component_container_mt-1]   what():  goal_handle attempted invalid transition from state ABORTED with event SUCCEED, at ./src/rcl_action/goal_handle.c:95
> ```
> 
> Maybe some lock might be good to make sure that only one state is sent.
> 
> ### Your environment
> * ROS2: [Humble]
> * OS Version: Ubuntu 22.04
> * Binary build
> 
> ### Steps to reproduce
> Maybe difficult to reproduce without running a hybrid planning demo often and for a long time.
> 
> ### Expected behaviour
> Hybrid planning aborts.
> 
> ### Actual behaviour
> Hybrid planning succeeds and node crashes.
> 
> ### Backtrace or Console output
> Console output (only last few lines since it was running for a while before this happened)

Hello, I've also came across this problem. Have you figure out how to solve this now?

skaeringur97, 2024-11-01:
Hi @frankchan12138 

Unfortunately I don't remember if I ever solved this. I stopped working on this such a long time ago.
Sorry to be of no help.. Good luck :)