moveit/moveit2 #1777: RobotTrajectory unwind() functions only account for one extra "wrap"

AndyZe, 2022-12-02:
Unwinding should account for any angle. For example, a UR3e robot's wrist can rotate infinitely. But, the [RobotTrajectory unwind()](https://github.com/ros-planning/moveit2/blob/f0d3202d637569ccac5fcc357a37e0d12a077c87/moveit_core/robot_trajectory/src/robot_trajectory.cpp#L158) function only accounts for one "unwrapping" of 2*pi.

```
      if (last_value > current_value + M_PI)
      {
        running_offset += 2.0 * M_PI;
      }
      else if (current_value > last_value + M_PI)
      {
        running_offset -= 2.0 * M_PI;
      }
```

AndyZe, 2022-12-02:
@mechwiz you've been deep in this code recently. Do you agree?

mechwiz, 2022-12-02:
Nice catch!
Yeah I agree that it looks like it only handles one wrap. This probably hasn't caused much issues since the waypoints within a trajectory are usually the way you'd expect them to be (i.e. if the first waypoint has a continuous joint at 20 rads, then the next waypoint will also be the same or close to it depending if that joint moves in the trajectory so they all should be winded up in the same wrap cycle correctly). The only time this would be an issue is if you explicitly change one of the waypoint's continuous joint values to be something more than one wrap different than the other waypoints or if you are chaining trajectories (and then call unwind) and the continuous joint values within each of them are more than one wrap apart

mechwiz, 2022-12-11:
I believe I fixed this in #1772 by ensuring all continuous joints have their bounds enforced when unwinding so there can never be more than one wrap

closed 2022-12-21 by sjahr